google-deepmind/mujoco #281: Stretching the tendons to drive the model

ycntzx, 2022-05-17:
In my model I used tendons as strings and I wanted to control the model by the length of the strings (Figure 1）. However, when I pull the string, the model starts to move accordingly (Figure2） but then it slowly changes back to its original shape (Figure 3）. Why does this happen? Is it because the tendon is elastic? What's wrong with my program? Thx
[image]
[image]
[image]
<mujoco model="roboter">
    <option timestep="0.000002" iterations="50" apirate="1000" gravity="0 0 0" solver="Newton" tolerance="1e-10" cone="elliptic"/>
    <compiler angle="degree" meshdir="meshes" balanceinertia="true"/>

        <default>
        <light castshadow="false" diffuse="0 0 0" />
        <camera fovy="60" />
        <site type="sphere" rgba="1 0 0 1" size="0.4"/>
    </default>

    <asset>

        <mesh file="ManipulatorDisk_start.STL" />
        <mesh file="ManipulatorDisk_grob.STL"/>
        <mesh file="ManipulatorEndDisk.STL"/>

        <texture name="floor_tile" file="meshes/floor_tile.png" width="3000" height="3000" type="2d"/>
        <material name="tile" texture="floor_tile" specular="0" shininess="0" reflectance="0" emission="1" />

        <material name="carbon" specular="1" shininess="1" reflectance="1" rgba="0.82 0.41 0.12 1" emission="1" />
        <texture name="foam_blocks" file="meshes/foam_blocks.png" width="2744" height="2744" type="2d"/>
        <material name="foam" texture="foam_blocks" specular="0" shininess="0" reflectance="0" emission="1" />
    </asset>

    <default>

		<default class="G">
                        <joint type="hinge" limited="true" range="-28.12 28.12" springdamper="0.005 10" />
                        </default>
		<default class="shoko">
						<geom type="mesh" pos="15 0 0"  euler="90 -90 0" rgba="1 0.5 0.31 1" friction="1 0.5 0.01" mesh="ManipulatorDisk_grob"/>
						</default>
		<default class="blue">
						<geom type="mesh" rgba="0 .31 .61 1" pos="0 0 0"  euler="90 0 0" friction="1 0.5 0.01" mesh="ManipulatorDisk_grob"/>
						</default>
        <default class="red">
						<geom type="mesh" rgba="1 0 0 1" pos="15 0 0"  euler="90 -90 0" friction="1 0.5 0.01" mesh="ManipulatorEndDisk"/>
						</default>

	 </default>

 <worldbody>
        <geom name="floor_foam_block" pos="0 0 0" size="100 100 0.1" type="plane" rgba="1 0.83 0.61 1" material="foam"/>
        <geom name="floor_grey_tile" pos="0 0 -0.001" size="120 120 0.1" type="plane" material="tile"/>
        <light directional="true" pos="-0.5 0.5 3" dir="0 0 -1" />

        <body name="base_Hand" pos="0 0 0">
             <geom name="ManipulatorDisk_start" type="mesh" mesh="ManipulatorDisk_start"  pos="0 0 0" euler="90 0 0" material="carbon" />
             <site name="b0" pos="2.49 -5.23 46"/> <site name="b00" pos="11.39 -3.61 46"/> <site name="b000" pos="9.77 -12.57 46"/>

        <body name="ball0" pos="2.49 -5.23 20">
             <inertial pos="0 0 0" mass="0.0000001" diaginertia="0.3 0.3 0.3"/>
             <site name="s0" type="sphere" size="0.4" pos="0 0 0"/>
             <joint name="j0" pos="0 -7.5 -0.75" type="slide" axis="0 0 1" />
        </body>

        <body name="ball00" pos="11.39 -3.61 20">
             <inertial pos="0 0 0" mass="0.0000001" diaginertia="0.3 0.3 0.3"/>
             <site name="s00" type="sphere" size="0.4"/>
             <joint name="j00" pos="0 -7.5 -0.75" type="slide" axis="0 0 1" />
        </body>

        <body name="ball000" pos="9.77 -12.57 20">
             <inertial pos="0 0 0" mass="0.0000001" diaginertia="0.3 0.3 0.3"/>
             <site name="s000" type="sphere" size="0.4"/>
             <joint name="j000" pos="0 -7.5 -0.75" type="slide" axis="0 0 1" />
[... 147 lines cut ...]
            <site site="B111" />
            <site site="b222" />
            <site site="B222" />
            <site site="b333" />
            <site site="B333" />
            <site site="b444" />
            <site site="B444" />
            <site site="b555" />
            <site site="B555" />
            <site site="b666" />
            <site site="B666" />
            <site site="b777" />
            <site site="B777" />
            <site site="b888" />
            <site site="B888" />
            <site site="b999" />
        </spatial>
</tendon>

    <actuator>
    <position  name="t1" ctrllimited="true" ctrlrange="-20 0" joint="j0" kp="100"/>
    <position  name="t2" ctrllimited="true" ctrlrange="-20 0" joint="j00" kp="100"/>
    <position  name="t3" ctrllimited="true" ctrlrange="-20 0" joint="j000" kp="100"/>
    </actuator>
</mujoco>.

ycntzx, 2022-05-17:
[roboter.zip](https://github.com/deepmind/mujoco/files/8709595/roboter.zip)
@yuvaltassa Thank you very much! I uploaded the zip file and everything is in there.
Just like in reality I wish the rope could only pull.
But doesn't range="70 71" mean that the length of the rope is limited to 70 to 71?
If I want to set up the tendon like a rope without elasticity and this rope is 70 in length，should I set range="0 70"?

ycntzx, 2022-05-18:
Hi！ @yuvaltassa 

1. I want to drive the model in two ways. One is to apply a force to the drive rope to drive the model and the other is to control the model by pulling on the drive rope and changing the stretch length. That is, different forces are applied to the tendons to change the posture of the model and changing the length of the tendons to change the posture of the model.

When I type these:
[image]

**Do these actuators on the tendon change the length of the tendon? For as can be seen in the perspective view of the Mujoco simulator, the rope (Purple) in the base is indeed stretched.**  
[image]

**And whether the value -12.2 for t3 in the control section of the diagram represents the length of the tendon stretch is 12.2*100=1220?**  This value has been bothering me for a long time :(

2. The same as before，when I control the tendon the model starts to move accordingly but then it slowly changes back to its original position. I have tried to change the range，the lower range is set to 0，but it doesn't work. **How can I keep the model from going back to its original state?**

3. my timestep is very tiny because there is always a warning:  Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. So had to make my time steps very small. When i removed the contact:
[image]

I noticed no difference compared to before either it seems. The current timestep setting is 0.00005, which is really small :(
**By the way, where can I set friction between disks when the contacts are removed?**

I'm so sorry I have so many questions, thanks again!

yuvaltassa, 2022-05-19:
Hi,

Please note I changed the actuators from `<position>` to `<motor>`, so they are now applying forces directly, not trying to maintain a particular length.

I recommend that you take a step back from your model and write a few small test models using primitive geoms (so no assets required). Read the relevant documentation section carefully ([actuation](https://mujoco.readthedocs.io/en/latest/computation.html#actuation-model), [actuator shortcuts](https://mujoco.readthedocs.io/en/latest/modeling.html#actuator-shortcuts), [tendons](https://mujoco.readthedocs.io/en/latest/XMLreference.html#tendon)), and if there is some behaviour you encounter that doesn't make sense to you, you can post your minimal model here (in a new issue), and we will help you.

It is much easier for us to help you if the model is as minimal as possible, yet demonstrates the issue that you are experiencing, rather than a complicated model like the one you are actually trying to build.

ycntzx, 2022-05-21:
@yuvaltassa Thank you for your help! 
when I use this actuator : <motor name="t1" tendon="t1" ctrllimited="true" ctrlrange="-20 0" gear="100"/>
The motor outputs a force. In the control window of the simulator, the value of t1 can be selected from -20 to 0. **If the value of t1 is chosen to be -10, does this mean that the magnitude of the force is 10?** But  I don't know what the unit of force is. I know that my unit of length is millimetres.
I have read the relevant documentation section about In MuJoCo basic physical units are undefined. But with my length in millimetres it is now neither MKS nor CGS. So what is the unit of force ?
Thank you again！

yuvaltassa, 2022-05-21:
Hi, 

Not to be rude, but the fact that you are asking this question implies that you should probably not be using millimetres. It is certainly possible to use millimetres, but then you need to be able to do the dimensional analysis yourself and figure out all your units.

I see that you already have clauses like `<inertial mass="0.0002" diaginertia="0.3 0.3 0.3"/>`, where did this come from? CAD software or auto-calculated by MuJoCo? `mass` is in units of mass and `diaginertia` is in units of mass*length^2.

I strongly recommend that you switch to MKS or, if your model is small (which I think it is), CGS. Remember that meshes have a `scale` attribute. If you switch to CGS, remember to set gravity to `0 0 -981`.

ycntzx, 2022-05-21:
@yuvaltassa  oh you are my hero! 
I also realised what the problem was, and I deleted all of those now. Like you said before, build a simple model first.
Maybe I understand what you are saying and I shouldn't have to get hung up on a specific numerical unit. Length in mm is the dimension drawn in Solidworks. In fact, I could totally scale it up and put it in the MKS, right? So everything is perfect, the length is meters, the mass is kilograms and the force is Newtons.
I just found something interesting, when I delete the code <inertial pos="0 0 0" mass="0.0002" diaginertia="0.3 0.3 0.3"/>, the motor that acts on the tendon doesn't seem to work.  Do you know the reason?