google-deepmind/mujoco #282: How to visualize xfrc_applied?

rohanpsingh, 2022-05-19:
Sorry if this is very basic.

I'm applying some forces to my robot programmatically through `data->xfrc_applied`,
but I can't see those forces in my renderer (I tried in python and c++).
I can see the perturbations applied through the mouse just not the ones in xfrc_applied...

Thanks~!

yuvaltassa, 2022-05-19:
There is not built in visualisation for that. How would you imagine these wrenches could be visualised?

However you can draw 3D visual objects directly into the scene, and make your own visualisation.  
See discussion in #132

closed 2022-05-19 by yuvaltassa

rohanpsingh, 2022-05-23:
Thanks for your response.

> How would you imagine these wrenches could be visualised?

Maybe something similar to mouse perturbations? It wouldn’t look so neat but at least we’ll get some sense of the force magnitudes. Thanks for the suggestion about making my own visualization. I’ll try that soon.