google-deepmind/mujoco #285: [Python] How to get the depth images

rollingball-3, 2022-05-20:
Hi, I'm a beginner. 
I'm trying to get both RGB and depth images during simulation. Here is my way:
I use the function `mjr_readPixels` similar to the `render_test.py`.

```python
 self.context = mujoco.MjrContext(self.model, mujoco.mjtFontScale.mjFONTSCALE_100)
mujoco.mjr_render(self.viewport, self.scene, self.context)
upside_down_image = np.empty((self.image_height, self.image_width, 3), dtype=np.uint8)
upside_down_depth = np.empty((self.image_height, self.image_width, 1))
mujoco.mjr_readPixels(rgb=upside_down_image, depth=upside_down_depth, viewport=self.viewport, con=self.context)
```

The RGB images can be well obtained but the depth images can not. 
Is there something wrong with my usage? Is it because of the python bindings? Any help would be appreciated.

yuvaltassa, 2022-05-20:
Check out our [tutorial](https://github.com/deepmind/dm_control) in the `dm_control` repo. Depth rendering is covered there, I believe.

closed 2022-05-20 by yuvaltassa

rollingball-3, 2022-05-20:
这是系统自动回复邮件。我已经收到了您的邮件并会尽快回复。麻烦您耐心等待。

Zialo, 2022-10-24:
Hi, you can obtain both images changeing the mode='human' to 'depth_array' on the render() function. You will receive a tuple with both arrays

rollingball-3, 2022-10-24:
> Hi, you can obtain both images changeing the mode='human' to 'depth_array' on the render() function. You will receive a tuple with both arrays
Did you mean the function in mujoco-py?

rollingball-3, 2022-10-24:
@Zialo Did you mean the function in mujoco-py?

Zialo, 2022-10-24:
That's it. The function within mujoco_env.py

rollingball-3, 2022-10-24:
Thanks! I use the render() function in dm_control right now, which can also obtain both arrays