google-deepmind/mujoco #294: Misalignment with self-compiled libmujoco.so

Trinkle23897, 2022-05-23:
This is a follow-up of https://github.com/deepmind/mujoco/issues/259#issuecomment-1113201972

To reproduce:

1. I strictly followed the step in https://github.com/deepmind/mujoco/blob/main/.github/workflows/build.yml to generate `libmujoco.so.2.2.0` under `build/lib`, with ubuntu 20.04;
2. I manually replaced the `libmujoco.so.2.2.0` in `~/.local/lib/python3.8/site-packages/mujoco` with my own copy created by step 1;
3. I write a script to check the result, however, it is not exactly matched.

```python
import numpy as np
from dm_control import suite

total = 500  # less than max 1000
seed = 0
np.random.seed(seed)
action = np.random.uniform(-1, 1, size=(total, 4))
print(action.sum())

env = suite.load("hopper", "hop", {"random": seed})
ts = env.reset()
for i in range(total):
    ts = env.step(action[i])
print(ts.observation["position"])
```

```bash
# run twice for the original .so
$ python3 align_suite.py
14.543042030504985
[-0.90433844  1.08425238  0.44559735  0.19416702  0.07538787 -0.75478539]

$ python3 align_suite.py
14.543042030504985
[-0.90433844  1.08425238  0.44559735  0.19416702  0.07538787 -0.75478539]

# run twice for the new .so
python3 align_suite.py
14.543042030504985
[-0.90433844  1.08425237  0.44559734  0.19416701  0.07538795 -0.7547854 ]

$ python3 align_suite.py
14.543042030504985
[-0.90433844  1.08425237  0.44559734  0.19416701  0.07538795 -0.7547854 ]
```

The two .so : [mujoco_so.tar.gz](https://github.com/deepmind/mujoco/files/8757681/mujoco_so.tar.gz), `libmujoco.so.2.2.0` is generated in my side and `libmujoco.so.2.2.0.old` is the original one.

Trinkle23897, 2022-05-23:
Three domains in suite are affected by this issue: `hopper`, `humanoid`, `walker`. `manipulator` is also affected but when step==1000 the difference is 1e-5, which is negligable.

nimrod-gileadi, 2022-05-23:
Hey, thanks for the detailed bug report.

One major difference between the binaries we shipped and the configuration in build.yml is that we built the release binaries with clang-13, rather than gcc-10. We also used libc++ rather than libstdc++. At the moment we're using GitHub Actions for testing this alternative build configuration.

Already, our aarch64 builds give numerically slightly different results from the x86_64 builds. We looked into making them identical, but found no way to specify strict floating point operations on aarch64 with Clang. For that reason we **don't** use ffp-model=strict, and floating point contraction is possible.

So it's not expected that the internal build and the code you build from source will give numerically identical results, unfortunately, and after 1000 random steps, errors will accumulate.

That said, we'll take a look to see if there is any surprising numerical difference.

Trinkle23897, 2022-05-24:
> we'll take a look to see if there is any surprising numerical difference.

If you're going to re-run the suite benchmark result, would you please use envpool to evaluate by the way? [That won't take too much effort](https://envpool.readthedocs.io/en/latest/env/dm_control.html) and can get a 2x free speedup compared with dm_control. I've aligned per step with dm_control but just want to make sure everything works correctly. Thanks in advance!

nimrod-gileadi, 2022-05-24:
I like the envpool effort, but for the purpose of this question, I rather not introduce new potential sources of mismatch.

On my own end, we have some internal golden value tests that exercise most features in MuJoCo. I ran a comparison of a build with clang-13 and libc++ vs the config in build.yml, and found no numeric differences.

I thought the differences might be due to your build not having AVX turned on, but I confirmed that there are AVX instructions in your binary.

Could you please paste the exact commands you used to build your library?

> I've aligned per step with dm_control but just want to make sure everything works correctly.

I don't understand this. Can you rephrase?

Trinkle23897, 2022-05-24:
> I don't understand this. Can you rephrase?

No need to rebuild. The newest version of envpool uses `.so` file provided by clang (instead of self-compiled), and just `pip install envpool dm_control --upgrade` then run the evaluation of the existing well-trained agent to see if there's any mismatch of the benchmark result.

nimrod-gileadi, 2022-05-24:
To follow up: I was able to reproduce the bug you described in the top comment, and confirmed that the difference is down to gcc vs clang. This issue didn't turn up in our golden tests, but when I used the hopper model they do show up.

As I explained in a previous comment, small differences are what we expect, and we won't be able to completely eliminate them.

Regarding running agents on envpool, it's a bit outside of the scope of my current work, but please let me know if you find any interesting results that way.