google-deepmind/mujoco #299: How to set the actuator on the tendon?

LuiseMq, 2022-05-25:
**How to ask for help**

- Take a step back and tell us what you're trying to accomplish, and in what context. The more information you give, the better the answers you will get.

- If relevant, include a ***minimal*** model that demonstrates the problem. If the model is small, include it as inline XML. If it requires binary assets, attach it as a zip file.

- Include a screenshot or video, if relevant.

nimrod-gileadi, 2022-05-26:
Proposed new text:

Actuators are used to apply forces - they don’t set quantities like tendon length directly.

The [docs](https://mujoco.readthedocs.io/en/latest/computation.html#geactuation) explain how these forces are calculated. When using actuators on tendons the `l(q)` in the equations is equal to the tendon length (as mentioned under the [`tendon` attribute docs](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-general)).

You could start with a `<position>` actuator rather than the `<general>` actuator to control the tendon length, and set `ctrllimited="true" ctrlrange="30 50"` to match the control input to the desired tendon length.

You’ll need to choose an appropriate value for `kp` on the actuator.

Once that works, you can try adding damping to your tendon if needed to stop oscillations, or move to the general actuator and tweak some of the parameters. See the [docs](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-position) for how a `<position>` actuator relates to `<general>`.

To put it together:

```
<mujoco>
  <option density="1000"/>
  <worldbody>
    <geom type="plane" size="10 10 .1"/>
    <light pos="0 0 10"/>
    <site name="anchor" pos="0 0 5" size="0.03" rgba="0 0 1 1"/>
    <body name="box" pos="0 0 2">
      <freejoint/>
      <geom type="box" size=".5 .5 .5"/>
      <site name="box" pos=".5 .5 .5" rgba="0 0 1 1" size="0.03"/>
    </body>
  </worldbody>
  <tendon>
    <spatial name="tendon" limited="true" range="0 2" damping="10000">
      <site site="anchor"/>
      <site site="box"/>
    </spatial>
  </tendon>
  <actuator>
    <position
      name="tendon" tendon="tendon"
      ctrllimited="true" ctrlrange="1 2"
      kp="100000"/>
  </actuator>
</mujoco>
```

closed 2022-05-26 by nimrod-gileadi

yuvaltassa, 2022-05-26:
Small followup, note the model above has `<option density="1000"/>`, this is used to create fluid damping -- density 1000 means the fluid is as dense as water -- in order to stop the box from wiggling too much. It is not critical for demonstration of tendon actuation.

LuiseMq, 2022-05-30:
@nimrod-gileadi @yuvaltassa Thank you very much! It worked!
I have another question, if I want to add friction to a hinge joint, do I just need to add friction to the two geoms connected to the joint?
Just like:
<body>
<geom name="geom1"  type="mesh" pos="0 0 0"  friction="0.51 0.51 0.2" condim="4"/>
<joint name="joint1" axis="0 1 0" type="hinge" limited="true" pos="0 0 10" stiffness="0.8" damping="0.4" frictionloss="0.51"/>?

yuvaltassa, 2022-05-30:
Joint friction is a property of a joint called [frictionloss](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-joint).

LuiseMq, 2022-05-30:
@yuvaltassa Yes, I've noticed this. So I just need to add frictionloss to the joint and that's it?

yuvaltassa, 2022-05-30:
Yes

LuiseMq, 2022-05-30:
@yuvaltassa  Does frictionloss prevent joint movement? I have noticed that the joint movement slows down, but does not affect the joint's ability to reach its designated position, as if there was no frictional loss. Why is this?

yuvaltassa, 2022-05-30:
Hi @LuiseMq,

These questions are easy to answer by loading test models in `simulate` and tweaking the parameters. Please try to make this minimal effort before asking questions. When you do, please include the model with which you are encountering a problem. Here is a model for you to play with:

```xml
<mujoco>
  <worldbody>
    <body>
      <joint axis="0 1 0"/>
      <geom type="capsule" size=".01" fromto="0 0 0 .3 0 0"/>
      <geom size=".05" pos=".3 0 0"/>
    </body>
    <body pos="0 .2 0">
      <joint axis="0 1 0" damping=".1"/>
      <geom type="capsule" size=".01" fromto="0 0 0 .3 0 0"/>
      <geom size=".05" pos=".3 0 0"/>
    </body> 
    <body pos="0 .4 0">
      <joint axis="0 1 0" frictionloss="1"/>
      <geom type="capsule" size=".01" fromto="0 0 0 .3 0 0"/>
      <geom size=".05" pos=".3 0 0"/>
    </body>        
  </worldbody>
</mujoco>
```

LuiseMq, 2022-05-30:
Thank you for the reminder. What I want to know is, why wouldn't friction stop the pendulum in a certain position? Why does the pendulum stop at the position where the forces are balanced, no matter how much friction there is?

fukangl, 2023-09-17:
Hi @yuvaltassa @nimrod-gileadi ,
If I want the tendon path is constrained to pass through one hole located in one object, I could make the tendon path be constrained to pass through the hole:
<tendon>
<spatial name="t1" limited="true" range="0 1" damping="100">
<site site="p1"/>
<geom geom="hole1" sidesite="hole1"/>
<site site="p2"/>
</spatial>
</tendon>

However, If I want the tendon path is constrained to pass through three holes located in three objects, I used 

<tendon>
<spatial name="t1" limited="true" range="0 1" damping="100">
<site site="p1"/>
<geom geom="hole1" sidesite="hole1"/>
<geom geom="hole2" sidesite="hole2"/>
<geom geom="hole3" sidesite="hole3"/>
<site site="p2"/>
</spatial>
</tendon>

But it does not work, do you know how to deal with that?

yuvaltassa, 2023-09-17:
From the documentation:

https://mujoco.readthedocs.io/en/latest/XMLreference.html?highlight=sidesite#spatial-geom-sidesite

> If the side site is inside the geom, the tendon is constrained to pass through the interior of the geom.

fukangl, 2023-09-17:
Thank you Yuval! @yuvaltassa Yeah, it works for one geom. However, when I attempt to constrain the tendon path to pass through two geoms, it doesn't seem to work. Is it possible for the tendon to pass through two geoms without specifying a site between them?"

yuvaltassa, 2023-09-17:
https://mujoco.readthedocs.io/en/latest/XMLreference.html?highlight=sidesite#tendon-spatial

> Each branch of the tendon path must start and end with a site, and if it has multiple obstacle geoms they must be separated by sites