google-deepmind/mujoco #301: python-bindings for Visualization: Capture an image of the view (equivalent to physics.render() von dm_control)

aliemil, 2022-05-26:
Hey all,
I am currently working on an RL problem for which I developed a Testing Env in MuJoCo and would like to document during the training process the actions e.g. after 10000 steps, 1M steps etc.
The dm_control library had a simple function to basically capture an image of the viewpoint at a certain state ( physics.render() ) which returns an image.

Is their any equivalent way to do it with just the python bindings?
I tried to import the Camera Class from dm_controll which contains the render function, but unfortunately some functions the bindings physics-equivalent do not overlap.

If not, any other ideas how to accomplish the capturing of parts of the training?

Cheers from Spain

[image]

nimrod-gileadi, 2022-06-29:
The new MuJoCo bindings don't have a convenience class like Camera, so you'll need to use the lower-level MuJoCo API. We'll work on a Colab notebook to demonstrate this.

In the meanwhile, you can look at the test code for python rendering, which shows how to use `mujoco.GLContext` and the rendering functions: https://github.com/deepmind/mujoco/blob/main/python/mujoco/render_test.py

aliemil, 2022-06-29:
Awesome thanks!

closed 2022-06-29 by aliemil