google-deepmind/mujoco #307: Element Nesting too Deep

tudorjnu, 2022-05-28:
**How to ask for help**

I am trying to implement a discretised form of a continuous robot which has n_bodies=100. For this purpose, I used a composite element (rope) which I further saved into an `.xml` using mujoco's function. Afterwards, I have added actuation within the last 10 links. 

When trying to load the `.xml` back, I get an error of `Error=XML_ELEMENT_DEPTH_EXCEEDED ErrorID=18 (0x12) Line number=494: Element nesting is too deep.`. 

This happens when I am just exporting and importing the model back as well. 

How should I approach this? Thank you!

yuvaltassa, 2022-05-28:
Oh wow. This looks like an issue at the tinyxml parser level. Let us investigate and get back to you...

yuvaltassa, 2022-05-28:
Ugh, yes it is. See [here](https://github.com/leethomason/tinyxml2/blob/master/tinyxml2.h#L115).

We'll see if we can fix this upstream or find some other workaround, but this is not something we can easily fix in our own code. I'll keep this open and report back when we have more info.

In the meantime, maybe you can reduce the number of bodies in the chain?

tudorjnu, 2022-05-29:
Thank you very much @yuvaltassa! 

That helped a lot! I managed to get away with 95 elements.

I was thinking maybe it will be a good feature to be able to attach sections of composite elements. For example one could make a section of 10 elements followed by another and so on... 

This will allow defining different properties (i.e. stiffness) to different blocks, thus making the block act like a stiffer or more flexible block. This property might help in the modeling of continuum robots.

quagla, 2023-01-16:
The maximum depth in tinyxml has now been increased to 500.

closed 2023-01-16 by quagla