google-deepmind/mujoco #310: Help using mujoco with CMake's FetchContent module

lucas-inacio, 2022-05-31:
Hello, everyone!

I'm trying to use mujoco as an external project using CMake's FetchContent feature. Below are the CMakeLists.txt scripts being used:

```cmake
# Main project CMakeLists.txt
cmake_minimum_required(VERSION 3.19)

project(Robot VERSION 0.0.1)

include(FetchContent)

FetchContent_Declare(
    mujoco
    GIT_REPOSITORY https://github.com/deepmind/mujoco.git
    GIT_TAG 2.2.0
)

set(MUJOCO_BUILD_EXAMPLES OFF)
set(MUJOCO_BUILD_TESTS OFF)
set(MUJOCO_TEST_PYTHON_UTIL OFF)

FetchContent_MakeAvailable(mujoco)

add_subdirectory(source)
```

```cmake
# source/CMakeLists.txt
add_executable(main main.cpp)
target_link_libraries(main mujoco)
```

The problem is when I build the project CMake complains about missing headers form mujoco/include directory. I was able to fix the problem by changing the `target_include_directories` at line 67 (mujoco's root CMakeLists.txt) from this:

```cmake
target_include_directories(
  mujoco
  PUBLIC $<BUILD_INTERFACE:${CMAKE_SOURCE_DIR}/include>
         $<INSTALL_INTERFACE:${CMAKE_INSTALL_INCLUDEDIR}/>
  PRIVATE src
)
```
To this:
```cmake
target_include_directories(
  mujoco
  PUBLIC $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>
         $<INSTALL_INTERFACE:${CMAKE_INSTALL_INCLUDEDIR}/>
  PRIVATE src
)
```
CMAKE_CURRENT_SOURCE_DIR makes reference to the correct directory when building the project.
So before opening any issue I wanted to ask if there's a better way to solve this. Is the inclusion of mujoco with FetchContent discouraged? Or is a pull request welcome in this case?

Notes: 
* I'm building with Visual Studio 17 2022 x64 Native Tools Command Prompt
* Using CMake 3.22.22040401-MSVC_2.

francesco-romano, 2022-05-31:
Hi @lucas-inacio 

Yes, your modification is definitely correct.
Please, do open a PR :)

lucas-inacio, 2022-05-31:
Hi @francesco-romano 

Thanks for the reply. I've created a new PR.

lucas-inacio, 2022-06-10:
I'm closing this issue since PR #311 solves the problem. Thank you everyone!

closed 2022-06-10 by lucas-inacio