google-deepmind/mujoco #312: [Feature Request] Java Bindings

madhephaestus, 2022-05-31:
As a Java developer, I would like to have Java binding for MuJoCo. 

To that end i have built the beginnings of Java binding using JavaCPP and the released binaries of MuJoCo. 

https://github.com/CommonWealthRobotics/mujoco-java

Building of the Java bindings is done entirely in Github Worflows. Releases are published to github release, as well as github packages for Maven integration. 

I would love to integrate this work into your build stack. It would be awesome to have each release automatically build and release the bundled jar file for all supported platforms.  

Is this feature something that y'all would like to integrate, or should I keep it as a separate project?

yuvaltassa, 2022-05-31:
Hi @madhephaestus !

Thanks so much for this. We would be delighted to feature your bindings in the README and let Java developers know that they have a home!

However, I doubt that integration into our build stack is a good idea. We have big plans for the MuJoCo API and while I can imagine small changes that would automatically get incorporated into a well-written bindings wrapper, it is just as easy to imagine changes that would break the wrapper and require some attention from the maintainer. Having written two wrappers ourselves ([Python](https://mujoco.readthedocs.io/en/latest/python.html), [Unity](https://mujoco.readthedocs.io/en/latest/unity.html)), we know all too well how easy it is to break them. If we or a contributor made a change that would break your build, we'd either be blocked from submitting or we'd have a red build-fail chip until the issue is fixed. Either way not a good look :)

Does that make sense?

madhephaestus, 2022-06-01:
I totally get it. 

I would very much appreciate being included in the readme (with username attribution as the maintainer of the bindings). I am setting up Workflow publishing to Maven Central via Sonatype, so the bindings will be availible as a regular Maven dependency. Once that is all configured i will message on this issue to pass along the usage instructions.

closed 2022-06-22 by copybara-service[bot]

yuvaltassa, 2022-06-22:
[Done](https://github.com/deepmind/mujoco#third-party-bindings) 🙂

madhephaestus, 2022-06-24:
Thank you so much! I look forward to working with MuJoCo in Java!

madhephaestus, 2022-10-11:
I would be delighted with inclusion into the README and will endeavor to
keep the bindings up to date, lagging the official C releases.

commit 2ea01bf2 (2022-06-22): Add Related Software section to README.md.

BEGIN PUBLIC

Add Related Software section to README.md.

Fixes #312.

END_PUBLIC

PiperOrigin-RevId: 456528399
Change-Id: Icf97a31a5ea16e93edf9856f70dec650c7b70791