google-deepmind/mujoco #314: Extend the composite class to support meshes from obj files

BOBaraki, 2022-05-31:
To make the simulator useful for deformable object manipulation, I believe it would be essential one to be able to introduce easily complex topology items. The problem is that the composite objects are not supported by meshes like the rigid classes. 

It would be useful if one is able to generate a deformable object by introducing a mesh from an OBJ file.

yuvaltassa, 2022-06-04:
This is a great feature request and we are indeed working on something along these lines. However note that OBJ files would not be appropriate for specifying such objects, since what you'd want is a volumetric (tetrahedral) mesh rather than a surface (triangular) mesh.

BOBaraki, 2022-06-05:
I made the request with obj files in my mind since most of the cloth/garment datasets consider the object as a 2-manifold and thus represent them with a mesh. An example is the [CLOTH3D dataset](https://chalearnlap.cvc.uab.es/dataset/38/description/). However, I agree that a volumetric mesh would be more appropriate to generate more complex shapes and dynamics.

yuvaltassa, 2022-06-05:
Of course, yes. I see now what you meant.

I'm going to close this issue now and re-open a new issue describing our plans for deformable mesh structures when they are finalised, hopefully over the summer. I will ping here when this happens.

Cheers.

closed 2022-06-05 by yuvaltassa

dblanm, 2022-09-13:
I wonder if there are any updates on this?