google-deepmind/mujoco #315: Failed to create Mujoco runtime Unity

ClaySant, 2022-06-01:
This message appear when I try to run the simulation in unity

yuvaltassa, 2022-06-02:
Please provide more details as described [here](https://github.com/deepmind/mujoco/blob/main/CONTRIBUTING.md#reporting-bugs)

ClaySant, 2022-06-04:
I'm using Mujoco 2.2.0, Unity 2021 and Linux Ubunto 20.04.3

erez-tom, 2022-06-04:
Did you install the plug-in as a package, or locally (i.e., have the files under your project's Assets folder)?  Did the installation process succeeded in copying the binary, or did you perform that process manually as described [here](https://mujoco.readthedocs.io/en/latest/unity.html#linux)?

ClaySant, 2022-06-04:
I installed the plug-in as a package, and then I copy the libmujoco.so to the Unity/Hub/Editor/2021.3.3f1/Editor (sorry, this is my first time using unity, so i may have made some mistakes)

erez-tom, 2022-06-04:
As the instructions suggest, you should copy the binary "into any location under your project’s Assets directory."

closed 2022-06-04 by erez-tom

tiago2820, 2023-04-24:
I haver the same issue, what did you do to resolve this @ClaySant ?

erez-tom, 2023-04-24:
This error message appears whenever the model failed to load.  The meaningful error is the one before it, which should explain why a model was not loaded.  If you're still stuck after fixing that error, you're welcome to start a new issue.  Good luck!

tiago2820, 2023-04-24:
the error before is: IOException: Error loading the model: XML Error: unsigned integer with an optional suffix {K,M,G,T,P,E} is expected in attribute 'memory' (or the size specified is too big)
Element 'size', line 1. You know how to solve it?

erez-tom, 2023-04-24:
I was trying to look into this error at #807, so let's move our conversation to that issue.  Can you answer the questions I asked there?

Wafeechan, 2023-09-24:
Hello, I had the same problem when trying out MuJoCo. In this case, I was trying to do the ball on the tilted plane stuff. I am currently testing it out on unity 2021.3.11f1 personal on Windows 10, using MuJoCo 2.3.7.

I have cloned a repository of the plugin from here: [https://github.com/google-deepmind/mujoco/tree/main](https://github.com/google-deepmind/mujoco/tree/main), and manually imported the package from disk. Do I need the package to be on the same disk partitions? (As in I downloaded the package on D: and imported it from there, while my user directory was on C:).

Secondly, it is probably because I didn't understand some terms, but I did "Expand the zip archive to a directory called MuJoCo in your user directory, and copy the file MuJoCo\bin\mujoco.dll." By this does it mean that I should make a new user directory called "MuJoCo", or should it be automatically created like when I import the package? Because I manually created an empty MuJoCo folder directory for it.

Thirdly, what does it actually mean by "The library can also be copied into any location under your project’s Assets directory."? Does it mean that I have to manually copy paste the mujoco.dll file to the project's Assets directory? Or was there actually another way to do this?

Also lastly, the line "On Unity version 2020.2 and later, the Package Manager will look for the native library file and copy it to the package directory when the package is imported.". I was actually assuming this do be done, but after importing the plugin, a warning prompt appeared telling me there's no bin directory on the user directory, which is why I did it manually as implied on the second and third question.

Sorry for the lengthy question, but I hope my question can be solved so I can finally start exploring MuJoCo for my final year project. It may be something dumb on my part, but any kind of help is appreciated. Thank you.