google-deepmind/mujoco #319: Windows Binary Build artifacts failing to execute (double click, powershell, and Cygwin execution)

madhephaestus, 2022-06-02:
1. Download the latest Windows x86_64 binary release
2. extract the zip on Windows 10 64 bit
3. double click simulate.exe or testspeed.exe
4. Get a series of 4 popups saying DLL's are missing

[image]

This issue is also effecting linking against the binary release of the library for applications.

madhephaestus, 2022-06-02:
For reference, the binary release on Linux produces the following result when run:
```
Running 10000 steps at dt = 0.005 ...

 Simulation time      : 0.23 s
 Steps per second     : 44310
 Realtime factor      : 221.55 x
 Time per step        : 0.0226 ms

 Contacts per step    : 11.44
 Constraints per step : 41.05
 Degrees of freedom   : 27

```

yuvaltassa, 2022-06-04:
Thanks @madhephaestus,
We'll look into this soon, leaving open. We all took some time off post open-sourcing, our response times will be back to normal in a week or two :)

madhephaestus, 2022-06-04:
Great! Thank you all for the hard work, we all very much appreciate this library becoming open source!

madhephaestus, 2022-06-06:
this issue is replicated in all published versions on GitHub, 

Except 2.1.0, this version works as expected.

All other versions exhibit this bug.

saran-t, 2022-06-14:
I was under the impression that MSVC redistributable comes bundled with Windows installations, but evidently this isn't always the case.

Please try downloading and installing VC++ redist from https://www.microsoft.com/en-gb/download/details.aspx?id=48145

madhephaestus, 2022-06-14:
that did indeed fix the issue. Why did MuJoCo work in 2.1.0 without this system depenancy?

saran-t, 2022-06-16:
MuJoCo 2.1.0 was statically linked against MSVCRT. We stopped doing this since we don't want to ship bits of binary that we do not build ourselves.

closed 2022-06-16 by saran-t

madhephaestus, 2022-06-16:
Is there a document of all the linked libraries that MuJoCo depends on for users that intend to make a statically linked application?

saran-t, 2022-06-16:
Do you mean for people who are building everything from source?

saran-t, 2022-06-16:
If you're using our official builds then it should really just be the Microsoft C runtime on Windows that's "missing".

I can check again to see if we can statically link in the Microsoft CRT again if that would help.

madhephaestus, 2022-06-16:
That would be a big help! It would be a nice comprimise if there were one archive statically linked, and a second archive of a thin release to satisfy those that want to avoid linked in binary blobs. This issue was all about the WIndows build, but the same would be true for all 3 platforms WRT the Java bindings.

saran-t, 2022-06-16:
See also https://docs.microsoft.com/en-us/cpp/c-runtime-library/crt-library-features

TL;DR it isn't really standard practice to link Microsoft's CRT statically into DLLs. What you might want to do is to ship the MS redistributable DLLs in your jar, see https://docs.microsoft.com/en-us/cpp/windows/redistributing-visual-cpp-files for Microsoft's guidance on this.

madhephaestus, 2022-06-17:
yeah, i can take a whack at that. If its just the CRT, then i can just bundle and deploy that at runtime. That said a statically linked lib would be kinda great :)

madhephaestus, 2022-06-18:
can you tell me which DLL's are needed for redistribution? https://docs.microsoft.com/en-us/cpp/windows/determining-which-dlls-to-redistribute?view=msvc-170