google-deepmind/mujoco #325: How to use the Unity Plugin in a URDF imported in Unity?

DinoDany, 2022-06-06:
Hello. I'm trying to use MuJoCo's plugin in a URDF that I imported into Unity. How can I map MujoCo and PhysX? Do I need to define the components for both? 

Thank you :)

Balint-H, 2022-06-29:
I'm working on a script that converts ArticulationBody chains to Mujoco ones and vice versa. Its still at around 70% ready, because of the differences in the way joint properties are defined in MjJoints and Articulation bodies. Have you found your own solution to this problem? I think an editor script to convert between the two sets of components could be a handy addition to the plugin!

erez-tom, 2022-07-18:
I just added a [file](https://github.com/deepmind/mujoco/blob/main/unity/Runtime/Tools/ConvertMj2Rca.cs) that converts a MuJoCo model (represented as a hierarchy of GameObjects with the relevant MuJoCo components) to a PhysX ArticulationBody hierarchy (sorry it took so long).  I hope it serve as a reference, even as it's missing some features.  I'll close the issue for now, but feel free to re-open or start new conversations on mapping between the two engines.

closed 2022-07-18 by erez-tom