google-deepmind/mujoco #326: Part of the model is lost in the mujoco

kankanzheli, 2022-06-07:
I used to use robots mainly in the gazebo environment. Use URDF files. All the model files used are in STL format. When using mujoco to directly open the URDF file, a part of the model is missing. Use/ The XML file converted from URDF by compile also shows that the box part in URDF is missing. How should this be solved?
[urdf.txt](https://github.com/deepmind/mujoco/files/8853199/urdf.txt)
[xml.txt](https://github.com/deepmind/mujoco/files/8853201/xml.txt)

Here, the "bone_mesh.stl" file is relatively large. 17mb

closed 2022-06-07 by yuvaltassa

yuvaltassa, 2022-06-07:
Just because it was used in other simulators doesn't mean it's not an ASCII file. Open it in a text editor to find out.

If it's a binary STL then it probably has more than 200000 faces. You can reduce that in e.g. MeshLab (and also convert ASCII-> binary if that's your issue.)

henbucuoshanghai, 2025-04-03:
when I covert urdf mdoel to xml ?

zhudisha, 2025-07-22:
> sanme error，why？and how to solve it?tks

面数太多导致mujoco识别错误。如果是自己的模型，那就导出时改精度。如果是别人的模型那就得用其他工具处理一下了