google-deepmind/mujoco #328: Question about texture

Lucky-hqj, 2022-06-07:
Recently, I tried to use [ Google Scanned Objects ] (https://app.gazebosim.org/GoogleResearch/fuel/collections/Google%20Scanned%20Objects) in mujoco, but I find the texture of the object like this:
[image]
I didn't find a proper way to use the texture like this.
Can you give me some advice？

Lucky-hqj, 2022-06-08:
Thank you very much! This is very inspiring to me. I'm not very familiar with mujoco yet. About "note that you are better off using simpler approximate collision geoms than the mesh itself ", Do you mean like this in the mug example:
```
  <default>
    <default class="collision">
      <geom type="box" group="3"/>
      <default class="cup">
        <geom size="0.0026 0.00704 0.033"/>
      </default>
      <default class="handle">
        <geom size="0.0025 0.00328 0.0055"/>
      </default>
    </default>
  </default>
```

kevinzakka, 2022-06-08:
@Lucky-hqj Correct, you may want to do similar approximations for the clock.

Lucky-hqj, 2022-06-09:
@kevinzakka Thank you very much!  I hope I can do some interesting thing about robot learning on mujoco！