google-deepmind/mujoco #335: How to use mjvMakeConnector to create a visual geom for circular tube before adding it to mjvScene.geoms?

dhruvthanki, 2022-06-10:
I am trying to show the circular ring trajectory that my end-effector is trying to track. from #88 I have to create the geom using mjvMakeConnector and then add it to the mjvScene.geoms. But the documentation for the mjvMakeConnector isn't clear to me. Could you explain or point to some documentation about how to do this?

yuvaltassa, 2022-06-12:
Please search our issues for e.g. `mjvScene` or `mjvMakeConnector` and you should find some examples. I think #194 might have some examples. If you try and fail please post some code here and we will help you.

closed 2022-06-12 by yuvaltassa

yuvaltassa, 2022-07-22:
There is a now a section in [this colab](https://github.com/deepmind/dm_control/blob/main/dm_control/mujoco/tutorial.ipynb) called "Adding arbitrary 3D geometry".

lgkimjy, 2023-11-02:
@yuvaltassa Currently, I am implementing the controller by modifying simulate/main.cc. I still am not sure how to use mjvMakeConnector in CPP. Can you provide any example sources just like python? Or can you provide any guides to use the given function?

yuvaltassa, 2023-11-02:
[mjv_connector](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mjv-connector) has a nicer interface. Just look in [engine_vis_visualize.c](https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_vis_visualize.c).