google-deepmind/mujoco #338: Auto Inertia Computation of concave shapes, meshes and thin bodies

rohit-kumar-j, 2022-06-10:
### Discussed in https://github.com/deepmind/mujoco/discussions/318

<div type='discussions-op-text'>

<sup>Originally posted by **rohit-kumar-j** June  2, 2022</sup>
By default, does mujoco support auto inertia computation from an STL file loaded as a body? Like a [hollow vase](https://cults3d.com/en/3d-model/home/vase-006)? (assuming uniform density)
</div>

yuvaltassa, 2022-06-11:
The answer is: 
1. Currently, no. Only convex STLs will get the correct inertia.
2. We are working on better computation that will work for all well formed meshes, convex or concave.
Leaving open, will update here when this is implemented.

closed 2022-07-31 by copybara-service[bot]

yuvaltassa, 2022-08-01:
Note this is now implemented, but not as a default option in order to not break existing models.

There is also an option to put all the inertia on the surface of the mesh.

arpit15, 2023-12-11:
@yuvaltassa can you kindly mention here where to get started for options to create models with concave meshes?

arpit15, 2023-12-11:
I found ` shellinertia="true"` as the parameter to set inertia at surfaces. However, its unclear how to properly set up concave meshes.

yuvaltassa, 2023-12-12:
There are two options for colliding with concave meshes:

1. Use a convex decomposition package like https://github.com/SarahWeiii/CoACD
2. Use Signed Distance Fields.

commit 5c5449bf (2022-07-31): Compute moment of inertia for concave and boundary meshes. Resolve #338.

- Mesh inertias can now be computed exactly for well-formed (no holes) non-convex meshes.
- To activate this feature, set `<compiler exactmeshinertia="true">` (defaults to `false`). This default may change in the future.
- Added `<geom shellinertia="true/false">` (defaults to `false`). When true, geom inertia is computed assuming all the mass is concentrated on the surface, and `density` is interpreted as surface density (mass/area). Currently only mesh geoms are supported.

PiperOrigin-RevId: 464368395
Change-Id: I17afd [... cut]