google-deepmind/mujoco #339: Model problem in Mujoco - Bbulges exist on each joint

JackTony123, 2022-06-10:
Hello everyone,

Recently I build up a model on the newest Mujoco 2.2.0 for WAM robot arm and hand The arm looks good but the hand there exists something weird for that
[image]
I don't know why there exist bulges on each joint. Looks so weird. Did someone meet the same problem like that? How to deal with that? Thanks!

kevinzakka, 2022-06-10:
Can you attach a zip file?

JackTony123, 2022-06-10:
[wam7_Pro.zip](https://github.com/deepmind/mujoco/files/8881741/wam7_Pro.zip)

kevinzakka, 2022-06-10:
These are related to the ` <geom mesh="bhand_finger_med_link_convex"/>` elements in the XML file.

JackTony123, 2022-06-10:
> 

Thanks for your reply Kevin! That helps me so much! Not only for this element, but also for the other <geom mesh> parts, Now the model is normal. Thanks again! 
[image]

closed 2022-06-11 by yuvaltassa