google-deepmind/mujoco #346: Can not open STL file for custom environment int Colab

aliemil, 2022-06-14:
Hey all,
I am trying to run my custom mujoco environment int google colab, but if I try to create the model from the xml it return that it cannot open the STL's (meshes) included.

To be said, on the local jupyter notebook/lab it runs without a problem.
Has anyone an idea what might cause the issue? 

[image]

kevinzakka, 2022-06-16:
Hey @aliemil, can you share the Colab? I can debug better that way.

aliemil, 2022-06-16:
@kevinzakka The error seems to have been pretty silly. It started to run just by turning on the GPU on runtime. 
But thanks for the effort! :)

closed 2022-06-16 by aliemil