google-deepmind/mujoco #349: “pip install mujoco” get stuck

zhihaocheng, 2022-06-17:
Hi,

I'm a student and I'm trying to use MuJoCo for my research. And I'm looking for some help with the installation of mujoco.

There are mujocos for C/CXX and for python. Hence, to distinguish, I denote them as Mujoco(C/CXX) and Mujoco(Python), respectively. Mujoco(C/CXX) can be installed by building from source or just download a precomplied version. Mujoco(Python) can be installed using pip or building from source.

After installing Mujoco(C/CXX) (The server maintainer installed Mujoco2.2.0 from source and provided a module that I can load), I tried to run “pip install mujoco” to install Mujoco(Python). There were no bugs reported but just got stuck. I waited for about two hours but the installation process was still not finished. Could you please help me? I have also tried to build Mujoco(Python) from source, but the same problem occurred.

"""
(mujocotest) [zhc]$ pip install mujoco
Collecting mujoco
  Using cached mujoco-2.2.0.tar.gz (138 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: absl-py in /home/zch/.local/lib/python3.7/site-packages (from mujoco) (0.15.0)
Requirement already satisfied: glfw in /home/zch/.local/lib/python3.7/site-packages (from mujoco) (2.5.3)
Requirement already satisfied: numpy in /home/zch/.local/lib/python3.7/site-packages (from mujoco) (1.19.5)
Requirement already satisfied: pyopengl in /home/zch/.local/lib/python3.7/site-packages (from mujoco) (3.1.6)
Requirement already satisfied: six in /usr/local/python/3.7.7/lib/python3.7/site-packages (from absl-py->mujoco) (1.15.0)
Building wheels for collected packages: mujoco
  Building wheel for mujoco (setup.py) ... /

"""

kevinzakka, 2022-06-17:
Hi @zhihaocheng, a copy of the MuJoCo binaries is provided as part of the pip package so you don't have to have the "C/CXX" binaries downloaded/installed separately.

Can you try creating a fresh python virtual environment (I recommend [miniconda](https://docs.conda.io/en/latest/miniconda.html)) and then installing MuJoCo? And then let me know if there are any errors.

saran-t, 2022-06-17:
I'm a bit surprised that `pip install mujoco` is triggering a source installation for you. Is your server configured to not connect to PyPI but use a local repository?

My recommendation if you're not intending to modify MuJoCo is to get a bdist wheel from PyPI, either via pip or by manually downloading it from pypi.org.

zhihaocheng, 2022-06-18:
Thanks @kevinzakka. I have followed your advice but the same problem occurred. The detailed installation process and information about the system is attached as follows.

First, create environment and then activate. I did not use miniconda or anaconda because the server does not support them well, and I must load provided python versions.

-> virtualenv --system-site-packages ~/envs/testmujoco
-> source ~/envs/testmujoco/bin/activate

Load necessary modules and upgrade pip.
-> module load python/3.7.7
-> module load cmake/3.21.1 (without this module, an error occurred. )
-> python -m pip install --upgrade pip

Export MUJOCO_PATH
-> export MUJOCO_PATH=~/mujocosource/mujoco-2.2.0 (this is the path to the mujoco . Without this path, an error occurred.) 

After that I can run pip install mujoco, but it got stuck. The server is provided by the university. So I have no access to root and must use some provideded modules such as python and tensorflow. Maybe the system of the server is to old such that mujoco does not support?

---------------------------------------------------------------------------------------------
Error without cmake.
(testmujoco) [zch]$ pip install --upgrade mujoco
Collecting mujoco
  Using cached mujoco-2.2.0.tar.gz (138 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: absl-py in ./.local/lib/python3.7/site-packages (from mujoco) (0.15.0)
Requirement already satisfied: glfw in ./.local/lib/python3.7/site-packages (from mujoco) (2.5.3)
Requirement already satisfied: numpy in ./.local/lib/python3.7/site-packages (from mujoco) (1.19.5)
Requirement already satisfied: pyopengl in ./.local/lib/python3.7/site-packages (from mujoco) (3.1.6)
Requirement already satisfied: six in /usr/local/python/3.7.7/lib/python3.7/site-packages (from absl-py->mujoco) (1.15.0)
Building wheels for collected packages: mujoco
  Building wheel for mujoco (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [34 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-37
      creating build/lib.linux-x86_64-cpython-37/mujoco
      copying mujoco/render_test.py -> build/lib.linux-x86_64-cpython-37/mujoco
      copying mujoco/rollout.py -> build/lib.linux-x86_64-cpython-37/mujoco
      copying mujoco/rollout_test.py -> build/lib.linux-x86_64-cpython-37/mujoco
      copying mujoco/bindings_test.py -> build/lib.linux-x86_64-cpython-37/mujoco
      copying mujoco/__init__.py -> build/lib.linux-x86_64-cpython-37/mujoco
      creating build/lib.linux-x86_64-cpython-37/mujoco/glfw
      copying mujoco/glfw/__init__.py -> build/lib.linux-x86_64-cpython-37/mujoco/glfw
      creating build/lib.linux-x86_64-cpython-37/mujoco/osmesa
      copying mujoco/osmesa/__init__.py -> build/lib.linux-x86_64-cpython-37/mujoco/osmesa
      creating build/lib.linux-x86_64-cpython-37/mujoco/egl
      copying mujoco/egl/egl_ext.py -> build/lib.linux-x86_64-cpython-37/mujoco/egl
      copying mujoco/egl/__init__.py -> build/lib.linux-x86_64-cpython-37/mujoco/egl
      running build_ext
      Configuring CMake with the following arguments:
          -DPython3_ROOT_DIR:PATH=~/envs/testmujoco
          -DPython3_EXECUTABLE:STRING=~/envs/testmujoco/bin/python
          -DCMAKE_MODULE_PATH:PATH=/tmp/pbs.6363806.pbsserver/pip-install-jv6zno69/mujoco_7813dcb2c5754bdf98268cdf968f35b7/cmake
          -DCMAKE_BUILD_TYPE:STRING=Release
          -DCMAKE_LIBRARY_OUTPUT_DIRECTORY:PATH=build/temp.linux-x86_64-cpython-37
          -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=:BOOLON
          -DCMAKE_Fortran_COMPILER:STRING=
          -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON
          -DBUILD_TESTING:BOOL=OFF
          -DMUJOCO_LIBRARY_DIR:PATH=~/mujocosource/mujoco-2.2.0/lib
          -DMUJOCO_INCLUDE_DIR:PATH=~/mujocosource/mujoco-2.2.0/include
          -DPython3_LIBRARY=/usr/local/python/3.7.7/lib/python3.7
          -DPython3_INCLUDE_DIR=/usr/local/python/3.7.7/include/python3.7m
      error: [Errno 2] No such file or directory: 'cmake': 'cmake'
      [end of output]
[... 72 lines cut ...]
          super().run_command(command)
        File "~/envs/testmujoco/lib/python3.7/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "/tmp/pbs.6363806.pbsserver/pip-install-9s9wfcn2/mujoco_673a74011cbe456d8d97c294fadfc071/setup.py", line 143, in run
          self._mujoco_framework_path) = self._find_mujoco()
        File "/tmp/pbs.6363806.pbsserver/pip-install-9s9wfcn2/mujoco_673a74011cbe456d8d97c294fadfc071/setup.py", line 154, in _find_mujoco
          raise RuntimeError(f'{MUJOCO_PATH} environment variable is not set')
      RuntimeError: MUJOCO_PATH environment variable is not set
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mujoco
  Running setup.py clean for mujoco
Failed to build mujoco

Server System Information
-> uname -a
Linux hpc055 2.6.32-696.16.1.el6.x86_64 #1 SMP Wed Nov 15 16:51:15 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

-> lsb_release -a
LSB Version:    :base-4.0-amd64:base-4.0-noarch:core-4.0-amd64:core-4.0-noarch:graphics-4.0-amd64:graphics-4.0-noarch:printing-4.0-amd64:printing-4.0-noarch
Distributor ID: CentOS
Description:    CentOS release 6.9 (Final)
Release:        6.9
Codename:       Final

zhihaocheng, 2022-06-18:
Thanks @saran-t. I am not clear whether the server uses a local repository. The server is provided by the university. So I have no access to root and must use some provideded modules such as python and tensorflow. But I guess mujoco could connect to Pypi because I am the first to configure mujoco on the server of the university.  

According to your advice, I downloaded Mujoco from Pypi and tried to install it. However, it failed again. Could you please refer to the above response to kevinzakka for more details on the environment and server. 

(testmujoco) [zhc]$ wget https://files.pythonhosted.org/packages/81/9b/c1f7d49c78ece63412c7f54fa0b89e5a81702e28cd46ea787f9bbd49b489/mujoco-2.2.0-cp37-cp37m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl
--2022-06-18 12:19:00--  https://files.pythonhosted.org/packages/81/9b/c1f7d49c78ece63412c7f54fa0b89e5a81702e28cd46ea787f9bbd49b489/mujoco-2.2.0-cp37-cp37m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl
Resolving files.pythonhosted.org... 151.101.29.63, 2a04:4e42:7::319
Connecting to files.pythonhosted.org|151.101.29.63|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3526854 (3.4M) [application/octet-stream]
Saving to: “mujoco-2.2.0-cp37-cp37m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl”

100%[===============================================================================================================================================>] 3,526,854   --.-K/s   in 0.08s

2022-06-18 12:19:00 (40.7 MB/s) - “mujoco-2.2.0-cp37-cp37m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl” saved [3526854/3526854]

(testmujoco) [zhc]$ pip install mujoco-2.2.0-cp37-cp37m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl
ERROR: mujoco-2.2.0-cp37-cp37m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl is not a supported wheel on this platform.

saran-t, 2022-06-19:
The problem is that the machine is running CentOS 6, which reached its end-of-life [since November 2020](https://cloud.google.com/compute/docs/eol/centos6).

Our Linux wheels are built to target [manylinux2014](https://peps.python.org/pep-0599/) compatibility requirements, which should be [good for CentOS 7 or later](https://github.com/pypa/manylinux).

We used to target manylinux2010 briefly, but since that's no longer recommended upstream due to CentOS 6 being EoL we migrated to manylinux2014. I realise this might not help you, but my recommendation would be to see if your sysadmin is willing to upgrade the server to CentOS 7 or later, since otherwise the machine wouldn't get any further upstream upgrade.

Failing that, you'll unfortunately have to build from source. To troubleshoot `pip` being stuck, please rerun it with `-vvv` so that we get some more message.

zhihaocheng, 2022-06-21:
Thanks @saran-t. The server is used for all university students, and there are no plans to update the system. I run pip with -vvv, and the log information is attached as follows.

(testmujoco) [zch]$ pip install -vvv mujoco
Using pip 22.1.2 from /project/RDS-FEI-ImitationLearn-RW/.local/lib/python3.7/site-packages/pip (python 3.7)
Defaulting to user installation because normal site-packages is not writeable
Created temporary directory: /tmp/pbs.6369584.pbsserver/pip-ephem-wheel-cache-ly7f56un
Created temporary directory: /tmp/pbs.6369584.pbsserver/pip-build-tracker-3h6o0oe0
Initialized build tracking at /tmp/pbs.6369584.pbsserver/pip-build-tracker-3h6o0oe0
Created build tracker: /tmp/pbs.6369584.pbsserver/pip-build-tracker-3h6o0oe0
Entered build tracker: /tmp/pbs.6369584.pbsserver/pip-build-tracker-3h6o0oe0
Created temporary directory: /tmp/pbs.6369584.pbsserver/pip-install-cu67c6ck
1 location(s) to search for versions of mujoco:
* https://pypi.org/simple/mujoco/
Fetching project page and analyzing links: https://pypi.org/simple/mujoco/
Getting page https://pypi.org/simple/mujoco/
Found index url https://pypi.org/simple
Looking up "https://pypi.org/simple/mujoco/" in the cache
Request header has "max_age" as 0, cache bypassed
Starting new HTTPS connection (1): pypi.org:443
https://pypi.org:443 "GET /simple/mujoco/ HTTP/1.1" 304 0
  Skipping link: none of the wheel's tags (cp310-cp310-macosx_10_12_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/78/45/f96ad37f30c3da1d4c339ed884d82d95502dc7d2ae1e5c65309d921f0012/mujoco-2.1.2-cp310-cp310-macosx_10_12_x86_64.whl#sha256=af0a6131438825c8b1689559b8dc4a60de9791949164496c1c08dafe4ad4948e (from https://pypi.org/simple/mujoco/) (requires-python:>=3.7)
  Skipping link: none of the wheel's tags (cp310-cp310-macosx_11_0_arm64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/ff/4a/fc284ad412a593881e46f5f3723bcc3dd3209213fb56367b05ed9f94b2ec/mujoco-2.1.2-cp310-cp310-macosx_11_0_arm64.whl#sha256=37b4a1cfa987bd1a20b5fe823271e01f02c286629d477c7063e0f53c5e7bbb88 (from https://pypi.org/simple/mujoco/) (requires-python:>=3.7)
  Skipping link: none of the wheel's tags (cp310-cp310-manylinux2014_aarch64, cp310-cp310-manylinux_2_17_aarch64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/4a/96/745d6578fa266f2c5158fa245ef4f6d05edd5418948c6dcf18cb92a91ff9/mujoco-2.1.2-cp310-cp310-manylinux_2_17_aarch64.manylinux2014_aarch64.whl#sha256=92599d7860e780753c2764cb242f768452a421f19993d2068c334c1b70779761 (from https://pypi.org/simple/mujoco/) (requires-python:>=3.7)
  Skipping link: none of the wheel's tags (cp310-cp310-manylinux2014_x86_64, cp310-cp310-manylinux_2_17_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/df/dc/c92802c6d399b8180e93af51d9a5360bad25bce8b7bd2b13f7faca99cf59/mujoco-2.1.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl#sha256=a86c67e181c186cdc78045ccc6ab196f789db04789673226d156188ddb468aca (from https://pypi.org/simple/mujoco/) (requires-python:>=3.7)
  Skipping link: none of the wheel's tags (cp310-cp310-win_amd64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/07/ec/a6cdd345ea2b14e942c59d32c467b3a17c42f365e7a5debfbc98e96a0549/mujoco-2.1.2-cp310-cp310-win_amd64.whl#sha256=7e3c48a14ad3c95ec27e9c63e41b2e9cfbfc0d201423743c7a225df274ae0dec (from https://pypi.org/simple/mujoco/) (requires-python:>=3.7)
  Skipping link: none of the wheel's tags (cp37-cp37m-macosx_10_12_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/bb/90/55b45786d20abb7b55689f6e63437efd2fdff05e5991665f93a6ad18f1c7/mujoco-2.1.2-cp37-cp37m-macosx_10_12_x86_64.whl#sha256=917b3293880b1bcb5b119815c75dca4f472ced0c91c8f1c828fee231ba52ba14 (from https://pypi.org/simple/mujoco/) (requires-python:>=3.7)
  Skipping link: none of the wheel's tags (cp37-cp37m-manylinux2014_aarch64, cp37-cp37m-manylinux_2_17_aarch64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/be/10/47f2b09bc07578f5e7bf0609617ea9e684041bb36e966302160a02f1e626/mujoco-2.1.2-cp37-cp37m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl#sha256=a98479ac8cc49e39a78fe85417b82df8db29f1f0fdccb933c64562070233a2e3 (from https://pypi.org/simple/mujoco/) (requires-python:>=3.7)
  Skipping link: none of the wheel's tags (cp37-cp37m-manylinux2014_x86_64, cp37-cp37m-manylinux_2_17_x86_64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/be/0f/d486b999698412ea2400df65b03f1ef05c9f74c55af3cd6e285a01582e54/mujoco-2.1.2-cp37-cp37m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl#sha256=845f12c3dd0be0487ac3398dc9ac092f8c086b9627f6426034477b6e7a1e76ca (from https://pypi.org/simple/mujoco/) (requires-python:>=3.7)
  Skipping link: none of the wheel's tags (cp37-cp37m-win_amd64) are compatible (run pip debug --verbose to show compatible tags): https://files.pythonhosted.org/packages/ed/56/afd8fdf826c8ba80d46e6be0c29ca3ce65e37cb6eb6d2f4c6475c2bd99cb/mujoco-2.1.2-cp37-cp37m-win_amd64.whl#sha256=8cbf4f8ac650a1d80becfd6f2d97755478db3380202b02c7e7a183c215195631 (from https://pypi.org/simple/mujoco/) (requires-python:>=3.7)
[... cut ...]
  -- Looking for pthread_create in pthreads
  -- Looking for pthread_create in pthreads - not found
  -- Looking for pthread_create in pthread
  -- Looking for pthread_create in pthread - found
  -- Found Threads: TRUE
  -- mujoco::FindOrFetch: Using FetchContent to retrieve `abseil-cpp` - Done
  -- mujoco::FindOrFetch: checking for targets in package `Eigen3`
  -- mujoco::FindOrFetch: checking for targets in package `Eigen3` - target `Eigen3::Eigen` not defined.
  -- mujoco::FindOrFetch: Using FetchContent to retrieve `eigen`

saran-t, 2022-06-21:
The last line that you see when it gets stuck is "mujoco::FindOrFetch: Using FetchContent to retrieve `eigen`" correct? I just want to make sure that the paste didn't get clipped.

zhihaocheng, 2022-06-22:
Thank you. I can confirm that the last line is "mujoco::FindOrFetch: Using FetchContent to retrieve eigen".

In terms of the GCC version, I also loaded and tried GCC 9.1.0, which is the newest version on the server. However, it still got stuck as my last post. Besides, I have tried to use llvm/14.0.0 and clang/6.0.1. I manually load these packages and export them as compliers. And then run pip install mujoco. However, I waited for about 1.5 hours and it got stuck again. The output is attached  at the end of this post. 

Actually, the server provides various compilers, which are listed as follows
(testmujoco) [zhc]$ module avail |& grep gcc
fftw/2.1.5-gcc
fftw/3.3.3-gcc(default)
fftw/3.3.4-gcc
gcc/12.1.0
gcc/4.8.4
gcc/4.9.0
gcc/4.9.3(default)
gcc/4.9.4
gcc/5.4.0
gcc/5.5.0
gcc/6.2.0
gcc/7.4.0
gcc/7.5.0
gcc/8.4.0
gcc/9.1.0
gromacs/4.6.7-openmpi-gcc
gromacs/5.1.4-openmpi-gcc
gromacs/5.1-openmpi-gcc
gromacs-plumed/5.1.4-openmpi-gcc-plumed-2.2.4
gromacs-plumed/5.1.4-openmpi-gcc-plumed-2.3.0(default)
hdf5-gcc-mpich-3.2/1.10.1(default)
hdf5-gcc-openmpi/1.10.1
hdf5-gcc-openmpi/1.8.16(default)
hdf5-gcc-openmpi-1.10.3/1.10.4(default)
hdf5-gcc-openmpi-3.0.0/1.10.1(default)
hdf5-gcc-openmpi-3.1.3/1.10.4(default)
hdf5-gcc-openmpi-3.1.3-cuda10/1.10.4(default)
hdf5-gcc-openmpi-cuda/1.10.1(default)
liggghts-public/3.8.0-openmpi-gcc
molpro/2015-gcc-mpi
molpro/2015-gcc-mpi-test2
namd/2.10-openmpi-gcc(default)
namd/2.11-openmpi-gcc
namd/2.12-openmpi-gcc
openmpi-gcc/1.10.2
openmpi-gcc/1.10.3
openmpi-gcc/1.10.7
openmpi-gcc/1.8.4(default)
openmpi-gcc/1.8.4-6.x
openmpi-gcc/1.8.4-backup
openmpi-gcc/2.1.2
openmpi-gcc/3.0.0
openmpi-gcc/3.0.0-64
openmpi-gcc/3.0.0-cuda
openmpi-gcc/3.0.1
openmpi-gcc/3.1.3
openmpi-gcc/3.1.3-cuda10
openmpi-gcc/3.1.4
openmpi-gcc/3.1.5
openmpi-gcc/3.1.5a
openmpi-gcc/4.0.0-cuda10
openmpi-gcc/4.0.3
openmpi-gcc/4.1.1
petsc-gcc/3.7.6(default)
petsc-gcc-mpich/3.11.1
petsc-gcc-mpich/3.13.6
petsc-gcc-mpich/3.14.1
petsc-gcc-mpich/3.15.2
petsc-gcc-mpich/3.7.6
petsc-gcc-mpich/3.9.3(default)
petsc-gcc-openmpi/3.7.6(default)
petsc-gcc-openmpi/3.8.4
petsc-gcc-openmpi/3.9.3

[... 254 lines cut ...]
  -- Performing Test SUPPORTS_GC_SECTIONS - Success
  -- Found Python3: /usr/local/python/3.7.7/bin/python3.7 (found version "3.7.7") found components: Interpreter Development Development.Module Development.Embed
  MuJoCo is at /project/RDS-FEI-ImitationLearn-RW/mujocosource/mujoco-2.2.0/lib/libmujoco.so
  MuJoCo headers are at /project/RDS-FEI-ImitationLearn-RW/mujocosource/mujoco-2.2.0/include
  -- mujoco::FindOrFetch: checking for targets in package `absl`
  -- mujoco::FindOrFetch: checking for targets in package `absl` - target `absl::core_headers` not defined.
  -- mujoco::FindOrFetch: Using FetchContent to retrieve `abseil-cpp`
  CMake Warning at /tmp/pbs.6371704.pbsserver/pip-install-y9bq5kdf/mujoco_4d5013efb38040138a506bb606318bb4/build/temp.linux-x86_64-3.7/_deps/abseil-cpp-src/CMakeLists.txt:70 (message):
    A future Abseil release will default ABSL_PROPAGATE_CXX_STD to ON for CMake
    3.8 and up.  We recommend enabling this option to ensure your project still
    builds correctly.

  -- Looking for pthread.h
  -- Looking for pthread.h - found
  -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
  -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
  -- Looking for pthread_create in pthreads
  -- Looking for pthread_create in pthreads - not found
  -- Looking for pthread_create in pthread
  -- Looking for pthread_create in pthread - found
  -- Found Threads: TRUE
  -- mujoco::FindOrFetch: Using FetchContent to retrieve `abseil-cpp` - Done
  -- mujoco::FindOrFetch: checking for targets in package `Eigen3`
  -- mujoco::FindOrFetch: checking for targets in package `Eigen3` - target `Eigen3::Eigen` not defined.
  -- mujoco::FindOrFetch: Using FetchContent to retrieve `eigen`

nimrod-gileadi, 2022-06-29:
Are you able to access the `eigen` git repository from the server? i.e:

```
git clone -n https://gitlab.com/libeigen/eigen
cd eigen
git checkout 3147391d946bb4b6c68edd901f2add6ac1f31f8c
```

saran-t, 2022-07-15:
@zhihaocheng any luck?

zhihaocheng, 2022-07-16:
Thanks @nimrod-gileadi. Thank you @saran-t. I have tried to git clone `eigen` and the server is able to access it. A difference is that the code working is as follows. However, even if I clone `eigen`, the same problem occurs as in my last post.

```
git clone -n https://gitlab.com/libeigen/eigen.git
```

saran-t, 2022-08-01:
Is this issue still affecting you?

zhihaocheng, 2022-08-07:
> Is this issue still affecting you?

Thanks @saran-t and sorry for the late reply. Actually, I cannot solve this problem until now. I can only run experiments on some small workstations that I have the root privilege rather than the HPC of the university, which slows down my research. Thanks again for you contact, and I will close this issue.

saran-t, 2022-08-07:
@zhihaocheng the only suggestion I have is to clone the Eigen repo manually and point CMake to it using the `-DFETCHCONTENT_SOURCE_DIR_EIGEN3:PATH=/path/to/your/eigen"` flag.

zhihaocheng, 2022-08-08:
Hi @saran-t , thanks for your advice. I will try it. In the previous discussion, you said that the problem is due to running on CentOS 6. Would it be possible that I build the mujoco on a CentOS 6 machine that I have full control from source and then migrate the compiled files to the HPC of university?

saran-t, 2022-08-08:
Yes that should work.

zhihaocheng, 2022-08-08:
Thanks @saran-t , I'm one step closer to solving the problem. Today, I tried to install the Python Mujoco by building from source. I find that the previous problem is resulted from the URL of eigen. If I replace "https://gitlab.com/libeigen/eigen" with "https://gitlab.com/libeigen/eigen.git" in mujoco-2.2.1/python/mujoco/CMakeLists.txt, the building wheel process with not be stuck. However, a further problem appeared as follows. I have searched in Google, but failed to find what is wrong with /usr/bin/ld. Could you please help me?

Currently Loaded Modulefiles:
  1) zlib/1.2.8       3) python/3.7.7     5) gmp/5.1.3        7) mpc/1.0.3        9) llvm/14.0.0
  2) sqlite/3080802   4) cmake/3.21.1     6) mpfr/3.1.4       8) gcc/8.4.0       10) clang/6.0.1

I also explicitly assign the complier.
export CC=/usr/local/llvm/14.0.0/bin/clang
export CXX=/usr/local/llvm/14.0.0/bin/clang++

pip wheel dist/mujoco-2.2.1.tar.gz, and then

  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/str_cat.cc.o
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/numbers.cc.o
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/str_replace.cc.o
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/str_split.cc.o
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/string_view.cc.o
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/substitute.cc.o
  [ 58%] Linking CXX static library libabsl_city.a
  [ 58%] Built target absl_city
  [ 58%] Linking CXX static library libabsl_malloc_internal.a
  [ 58%] Built target absl_malloc_internal
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/synchronization/CMakeFiles/absl_graphcycles_internal.dir/internal/graphcycles.cc.o
  [ 58%] Linking CXX static library libabsl_demangle_internal.a
  [ 58%] Built target absl_demangle_internal
  [ 58%] Linking CXX static library libabsl_graphcycles_internal.a
  [ 58%] Built target absl_graphcycles_internal
  [ 58%] Linking CXX static library libabsl_strings.a
  [ 58%] Built target absl_strings
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/debugging/CMakeFiles/absl_symbolize.dir/symbolize.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/hash/CMakeFiles/absl_hash.dir/internal/hash.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time.dir/civil_time.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_internal.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time.dir/clock.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time.dir/duration.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_btree.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time.dir/time.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time.dir/format.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_btree_navigator.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_btree_reader.cc.o
  [ 75%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_concat.cc.o
  [ 75%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_crc.cc.o
  [ 75%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_ring.cc.o
  [ 75%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_consume.cc.o
  [ 75%] Linking CXX shared module _constants.cpython-37m-x86_64-linux-gnu.so
  /usr/bin/ld: unrecognized option '-plugin'
  /usr/bin/ld: use the --help option for usage information
  clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
  gmake[2]: *** [_constants.cpython-37m-x86_64-linux-gnu.so] Error 1
  gmake[1]: *** [CMakeFiles/_constants.dir/all] Error 2
  gmake[1]: *** Waiting for unfinished jobs....
  [ 75%] Linking CXX static library libabsl_symbolize.a
  [ 75%] Built target absl_symbolize
  [ 75%] Linking CXX static library libabsl_time.a
  [ 75%] Built target absl_time
  [ 75%] Linking CXX static library libabsl_hash.a
  [ 75%] Built target absl_hash
  [ 75%] Linking CXX static library libabsl_cord_internal.a
  [ 75%] Built target absl_cord_internal
  [ 75%] Linking CXX shared module _errors.cpython-37m-x86_64-linux-gnu.so
  /usr/bin/ld: unrecognized option '-plugin'
  /usr/bin/ld: use the --help option for usage information
  clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
  gmake[2]: *** [_errors.cpython-37m-x86_64-linux-gnu.so] Error 1
  gmake[1]: *** [CMakeFiles/_errors.dir/all] Error 2
  [ 75%] Linking CXX shared module _enums.cpython-37m-x86_64-linux-gnu.so
  /usr/bin/ld: unrecognized option '-plugin'
  /usr/bin/ld: use the --help option for usage information
  clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
  gmake[2]: *** [_enums.cpython-37m-x86_64-linux-gnu.so] Error 1
[... cut ...]
  File "/home/zhc/.local/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 247, in wrapper
    return func(self, options, args)
  File "/home/zhc/.local/lib/python3.7/site-packages/pip/_internal/commands/wheel.py", line 176, in run
    raise CommandError("Failed to build one or more wheels")
pip._internal.exceptions.CommandError: Failed to build one or more wheels
Remote version of pip: 22.2.2
Local version of pip:  22.2.2
Was pip installed by pip? True
Removed build tracker: '/tmp/pbs.6440532.pbsserver/pip-build-tracker-ey5d2xv0'

zhihaocheng, 2022-08-08:
Thanks @saran-t , I'm one step closer to solving the problem. Today, I tried to install the Python Mujoco by building from source. I find that the previous problem is resulted from the URL of eigen. If I replace "https://gitlab.com/libeigen/eigen" with "https://gitlab.com/libeigen/eigen.git" in mujoco-2.2.1/python/mujoco/CMakeLists.txt, the building wheel process with not be stuck. However, a further problem appeared as follows. I have searched in Google, but failed to find what is wrong with /usr/bin/ld. Could you please help me?

Currently Loaded Modulefiles:
  1) zlib/1.2.8       3) python/3.7.7     5) gmp/5.1.3        7) mpc/1.0.3        9) llvm/14.0.0
  2) sqlite/3080802   4) cmake/3.21.1     6) mpfr/3.1.4       8) gcc/8.4.0       10) clang/6.0.1

I also explicitly assign the complier.
export CC=/usr/local/llvm/14.0.0/bin/clang
export CXX=/usr/local/llvm/14.0.0/bin/clang++

pip wheel dist/mujoco-2.2.1.tar.gz, and then

  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/str_cat.cc.o
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/numbers.cc.o
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/str_replace.cc.o
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/str_split.cc.o
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/string_view.cc.o
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_strings.dir/substitute.cc.o
  [ 58%] Linking CXX static library libabsl_city.a
  [ 58%] Built target absl_city
  [ 58%] Linking CXX static library libabsl_malloc_internal.a
  [ 58%] Built target absl_malloc_internal
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/synchronization/CMakeFiles/absl_graphcycles_internal.dir/internal/graphcycles.cc.o
  [ 58%] Linking CXX static library libabsl_demangle_internal.a
  [ 58%] Built target absl_demangle_internal
  [ 58%] Linking CXX static library libabsl_graphcycles_internal.a
  [ 58%] Built target absl_graphcycles_internal
  [ 58%] Linking CXX static library libabsl_strings.a
  [ 58%] Built target absl_strings
  [ 58%] Building CXX object _deps/abseil-cpp-build/absl/debugging/CMakeFiles/absl_symbolize.dir/symbolize.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/hash/CMakeFiles/absl_hash.dir/internal/hash.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time.dir/civil_time.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_internal.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time.dir/clock.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time.dir/duration.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_btree.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time.dir/time.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time.dir/format.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_btree_navigator.cc.o
  [ 66%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_btree_reader.cc.o
  [ 75%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_concat.cc.o
  [ 75%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_crc.cc.o
  [ 75%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_ring.cc.o
  [ 75%] Building CXX object _deps/abseil-cpp-build/absl/strings/CMakeFiles/absl_cord_internal.dir/internal/cord_rep_consume.cc.o
  [ 75%] Linking CXX shared module _constants.cpython-37m-x86_64-linux-gnu.so
  /usr/bin/ld: unrecognized option '-plugin'
  /usr/bin/ld: use the --help option for usage information
  clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
  gmake[2]: *** [_constants.cpython-37m-x86_64-linux-gnu.so] Error 1
  gmake[1]: *** [CMakeFiles/_constants.dir/all] Error 2
  gmake[1]: *** Waiting for unfinished jobs....
  [ 75%] Linking CXX static library libabsl_symbolize.a
  [ 75%] Built target absl_symbolize
  [ 75%] Linking CXX static library libabsl_time.a
  [ 75%] Built target absl_time
  [ 75%] Linking CXX static library libabsl_hash.a
  [ 75%] Built target absl_hash
  [ 75%] Linking CXX static library libabsl_cord_internal.a
  [ 75%] Built target absl_cord_internal
  [ 75%] Linking CXX shared module _errors.cpython-37m-x86_64-linux-gnu.so
  /usr/bin/ld: unrecognized option '-plugin'
  /usr/bin/ld: use the --help option for usage information
  clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
  gmake[2]: *** [_errors.cpython-37m-x86_64-linux-gnu.so] Error 1
  gmake[1]: *** [CMakeFiles/_errors.dir/all] Error 2
  [ 75%] Linking CXX shared module _enums.cpython-37m-x86_64-linux-gnu.so
  /usr/bin/ld: unrecognized option '-plugin'
  /usr/bin/ld: use the --help option for usage information
  clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
  gmake[2]: *** [_enums.cpython-37m-x86_64-linux-gnu.so] Error 1
[... cut ...]
  File "/home/zhc/.local/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 247, in wrapper
    return func(self, options, args)
  File "/home/zhc/.local/lib/python3.7/site-packages/pip/_internal/commands/wheel.py", line 176, in run
    raise CommandError("Failed to build one or more wheels")
pip._internal.exceptions.CommandError: Failed to build one or more wheels
Remote version of pip: 22.2.2
Local version of pip:  22.2.2
Was pip installed by pip? True
Removed build tracker: '/tmp/pbs.6440532.pbsserver/pip-build-tracker-ey5d2xv0'

closed 2022-08-22 by zhihaocheng

raviwaarr, 2023-10-04:
I am having issue building mujoco on apple sillicon m1

  Building wheel for mujoco (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for mujoco (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [71 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.macosx-10.9-x86_64-cpython-311
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/rollout.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/rollout_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/bindings_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/render_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      copying mujoco/egl/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      copying mujoco/egl/egl_ext.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/osmesa
      copying mujoco/osmesa/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/osmesa
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/glfw
      copying mujoco/glfw/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/glfw
      running build_ext
      Traceback (most recent call last):
        File "/tmp/mujoco/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/tmp/mujoco/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/mujoco/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 434, in build_wheel
          return self._build_with_temp_dir(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 419, in _build_with_temp_dir
          self.run_setup()
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 507, in run_setup
          super(_BuildMetaLegacyBackend, self).run_setup(setup_script=setup_script)
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 341, in run_setup
          exec(code, locals())
        File "<string>", line 265, in <module>
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/wheel/bdist_wheel.py", line 364, in run
          self.run_command("build")
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-build-env-bahz24fd/overlay/lib/python3.11/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
[... cut ...]
  × Building wheel for pygame (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [13 lines of output]
      src_c/_sdl2/sdl2.c:211:12: fatal error: 'longintrepr.h' file not found
        #include "longintrepr.h"
                 ^~~~~~~~~~~~~~~
      1 error generated.

      ---
      For help with compilation see:
          https://www.pygame.org/wiki/MacCompile
      To contribute to pygame development see:
          https://www.pygame.org/contribute.html
      ---

      error: command '/usr/bin/clang' failed with exit code 1

raviwaarr, 2023-10-04:
I am stuck with installing mujoco on apple siliicon m1

Building wheels for collected packages: mujoco
  Building wheel for mujoco (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [68 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.macosx-10.9-x86_64-cpython-311
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/viewer.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/renderer.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/renderer_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/rollout.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/rollout_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/bindings_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/viewer_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/render_test.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      copying mujoco/gl_context.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/cgl
      copying mujoco/cgl/cgl.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/cgl
      copying mujoco/cgl/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/cgl
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      copying mujoco/egl/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      copying mujoco/egl/egl_ext.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/egl
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/osmesa
      copying mujoco/osmesa/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/osmesa
      creating build/lib.macosx-10.9-x86_64-cpython-311/mujoco/glfw
      copying mujoco/glfw/__init__.py -> build/lib.macosx-10.9-x86_64-cpython-311/mujoco/glfw
      running build_ext
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-install-pscr4xgl/mujoco_d4b94e5e211c43dcbadb4e0ddb330867/setup.py", line 418, in <module>
          setup(**SETUP_KWARGS)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/wheel/bdist_wheel.py", line 364, in run
          self.run_command("build")
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/command/build.py", line 131, in run
          self.run_command(cmd_name)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/Users/raviwar/miniconda/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-install-pscr4xgl/mujoco_d4b94e5e211c43dcbadb4e0ddb330867/setup.py", line 156, in run
          self._mujoco_framework_path) = self._find_mujoco()
                                         ^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/25/3c_cvnxx415_zr7xxz5k1_zh0000gn/T/pip-install-pscr4xgl/mujoco_d4b94e5e211c43dcbadb4e0ddb330867/setup.py", line 170, in _find_mujoco
          raise RuntimeError(
      RuntimeError: MUJOCO_PATH environment variable is not set
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mujoco
  Running setup.py clean for mujoco
Failed to build mujoco

bennevans, 2023-10-18:
@raviwaarr Make sure to set your environment variables MUJOCO_PATH and MUJOCO_PLUGIN_PATH to point to the correct directories (the root mujoco path like /home/users/ben/.mujoco/mujoco-3.0.0 and /home/users/ben/.mujoco/mujoco-3.0.0/bin/mujoco_plugin)

ocg2347, 2023-11-14:
best not use conda environments if possible. I switched to python venv and nothing went wrong :)