google-deepmind/mujoco #350: Is it possible to arbitrary relative position for weld constraint?

XinyuWang2, 2022-06-18:
Hi,

I'm building an environment that uses a box-shaped cursor to select objects. It should like very much like the video from IKEA env (https://clvrai.github.io/furniture/), but they didn't use actuators for the cursor.

The expected design is that the weld constraint is activated when the cursor collides with an object. This is the "grab". The cursor is allowed to overlap with any objects. I find 

I'm looking for some help with the following:
Make the relative pose for the constraint as what it is at the moment of activation, rather than some fixed one. What I have right now is every time the constraint is activated, the cursor or the object suddenly moves to another and even farther because of its moment.

Here is a model which explains my question:

```XML
<mujoco model="base">
  <actuator>
      <motor joint='t_joint_x' ctrllimited='true' ctrlrange='-2.0 2.0'/>
      <motor joint='t_joint_y' ctrllimited='true' ctrlrange='-2.0 2.0'/>
  </actuator>

  <equality>
    <weld active="true" body1="cursor0" body2="block1" solimp="1 1 0.5" solref="0.01 0.3" />
  </equality>

  <worldbody>
    <geom condim="3" friction="2.0 0.005 0.0001" name="FLOOR" pos="0 0 0" rgba="0.7 0.7 0.7 1" size="0.5 0.5 0.0625" type="plane" />
    <light castshadow="true" diffuse=".8 .8 .8" dir=".5 -.3 -.8" directional="true" pos="0 0 2.0" specular="0.3 0.3 0.3" />

    <body name="cursor0" pos="0.2 -0.3 0.05">
      <geom condim="3" gap="10" margin="0.05" name="cursor0" rgba="1 0 0 0.5" size="0.05 0.05 0.05" type="box" />
      <joint type='slide' axis='1 0 0' damping='0.5' pos='0 0 0' name='t_joint_x' range='-10.0 10.0'/>
      <joint type='slide' axis='0 1 0' damping='0.5' pos='0 0 0' name='t_joint_y' range='-10.0 10.0'/>
    </body>

    <body name="block1" pos="0.201 0.0 0.05" quat="1 0 0 0">
      <geom density="100" friction="1 10 0.5"  name="block1-0" rgba="0.82 0.71 0.55 1" size="0.1 0.02 0.05" solref="0.001 1" type="box" />
      <joint damping="0.0001" name="block1" type="free" />
    </body>

  </worldbody>
</mujoco>

```

Thank you so much!

yuvaltassa, 2022-06-20:
Hi, thanks for the model, but it did not clarify what you are trying to do.

Are you trying to set the relative pose for the weld constraint at run time or at design time?

XinyuWang2, 2022-06-20:
Yes, I want to set the relative pose for the weld constraint at run time. Say there is a big box object, and a cursor (very small). The cursor moves around and can touch any part of the box. I wish to activate the constraint when the cursor touches the box, and the relative pose should just be like that, with no more changes.

closed 2022-06-22 by yuvaltassa