google-deepmind/mujoco #355: compilation on Windows

alexxlzhou, 2022-06-22:
Hi there,

I am trying to compile mujoco on Windows (Win10, VS2019) and have some troubles with it.

The first issue comes from applying qhull patch, which generates the error below:
  Performing patch step for 'qhull-populate'
CUSTOMBUILD : error : unknown switch `q'
  usage: git apply [<options>] [<patch>...]

After fixing this and generating project files, there are more errors during compilation, most errors are 
error C2059: syntax error: 'type'
related to the X Macros.

Anyone has similar issues or know how to fix this?

Thank you.

Alex

saran-t, 2022-06-23:
Not sure what happened with the patch step, but since you seem to have fixed it could you please send a PR so that we can take a look?

The compilation error is almost definitely because of the use of C11 features (specifically in `engine_array_safety.h`) which aren't supported until MSVC 2021. You should be able to insert some `#ifdef` guards in that file to detect MSVC < 2021 and bypass the typesafety checks.

alexxlzhou, 2022-06-26:
Very helpful, thanks! 
The code compile fine after I updated my VC2019!
For the qhull issue, I simply removed the "-q". 
And according to the comment in the cmake file, the patch will be removed when https://github.com/qhull/qhull/pull/112 is merged. 
So I don't think it is necessary to send a PR.

closed 2022-06-29 by alexxlzhou