google-deepmind/mujoco #358: Defining a Mesh as Floor for simulation

hosseinfeiz, 2022-06-23:
Hi,

I'm a student and I'm trying to use MuJoCo for Correcting the Poses estimated from videos.

I'm looking for some help with XML part of simulation.
I want to use a mesh of the ring as the plane(the whole possible space of motion) to restrict the motion to inside the ring. What should I add the ring mesh? 
Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
  <asset>
    <texture type="skybox" builtin="gradient" rgb1=".3 .5 .7" rgb2="0 0 0" width="512" height="512"/>
    <texture name="body" type="cube" builtin="flat" mark="cross" width="127" height="1278"
             rgb1="0.8 0.6 0.4" rgb2="0.8 0.6 0.4" markrgb="1 1 1" random="0.01"/>
    <material name="body" texture="body" texuniform="true" rgba="0.8 0.6 .4 1"/>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1=".1 .2 .3" rgb2=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".4"/>
    <mesh file="ring.stl"/>
  </asset>

  <worldbody>
    <geom name="floor" type="plane" conaffinity="1" size="100 100 .2" material="grid"/>
    <geom type="mesh"  mesh="ring" pos="0.2 -0.7 0" rgba="0.3 0.3 1.3 1"/>

</details>

Here is a  video, illustrating my question:

https://user-images.githubusercontent.com/64847793/175394608-2dda4e5d-8509-47f6-adde-42ceab17732f.mov

kevinzakka, 2022-06-24:
I'm confused at what you are trying to do, could you be more specific?

hosseinfeiz, 2022-06-24:
Hi, How can I add a mesh as a rigid body? I need the mesh to act as the floor for the humanoid that  I am using. Can you watch the video the I send?
[image]

yuvaltassa, 2022-06-25:
If I'm understanding your question, the answer is: you cannot do that.
Meshes collisions use the mesh's convex hull, not the actual mesh. So your options are:
1. Decompose the mesh into convex parts using e.g. [VHACD](https://github.com/kmammou/v-hacd).
2. Build this decomposition yourself from primitives like e.g. in the [mug](https://github.com/deepmind/mujoco/blob/main/model/mug/mug.xml) example.

Looking at your boxing ring mesh, I think option 2 is very easy to do, the whole thing can be well approximated by 5 boxes, no?

closed 2022-06-25 by yuvaltassa