google-deepmind/mujoco #359: Can't download older versions of mujoco

RamiAI, 2022-06-24:
Hi,

I have a software that requires an older version of mujoco (mjpro150_linux), but it seems like the website where I download the code has an expired SSL certificate (specifically the download of the file https://roboti.us/download/mjpro150_linux.zip and the key https://roboti.us/file/mjkey.txt doesn't work). If you are responsible for this, would it be possible to fix this problem?

If you are not responsible for this, can you tell me where I can get the code and the key for these older versions? I have not found anything in the branches and tags of this repo.

thowell, 2022-06-24:
Wayback machine works for me: 

https://web.archive.org/web/20200918071752/https://roboti.us/download/mjpro150_linux.zip
https://web.archive.org/web/20220315114911/https://roboti.us/file/mjkey.txt

saran-t, 2022-06-24:
Another workaround for the time being is to use the non-secure server, i.e. http://roboti.us/ rather than https://roboti.us/

Roboti is responsible for hosting binaries and keys for the legacy, pre-DeepMind versions of MuJoCo. We have asked them to look into this as soon as possible. Thanks for pointing this out to us!

yuvaltassa, 2022-06-25:
Since using `http://` rather than `https://` is an easy workaround until this is fixed properly, I am closing the issue.

closed 2022-06-25 by yuvaltassa