google-deepmind/mujoco #362: Rotation mesh frame without rotating its geom (picture)

aliemil, 2022-06-26:
Does anyone know how to change the frame orientation of that middle geometry's mesh, so it equals the neighbouring ones? 
Hence rotating the frame around the z achsis (blue), without rotating the geom/mesh itself.

I tried to change the orientation of the STL file but the frame is always assigned that way from mujoco itself.

[image]

aliemil, 2022-06-29:
I created a custom python environment where a robot needs to position several into their designated position. Once one segment is positioned (the red segments on the picture are showing the target position of their respective segment. 
After a segment is positioned, the end-effector segment shall change to initiate the next placement.
This I am achieving changing the mesh ID with `model.geom_dataid` .
The problem is that when mujoco changes the mesh ID it seems to reset the orientation of the mesh to the one declared by mujoco. Therefore I dont think that I can easily use the body frame, as it switching the mesh. Hence I would need to change the body orientation of that particular end-effector body, but only for the short segment. 

Here is a sample of the issue: https://github.com/aliemil/tbm

Uploaded are a sample xml and 3 STL's:
1. The long segment with the desired frame
2. The short segment with the undesired frame
3. Same short segment rotated in CAD

Also thanks for the help! :)

nimrod-gileadi, 2022-06-30:
Very cool application!

Are these geoms just visual, or do you want them to have inertia and generate contacts?

1. If they're just visual, rather than changing the `geom_dataid` at run time, have you considered adding all the possible geoms as children of the end effector, but putting them all in a visualization group that's hidden? At run time you can toggle the visibility of a geom by changing its visualization group.
2. If these geoms aren't just visual geoms, and you want them to have inertia and generate contacts, then switching the `geom_dataid` or toggling visualization groups isn't going to do the right thing. If that's the case, you may want to make the geoms separate free bodies, and use [`<weld>`](https://mujoco.readthedocs.io/en/latest/XMLreference.html#equality-weld) constraints to glue them to the end effector. This will be slightly tricky. You would define one weld constraint in your XML, and modify `model.eq_obj2id` at runtime.

aliemil, 2022-06-30:
Thanks, its the ring building phase of a tunnel boring machine - a part of my Master Thesis :) 
In the video you can see what exactly I mean. The collisions are turnt of, as my agent is not skilled good enought to avoid them yet.

https://user-images.githubusercontent.com/74488721/176778832-fb691ef1-205b-4fa1-a4aa-507c3b4db046.mp4

The end-effector segment shall be able to generate contacts with the white colored segments.
You brought me to an idea! In fact I could do what you suggested in 1. and set all non selected segments transparent, conaffinity = contype = 0 and their masses ~ 0. 
I think that should work. Thanks a lot! :)