google-deepmind/mujoco #365: Computing Jacobian of the Center on Mass of the Robot

dhruvthanki, 2022-07-01:
Hi,

I'm trying to calculate the jacobian of the COM of the robot. Is this the right way to do it?
```python
def jacpCOM(model, data):
    total_mass = mujoco.mj_getTotalmass(model)
    jacp_i = np.zeros((3,model.nv))
    jacp_COM = np.zeros((3,model.nv))
    for i in range(model.nbody):
        mi = model.body_mass[i]/total_mass
        mujoco.mj_jacBodyCom(model, data, jacp_i, None, i)
        jacp_COM += mi*jacp_i
    return jacp_COM
```

yuvaltassa, 2022-07-01:
The short answer is yes 🙂 

The slightly longer answer is: You are looping over all bodies, this might not be what you want. Specifically you are including static bodies with no DoFs, which you probably don't want. Better to loop over a subtree passed as input parameter (bodies with `model.body_rootid==n`).

closed 2022-07-01 by yuvaltassa

yuvaltassa, 2022-07-29:
This function is now [implemented natively](https://github.com/deepmind/mujoco/blob/fcf41317710afa718b31b66ec4191f2dd5779cfa/src/engine/engine_support.c#L158).