google-deepmind/mujoco #366: Bin packing not stable

caiqi, 2022-07-02:
I am trying to build a bin packing model with mujoco. I found that the simulation is not stable even though the packed objects "should" be stable. Following is the code to reproduce. I have searched the Google but could not find any solutions. Is there anything I miss to simulate optimal 3D bin packing only considering gravity stability? Thanks!

``` python
# The basic mujoco wrapper.
from dm_control import mujoco
import numpy as np
import cv2
n_frames = 1000
frame_rate = 10
height = 480
width = 640

def create_videos(images):
    for image in images:
        cv2.imshow('Frame', image)
        # Press Q on keyboard to  exit
        if cv2.waitKey(25) & 0xFF == ord('q'):
            break
base_output = ""
packing_str = np.zeros((10, 6))
packing_str[:, 0] = 0
packing_str[:, 1] = 0
packing_str[:, 2] = np.arange(0, packing_str.shape[0]) * 0.1 + 0.05
packing_str[:, 3] = np.random.rand(packing_str.shape[0]) * 0.1
packing_str[:, 4] = np.random.rand(packing_str.shape[0]) * 0.1
packing_str[:, 5] = 0.05

template = """
    <body name="box{}" pos="{} {} {}">  
        <freejoint/>
        <geom name="geom_box{}" type="box" size="{} {} {}" rgba="{} {} {} 1" condim="1"/>
    </body>
"""

for idx in range(len(packing_str)):
    packing_str_ = packing_str[idx]
    colors = np.random.rand(3).tolist()
    output = template.format(idx, packing_str_[0], packing_str_[1], packing_str_[2], idx, packing_str_[3],
                             packing_str_[4], packing_str_[5], colors[0], colors[1], colors[2])
    base_output += output + "\n"
"""    <body name="box1" pos="0 0 0">
        <freejoint/>
        <geom name="geom_box1" type="box" size="0.1 0.1 0.1" rgba="1 0 0 1"/>
    </body>
        <body name="box2" pos="0.2 0.2 0.2">
        <freejoint/>
        <geom name="geom_box2" type="box" size="0.1 0.1 0.1" rgba="1 1 0 1"/>
    </body>"""
free_body_MJCF = """
<mujoco>
<size njmax="5000" nconmax="5000" />
  <asset>
    <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3" 
    rgb2=".2 .3 .4" width="300" height="300" mark="edge" markrgb=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="2 2" texuniform="true" 
    reflectance=".2"/>
  </asset>
  <worldbody>
    <light pos="0 0 1"/>
    <geom name="ground" type="plane" pos="0 0 0" size="2 2 .1" material="grid" solimp=".99 .99 .01" solref=".001 1"/>
    {}
  </worldbody>
</mujoco>
""".format(base_output)

physics = mujoco.Physics.from_xml_string(free_body_MJCF)

frames = np.zeros((n_frames // frame_rate, height, width, 3), dtype=np.uint8)
for i in range(n_frames):
    physics.step()
    if i % frame_rate == 0 and i > 0:
        frames[i // frame_rate] = physics.render(height, width)
create_videos(frames)
```

https://user-images.githubusercontent.com/8349599/176992610-68b1a4fd-c5ca-480f-aec0-7c7cd0eaef97.mp4

yuvaltassa, 2022-07-03:
Hi,

Could you please paste a specific model string (the input to `from_xml_string`)?

caiqi, 2022-07-04:
@yuvaltassa This is the body MJCF. 
```python
free_body_MJCF = """
<mujoco>
<size njmax="5000" nconmax="5000" />
  <asset>
    <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3" 
    rgb2=".2 .3 .4" width="300" height="300" mark="edge" markrgb=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="2 2" texuniform="true" 
    reflectance=".2"/>
  </asset>
  <worldbody>
    <light pos="0 0 1"/>
    <geom name="ground" type="plane" pos="0 0 0" size="2 2 .1" material="grid" solimp=".99 .99 .01" solref=".001 1"/>
    
    <body name="box0" pos="0.0 0.0 0.05">  
        <freejoint/>
        <geom name="geom_box0" type="box" size="0.06561746120008799 0.04738517796267593 0.05" rgba="0.009696437179144923 0.420189016584275 0.8829982521623548 1" condim="1"/>
    </body>
    <body name="box1" pos="0.0 0.0 0.15000000000000002">  
        <freejoint/>
        <geom name="geom_box1" type="box" size="0.06652134085772976 0.028893415419341262 0.05" rgba="0.2956480063844368 0.3001904997982424 0.11454123968493557 1" condim="1"/>
    </body>
    <body name="box2" pos="0.0 0.0 0.25">  
        <freejoint/>
        <geom name="geom_box2" type="box" size="0.07264730840920187 0.043260497746420525 0.05" rgba="0.2483086875372491 0.3573402529311226 0.11723846289193163 1" condim="1"/>
    </body>
    <body name="box3" pos="0.0 0.0 0.35000000000000003">  
        <freejoint/>
        <geom name="geom_box3" type="box" size="0.08470889362891687 0.07940626251864058 0.05" rgba="0.36447724022801886 0.7303387927072464 0.5102924664706513 1" condim="1"/>
    </body>
    <body name="box4" pos="0.0 0.0 0.45">  
        <freejoint/>
        <geom name="geom_box4" type="box" size="0.07724942546487484 0.009476644296396075 0.05" rgba="0.5684318864291983 0.9194317957100564 0.15405505852149515 1" condim="1"/>
    </body>
    <body name="box5" pos="0.0 0.0 0.55">  
        <freejoint/>
        <geom name="geom_box5" type="box" size="0.08905100369025644 0.09008062415663218 0.05" rgba="0.07378669297875662 0.8103308418005956 0.40403368088143954 1" condim="1"/>
    </body>
    <body name="box6" pos="0.0 0.0 0.6500000000000001">  
        <freejoint/>
        <geom name="geom_box6" type="box" size="0.025227385668945425 0.04933448820680403 0.05" rgba="0.6536745713784758 0.5993782870623209 0.6960554085905803 1" condim="1"/>
    </body>
    <body name="box7" pos="0.0 0.0 0.7500000000000001">  
        <freejoint/>
        <geom name="geom_box7" type="box" size="0.06504517831276206 0.09655778520946809 0.05" rgba="0.7181298817053159 0.9901195646781428 0.68003614190625 1" condim="1"/>
    </body>
    <body name="box8" pos="0.0 0.0 0.8500000000000001">  
        <freejoint/>
        <geom name="geom_box8" type="box" size="0.08593173127241356 0.0931113020003568 0.05" rgba="0.7944109873466195 0.2781012934132022 0.505847600688515 1" condim="1"/>
    </body>
    <body name="box9" pos="0.0 0.0 0.9500000000000001">  
        <freejoint/>
        <geom name="geom_box9" type="box" size="0.08766908479918473 0.04666486973186093 0.05" rgba="0.34085541595643054 0.27930845974814156 0.12149603872973636 1" condim="1"/>
    </body>
  </worldbody>
</mujoco>
"""
```

The following is a minimal running script: 
```python
# The basic mujoco wrapper.
from dm_control import mujoco
import numpy as np
import cv2
n_frames = 1000
frame_rate = 10
height = 480
width = 640
[... 50 lines cut ...]
        <geom name="geom_box7" type="box" size="0.06504517831276206 0.09655778520946809 0.05" rgba="0.7181298817053159 0.9901195646781428 0.68003614190625 1" condim="1"/>
    </body>
    <body name="box8" pos="0.0 0.0 0.8500000000000001">  
        <freejoint/>
        <geom name="geom_box8" type="box" size="0.08593173127241356 0.0931113020003568 0.05" rgba="0.7944109873466195 0.2781012934132022 0.505847600688515 1" condim="1"/>
    </body>
    <body name="box9" pos="0.0 0.0 0.9500000000000001">  
        <freejoint/>
        <geom name="geom_box9" type="box" size="0.08766908479918473 0.04666486973186093 0.05" rgba="0.34085541595643054 0.27930845974814156 0.12149603872973636 1" condim="1"/>
    </body>
  </worldbody>
</mujoco>
"""
print(free_body_MJCF)
physics = mujoco.Physics.from_xml_string(free_body_MJCF)
frames = np.zeros((n_frames // frame_rate, height, width, 3), dtype=np.uint8)
for i in range(n_frames):
    physics.step()
    if i % frame_rate == 0 and i > 0:
        frames[i // frame_rate] = physics.render(height, width)
create_videos(frames)
``` 
The simulation result is attached below. Thanks!

https://user-images.githubusercontent.com/8349599/177068845-fe1055d7-d2bb-4b19-8e86-7ae008bc0b52.mp4

yuvaltassa, 2022-07-04:
Hi,

I managed to stabilize this. 

1. The biggest problem with your model is that you set `condim="1"`. You cannot expect this to work. It is true that for infinite precision computations your model is technically stable, but MuJoCo is not infinite precision. Therefore there needs to be a mechanism to restabilize perturbations that are due to finite precision, otherwise errors will simply accumulate leading to toppling. This stabilising mechanism is friction, but you had disabled it, so there was no way for the model to restabilize. I just replaced `condim="1"` with `"3"` everywhere.
2. The second problem is that your contacts are too soft. This is because MuJoCo's default contacts are fairly soft by design, so the contacts are not stiff enough to reject the perturbations due to finite precision. You seemed to understand this since you stiffened the floor, but you forgot to stiffen everything else. I added the below to you model and deleted the custom contact params for the floor:
```
<default>
  <geom solimp="0.9 .99 .0001" solref=".001 1"/>
</default>
```

After these two changes the model is stable. 

Note however that if you make your thin boxes thinner it will topple again. You need to imagine that there is always small noise injected into your model (corresponding to finite precision errors), therefore unstable equilibria (like a ball balancing on a ball) will always fall at some point, just like they do real life...

closed 2022-07-04 by yuvaltassa