google-deepmind/mujoco #370: mjUI Python Bindings

rohit-kumar-j, 2022-07-03:
Python Bindings for mjUI

**Additional context**
[Add and access UI elements? #369](https://github.com/deepmind/mujoco/discussions/369)

nassir90, 2024-01-08:
Has there been any work on this? I am interested in this feature. If not, what are the main roadblocks and how feasible would it be for a new maintainer to add this API? Also, if there any alternatives to get custom UI elements in a project based on mujoco-python in a cross-platform manner anybody knows of I'd like to hear of them.

nassir90, 2024-01-12:
Just an update for anybody who is using python but wants a GUI system, I am now using [dearpygui](https://dearpygui.readthedocs.io/en/latest/) combined with mujoco's offscreen rendering capabilities here is an example of what it looks like [image].