google-deepmind/mujoco #371: Joint limit didn't work

Ddeconelee, 2022-07-04:
Hi,

I'm a student and I'm trying to use MuJoCo for simulation modeling of dexterous hand.

I'm looking for some help with joint limit setting.

Here is a model which explains my question:
[https://github.com/Ddeconelee/hand/blob/master/UR5_three_fingers_hand.xml](url)

Here is a screenshot, illustrating my question:
[image]
I set joint range in [0 : 1.5708] and joint limit as true, but when I use the actuator6 "IandMbending02", those joints' limits don't work, rotate joints such as mid_ joint2 become negative.

kevinzakka, 2022-07-08:
Hi @Ddeconelee, were you able to fix your issue?

yuvaltassa, 2022-07-08:
Hi @Ddeconelee, I suspect the problem is [here](https://github.com/Ddeconelee/hand/blob/master/UR5_three_fingers_hand.xml#L446), your position actuator is set to be allowed to push the joint to 12.217 radians, and has a quite a high gain (100 Nm/radian). Joint limits are soft in MuJoCo, so if you push on them hard enough you will be able to take them past their limit. Try applying `(12.217 - 1.57) * 100` Nm to one of your fingers and see what happens... 
:stuck_out_tongue_winking_eye:  (please don't do that)

closed 2022-07-08 by yuvaltassa

Ddeconelee, 2022-07-11:
@yuvaltassa Thank you very much for your answer. I tried to change Ctrlrange and Kp, but it didn't work. Is there any other way to restrict the joint to rotate in a negative direction (this obviously doesn't conform to common sense)?

yuvaltassa, 2022-07-11:
If you attach a zip file with a loadable model I will take a look.

Ddeconelee, 2022-07-11:
@yuvaltassa [git@github.com:Ddeconelee/three_finger_hand_zip.git](url) Thanks.

Ddeconelee, 2022-07-11:
[three_finger_hand.zip](https://github.com/deepmind/mujoco/files/9082415/three_finger_hand.zip)
This is the file, thanks!

yuvaltassa, 2022-07-11:
Okay, the model loads fine, I don't see any joints that bend in a strange way. What am supposed to look at?

Ddeconelee, 2022-07-11:
@yuvaltassa 
I want to make the proximal joint (joint1) of the finger bend forward, that is, when I control the actuator IandMbending02, the middle joint (joint2) and the distal joint (joint3) keep the zero angle unchanged (so I set the range (0:1.57)), but due to the angular coupling relationship between the three joints (tendon matrix Index&Middlefinger bending), when the proximal joint bends, the middle and distal joints bend to a negative angle, and the angle restriction is unsuccessful, as shown in the figure. In addition to setting the angle range, what ways can I keep the mid and distal joint at zero when the proximal joint bends forward?
[image]
Thanks!

yuvaltassa, 2022-07-11:
Sorry, none of this is happening in the model you attached above. In the model I have the fingers barely move in response to the actuators.

Ddeconelee, 2022-07-11:
[three_finger_hand_zip.zip](https://github.com/deepmind/mujoco/files/9083634/three_finger_hand_zip.zip)
Sorry,I have uploaded the correct version.

yuvaltassa, 2022-07-11:
I "fixed" this by making the joint limits stiffer in line 29:
```xml
<joint armature="0.01" damping=".1" solreflimit="4e-3 1" solimplimit=".95 .99 1e-3"/>
```

But really you shouldn't need this. Having these crazy high gains like this is a sign of bad design. You need to think carefully about your tendon coefs. Also you have tendon `range` defined but `limited` is not set to true so the tendon ranges are not doing anything...