google-deepmind/mujoco #373: building from source on jetson tx1 aarch64.

pixar0407, 2022-07-05:
Hi, thank you for the good work on mujoco.

I'm a (engineer) and I'm trying to use MuJoCo for my research.

I'm looking for some help with building from source on jetson tx1 aarch64_.

Here is a model which explains my question:

I followed the instructions from [here](https://mujoco.readthedocs.io/en/latest/programming.html#building-mujoco-from-source).
But i got error while doing _cmake --build ._ saying below.

I googled it. it seems like related to cmake problem. But i cannot solve it. Help me!

`[ 62%] Linking CXX static library ../../../lib/libgtest_main.a
[ 62%] Built target gtest_main
[ 62%] Building CXX object test/CMakeFiles/fixture_test.dir/fixture_test.cc.o
[ 62%] Linking CXX executable ../bin/fixture_test
../lib/libfixture.a(fixture.cc.o): In function `mujoco::SaveAndReadXml[abi:cxx11](mjModel_ const*)':
fixture.cc:(.text._ZN6mujoco14SaveAndReadXmlB5cxx11EPK8mjModel_+0x164): undefined reference to `std::filesystem::temp_directory_path[abi:cxx11]()'
fixture.cc:(.text._ZN6mujoco14SaveAndReadXmlB5cxx11EPK8mjModel_+0x1ac): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
fixture.cc:(.text._ZN6mujoco14SaveAndReadXmlB5cxx11EPK8mjModel_+0x1b4): undefined reference to `std::filesystem::__cxx11::path::has_root_directory() const'
fixture.cc:(.text._ZN6mujoco14SaveAndReadXmlB5cxx11EPK8mjModel_+0x284): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
fixture.cc:(.text._ZN6mujoco14SaveAndReadXmlB5cxx11EPK8mjModel_+0xd20): undefined reference to `std::filesystem::__cxx11::path::has_root_directory() const'
fixture.cc:(.text._ZN6mujoco14SaveAndReadXmlB5cxx11EPK8mjModel_+0xd30): undefined reference to `std::filesystem::__cxx11::path::has_filename() const'
fixture.cc:(.text._ZN6mujoco14SaveAndReadXmlB5cxx11EPK8mjModel_+0xda0): undefined reference to `std::filesystem::__cxx11::path::_M_split_cmpts()'
collect2: error: ld returned 1 exit status
test/CMakeFiles/fixture_test.dir/build.make:119: recipe for target 'bin/fixture_test' failed
make[2]: *** [bin/fixture_test] Error 1
CMakeFiles/Makefile2:5356: recipe for target 'test/CMakeFiles/fixture_test.dir/all' failed
make[1]: *** [test/CMakeFiles/fixture_test.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
`

kevinzakka, 2022-07-05:
You need c++17 to build from source.

pixar0407, 2022-07-06:
@kevinzakka 

Hi, thank you for your attention.
But i am confused with your answer. 

Because i already have the g++/cc -8 and -7 which are available to c++17.
(i read c++17 features are available  since gcc-5)

so, i compile g++/cc version 8.4.0. But i got this error.

hmm... does g++/cc 8.4.0 have c++17 compatibility, right?

kevinzakka, 2022-07-06:
The error log says `cxx11` so it looks like it's not configured correctly. Make sure CXX and CC are exported correctly and double check things with `which`.

pixar0407, 2022-07-06:
@kevinzakka 

Thank you again for quick answer.

So, does it need gcc/++ version 11?
(i am new to this. so, sorry for messing around.)

pixar0407, 2022-07-06:
@kevinzakka 

you are right! after upgrading to gcc version 11, the error got resolved.

But i got another error saying, (any suggestion?)

[ 79%] Building CXX object test/engine/CMakeFiles/engine_ray_test.dir/engine_ray_test.cc.o
[ 79%] Linking CXX executable ../../bin/engine_ray_test
[ 79%] Built target engine_ray_test
[ 80%] Building CXX object test/engine/CMakeFiles/engine_sensor_test.dir/engine_sensor_test.cc.o
/home/jetson/mujoco/test/engine/engine_sensor_test.cc: In member function ‘virtual void mujoco::{anonymous}::RelativeFrameSensorTest_ReferencePosMatQuat_Test::TestBody()’:
/home/jetson/mujoco/test/engine/engine_sensor_test.cc:159:79: error: class template argument deduction failed:
   std::vector expected_values(data->sensordata, data->sensordata+nsensordata/2);
                                                                               ^
/home/jetson/mujoco/test/engine/engine_sensor_test.cc:159:79: error: no matching function for call to ‘vector(mjtNum*&, mjtNum*)’
In file included from /usr/include/c++/7/vector:64:0,
                 from /usr/include/c++/7/functional:61,
                 from /home/jetson/mujoco_build/_deps/googletest-src/googlemock/include/gmock/gmock-actions.h:138,
                 from /home/jetson/mujoco_build/_deps/googletest-src/googlemock/include/gmock/gmock.h:59,
                 from /home/jetson/mujoco/test/engine/engine_sensor_test.cc:22:
/usr/include/c++/7/bits/stl_vector.h:411:2: note: candidate: template<class _Tp, class _Alloc, class _InputIterator, class> vector(_InputIterator, _InputIterator, const _Alloc&)-> std::vector<_Tp, _Alloc>
  vector(_InputIterator __first, _InputIterator __last,
  ^~~~~~
/usr/include/c++/7/bits/stl_vector.h:411:2: note:   template argument deduction/substitution failed:
/home/jetson/mujoco/test/engine/engine_sensor_test.cc:159:79: note:   couldn't deduce template parameter ‘_Tp’
   std::vector expected_values(data->sensordata, data->sensordata+nsensordata/2);
                                                                               ^
In file included from /usr/include/c++/7/vector:64:0,
                 from /usr/include/c++/7/functional:61,
                 from /home/jetson/mujoco_build/_deps/googletest-src/googlemock/include/gmock/gmock-actions.h:138,
                 from /home/jetson/mujoco_build/_deps/googletest-src/googlemock/include/gmock/gmock.h:59,
                 from /home/jetson/mujoco/test/engine/engine_sensor_test.cc:22:
/usr/include/c++/7/bits/stl_vector.h:383:7: note: candidate: template<class _Tp, class _Alloc> vector(std::initializer_list<_Tp>, const allocator_type&)-> std::vector<_Tp, _Alloc>
       vector(initializer_list<value_type> __l,
       ^~~~~~
/usr/include/c++/7/bits/stl_vector.h:383:7: note:   template argument deduction/substitution failed:
/home/jetson/mujoco/test/engine/engine_sensor_test.cc:159:79: note:   mismatched types ‘std::initializer_list<_Tp>’ and ‘mjtNum* {aka double*}’
   std::vector expected_values(data->sensordata, data->sensordata+nsensordata/2);
                                                                               ^
In file included from /usr/include/c++/7/vector:64:0,
                 from /usr/include/c++/7/functional:61,
                 from /home/jetson/mujoco_build/_deps/googletest-src/googlemock/include/gmock/gmock-actions.h:138,
                 from /home/jetson/mujoco_build/_deps/googletest-src/googlemock/include/gmock/gmock.h:59,
                 from /home/jetson/mujoco/test/engine/engine_sensor_test.cc:22:
/usr/include/c++/7/bits/stl_vector.h:358:7: note: candidate: template<class _Tp, class _Alloc> vector(std::vector<_Tp, _Alloc>&&, const allocator_type&)-> std::vector<_Tp, _Alloc>
       vector(vector&& __rv, const allocator_type& __m)
       ^~~~~~
/usr/include/c++/7/bits/stl_vector.h:358:7: note:   template argument deduction/substitution failed:
/home/jetson/mujoco/test/engine/engine_sensor_test.cc:159:79: note:   mismatched types ‘std::vector<_Tp, _Alloc>’ and ‘mjtNum* {aka double*}’
   std::vector expected_values(data->sensordata, data->sensordata+nsensordata/2);
                                                                               ^
In file included from /usr/include/c++/7/vector:64:0,
                 from /usr/include/c++/7/functional:61,
                 from /home/jetson/mujoco_build/_deps/googletest-src/googlemock/include/gmock/gmock-actions.h:138,
                 from /home/jetson/mujoco_build/_deps/googletest-src/googlemock/include/gmock/gmock.h:59,
                 from /home/jetson/mujoco/test/engine/engine_sensor_test.cc:22:
/usr/include/c++/7/bits/stl_vector.h:348:7: note: candidate: template<class _Tp, class _Alloc> vector(const std::vector<_Tp, _Alloc>&, const allocator_type&)-> std::vector<_Tp, _Alloc>
       vector(const vector& __x, const allocator_type& __a)
       ^~~~~~
/usr/include/c++/7/bits/stl_vector.h:348:7: note:   template argument deduction/substitution failed:
/home/jetson/mujoco/test/engine/engine_sensor_test.cc:159:79: note:   mismatched types ‘const std::vector<_Tp, _Alloc>’ and ‘mjtNum* {aka double*}’
   std::vector expected_values(data->sensordata, data->sensordata+nsensordata/2);
                                                                               ^
In file included from /usr/include/c++/7/vector:64:0,
                 from /usr/include/c++/7/functional:61,
                 from /home/jetson/mujoco_build/_deps/googletest-src/googlemock/include/gmock/gmock-actions.h:138,
[... cut ...]
                                                         ^
test/engine/CMakeFiles/engine_sensor_test.dir/build.make:75: recipe for target 'test/engine/CMakeFiles/engine_sensor_test.dir/engine_sensor_test.cc.o' failed
make[2]: *** [test/engine/CMakeFiles/engine_sensor_test.dir/engine_sensor_test.cc.o] Error 1
CMakeFiles/Makefile2:6032: recipe for target 'test/engine/CMakeFiles/engine_sensor_test.dir/all' failed
make[1]: *** [test/engine/CMakeFiles/engine_sensor_test.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2

pixar0407, 2022-07-06:
You are right, again.

For the second error, i resolve it after switching to gcc/++ version to 8 (maybe 7 would be enough).

Thank you for the reply and I learn something from you.
I will close this 👍

closed 2022-07-06 by pixar0407