google-deepmind/mujoco #375: Loading unbalanced inertias

AlexanderFabisch, 2022-07-08:
It is currently not possible to load long links because of the requirement of balanced inertia values. (see [here](https://github.com/deepmind/mujoco/blob/09a5efc09ecd19a439f92f3416c6b2ef0554c1c3/src/user/user_objects.cc#L728))

It would be nice to have an option to turn off this requirement. Automatic balancing of inertias does not work for me.

AlexanderFabisch, 2022-07-08:
Hi @yuvaltassa ,

I am starting the GUI `bin/simulate` from the latest release [2.2.0](https://github.com/deepmind/mujoco/releases/tag/2.2.0).

I want to load the URDF file `mia_hand_ros_pkgs/mia_hand_description/urdf/mia_hand.urdf` from [this repository](https://bitbucket.org/prensiliasrl/mia_hand_ros_pkgs/src/master/). The error that I get is `Error: inertia must satisfy A + B >= C; use 'balanceinertia' to fix / Object name = index_fle, id = 2`.

The error message makes sense because the inertia tag for `index_fle` is

```xml
<inertia ixx="1.486e-06" ixy="0" ixz="0" iyy="2e-07" iyz="0" izz="8.31e-07"/>
```

and ixx > iyy + izz: 1.486e-6 > 2e-7 + 8.31e-7.

kevinzakka, 2022-07-08:
@AlexanderFabisch Could you attach a zip containing the URDF? I'd like to check it out but don't want to go through bitbucket :)

AlexanderFabisch, 2022-07-08:
Here you go. 

[prensiliasrl-mia_hand_ros_pkgs-3e7e80b9ff52.zip](https://github.com/deepmind/mujoco/files/9073304/prensiliasrl-mia_hand_ros_pkgs-3e7e80b9ff52.zip)

Bitbucket is not totally different from Github though, since it's a public repository:

    git clone https://bitbucket.org/prensiliasrl/mia_hand_ros_pkgs.git

kevinzakka, 2022-07-08:
Good to have a trace here for posterity :) I'll check it out!

kevinzakka, 2022-07-08:
@AlexanderFabisch It loads fine for me with the following:

```xml
<mujoco>
  <compiler meshdir="../meshes" balanceinertia="true" discardvisual="false" strippath="false"/>
</mujoco>
```

Are you saying you don't want to have `balanceinertia="true"`?

kevinzakka, 2022-07-08:
@AlexanderFabisch Indeed this is a bug in MuJoCo, the expression to check the condition `OR`s multiple cases which fails even when A+B>=C, see [here](https://github.com/deepmind/mujoco/blob/main/src/user/user_objects.cc#L723).

cc @yuvaltassa

kevinzakka, 2022-07-08:
@AlexanderFabisch Could you explain what you mean by "automatic balancing does not work for me"? If I turn it on, it loads fine but the inertia boxes look different than the CAD model you uploaded.

[image]

AlexanderFabisch, 2022-07-08:
> Could you explain what you mean by "automatic balancing does not work for me"? If I turn it on, it loads fine but the inertia boxes look different than the CAD model you uploaded.

[This line](https://github.com/deepmind/mujoco/blob/09a5efc09ecd19a439f92f3416c6b2ef0554c1c3/src/user/user_objects.cc#L727) looks like it averages the main diagonal of the inertia matrix. This is not a behavior that I would like to have. However, the boxes look fine. It seems seems like only one irrelevant link is affected.

Thanks for the reference. I get it now. Apparently, I didn't know much about inertia matrices. :) Thanks for your quick reply!

I guess I can close the issue and will look into how to correct the inertia values in the URDF.

closed 2022-07-08 by AlexanderFabisch