google-deepmind/mujoco #379: Python bindings won't compile - complaints of #pragma

scottastone, 2022-07-13:
Hi there

I'm trying to use MuJoCo on a cluster (Compute Canada if it makes any difference). I am trying to compile the Python bindings on MuJoCo 2.2.0. Amazingly they don't have the precompiled bindings on the server, but they do have the MuJoCo binaries.

`gcc` is 11.3.0 

My workflow is as follows:
```
$ module load StdEnv/2020 # note this just loads the compiler, etc
$ module load mujoco/2.2.0
$ export MUJOCO_PATH=/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/mujoco/2.2.0
$ wget https://github.com/deepmind/mujoco/archive/refs/tags/2.2.0.zip
$ unzip 2.2.0.zip
$ cd mujoco-2.2.0/python
$ virtualenv tmp/env
$ source tmp/env/bin/activate
$ ./make_sdist.sh
$ pip wheel mujoco-2.2.0.tar.gz
```

I came across this [issue](https://github.com/deepmind/mujoco/issues/354), but it wasn't obvious how @berniekirby solved the issue. I've tried removing everything, etc. I'm just trying to get this working with `dm_control`. Any help would be appreciated.

I end up getting a bunch of output like below:

<details>
```
     < ... lots cut >

      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:298:1: error: ‘#pragma’ is not allowed here
        298 | MJMODEL_SITE
            | ^~~
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSiteViews::bodyid()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:298:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSiteViews::matid()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:298:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSiteViews::group()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:298:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<unsigned char> mujoco::python::MjModelSiteViews::sameframe()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:298:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<double> mujoco::python::MjModelSiteViews::size()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:298:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<double> mujoco::python::MjModelSiteViews::pos()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:298:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<double> mujoco::python::MjModelSiteViews::quat()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:298:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<double> mujoco::python::MjModelSiteViews::user()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:298:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<float> mujoco::python::MjModelSiteViews::rgba()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:298:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSkinViews::matid()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:302:1: error: ‘#pragma’ is not allowed here
        302 | MJMODEL_SKIN
            | ^~~
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<float> mujoco::python::MjModelSkinViews::rgba()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:302:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<float> mujoco::python::MjModelSkinViews::inflate()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:302:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSkinViews::vertadr()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:302:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSkinViews::vertnum()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:302:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSkinViews::texcoordadr()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:302:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSkinViews::faceadr()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:302:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSkinViews::facenum()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:302:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSkinViews::boneadr()’:
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc:302:1: error: ‘#pragma’ is not allowed here
      /tmp/pip-req-build-y63uhm9z/mujoco/indexers.cc: In member function ‘pybind11::array_t<int> mujoco::python::MjModelSkinViews::bonenum()’:
[... 278 lines cut ...]
          cmd_obj.run()
        File "/project/6029978/stone/Work/mujoco-2.2.0/python/tmp/env/lib/python3.9/site-packages/setuptools/command/build.py", line 24, in run
          super().run()
        File "/project/6029978/stone/Work/mujoco-2.2.0/python/tmp/env/lib/python3.9/site-packages/setuptools/_distutils/command/build.py", line 131, in run
          self.run_command(cmd_name)
        File "/project/6029978/stone/Work/mujoco-2.2.0/python/tmp/env/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 317, in run_command
          self.distribution.run_command(command)
        File "/project/6029978/stone/Work/mujoco-2.2.0/python/tmp/env/lib/python3.9/site-packages/setuptools/dist.py", line 1217, in run_command
          super().run_command(command)
        File "/project/6029978/stone/Work/mujoco-2.2.0/python/tmp/env/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 987, in run_command
          cmd_obj.run()
        File "/tmp/pip-req-build-y63uhm9z/setup.py", line 144, in run
          self._configure_cmake()
        File "/tmp/pip-req-build-y63uhm9z/setup.py", line 243, in _configure_cmake
          subprocess.check_call(
        File "/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/python/3.9.6/lib/python3.9/subprocess.py", line 373, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['cmake', '--build', '.', '-j64', '--config', 'Release']' returned non-zero exit status 2.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mujoco

```
</details>

berniekirby, 2022-07-13:
Hi, this is my environment: 

Ended up using gcc/9.1.0 Needs to use gcc >8

module load gcc/9.1.0 binutils/2.28 cmake/3.21.1 git python/3.7.7

 

Something is making part of cmake use the default /usr/bin/gcc which is 4.4 and so add this:

export CC=gcc

export CXX=g++

(Can’t be bothered to figure out why. Seems common on this system. By setting CC and CXX as above it should just use the one in the PATH).

 

This is just convenience in our environment.

export LDFLAGS="$LDFLAGS -Wl,-rpath,/usr/local/gcc/9.1.0/lib64"

 

Also I git cloned the repo (didn’t down load v2.2.0.tar.gz)

git clone https://github.com/deepmind/mujoco.git

 

From: Scott Stone ***@***.***> 
Sent: Wednesday, 13 July 2022 1:25 PM
To: deepmind/mujoco ***@***.***>
Cc: berniekirby ***@***.***>; Mention ***@***.***>
Subject: [deepmind/mujoco] Python bindings won't compile - complaints of #pragma (Issue #379)

 

Hi there

I'm trying to use MuJoCo on a cluster (Compute Canada if it makes any difference). I am trying to compile the Python bindings on MuJoCo 2.2.0. Amazingly they don't have the precompiled bindings on the server, but they do have the MuJoCo binaries.

gcc is 11.3.0

My workflow is as follows:

$ module load StdEnv/2020 # note this just loads the compiler, etc
 
$ module load mujoco/2.2.0
 
$ export MUJOCO_PATH=/cvmfs/soft.computecanada.ca/easybuild/software/2020/avx2/Core/mujoco/2.2.0
 
$ wget https://github.com/deepmind/mujoco/archive/refs/tags/2.2.0.zip
 
$ unzip 2.2.0.zip
 
$ cd mujoco-2.2.0/python
 
$ virtualenv tmp/env
 
$ source tmp/env/bin/activate
 
$ ./make_sdist.sh
 
$ pip wheel mujoco-2.2.0.tar.gz
 

I came across this issue <https://github.com/deepmind/mujoco/issues/354> , but it wasn't obvious how @berniekirby <https://github.com/berniekirby>  solved the issue. I've tried removing everything, etc. I'm just trying to get this working with dm_control. Any help would be appreciated.

I end up getting a bunch of output like below:

….. lots deleted.

scottastone, 2022-07-13:
Thanks for the reply and suggestions @berniekirby. I tried adding `export CC=gcc` and `export CXX=g++` (as on this cluster, the default `/usr/bin/gcc` is 4.8.5) to no avail. Seems to just produce the same error. 

I actually also have a local machine that I've tried this build process on (running Ubuntu 22.04 & gcc 11.2.0) and when compiling I actually get the same error. So I am scratching my head even more.

I'll keep plugging away and see if something works. If someone from the MuJoCo team sees an obvious flaw, I'd super appreciate it!

saran-t, 2022-07-13:
@scottastone Are you not able to install via `pip`?

scottastone, 2022-07-13:
@saran-t Even if I try to install via `pip` I still get the build error. Worse though, if I deploy the job the nodes do not have access to the internet. My solution was if I could at least build the wheel then I could deploy it with the job so it could install the local copy.

scottastone, 2022-07-13:
Just another update

I figured out the OS is CentOS 7, which _should_ support `manylinux2014` but doesn't seem to.
```
$ cat /etc/os-release
NAME="CentOS Linux"
VERSION="7 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="7"
PRETTY_NAME="CentOS Linux 7 (Core)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:7"
HOME_URL="https://www.centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"

CENTOS_MANTISBT_PROJECT="CentOS-7"
CENTOS_MANTISBT_PROJECT_VERSION="7"
REDHAT_SUPPORT_PRODUCT="centos"
REDHAT_SUPPORT_PRODUCT_VERSION="7"
```

Thought I'd try building on a different machine and deploying the wheel, but no success yet:
```
$ pip install mujoco-2.2.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl
Ignoring pip: markers 'python_version < "3"' don't match your environment
ERROR: mujoco-2.2.0-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl is not a supported wheel on this platform.
```

saran-t, 2022-07-14:
Could you please paste the full output of `pip debug --verbose 2> /dev/null | grep cp3` ?

scottastone, 2022-07-14:
```
$ pip debug --verbose 2> /dev/null | grep cp3
  cp39-cp39-linux_x86_64
  cp39-abi3-linux_x86_64
  cp39-none-linux_x86_64
  cp38-abi3-linux_x86_64
  cp37-abi3-linux_x86_64
  cp36-abi3-linux_x86_64
  cp35-abi3-linux_x86_64
  cp34-abi3-linux_x86_64
  cp33-abi3-linux_x86_64
  cp32-abi3-linux_x86_64
  cp39-none-any
```

saran-t, 2022-07-15:
This is very strange... somehow pip decides that your system doesn't conform to _any_ of the `manylinux` requirement, not even the oldest `manylinux1`, which means that you won't be able to install anything via prebuilt wheels.

Is there some way that I can make a VM with the same environment as what you're using?

saran-t, 2022-07-15:
To try and answer you original question: the pragma in question most definitely comes from https://github.com/deepmind/mujoco/blob/main/python/mujoco/util/crossplatform.h#L54

As you can see that pragma is supposed to only be there when the `__clang__` macro is defined, which implies that for some reason your GCC is pretending to be Clang.

scottastone, 2022-07-16:
Very interesting - I can try to get in contact with the cluster engineers to get a better understanding of what is going on, but they are quite slow to help. If you think it would be worthwhile, I'm happy to grant you access to the cluster as I'm not sure of the best way to create an effective VM clone.

scottastone, 2022-07-16:
Update:

I was able to successfully build mujoco by using clang `13.0.1`. However, when I go to `import mujoco` in python, I get the following error with glfw:

```
(env) [stone@cedar1 dist]$ python
Python 3.9.6 (default, Jul 12 2021, 18:23:59)
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/project/6029978/stone/work/mujoco-2.2.0/python/tmp/env/lib/python3.9/site-packages/mujoco/__init__.py", line 54, in <module>
    from mujoco.glfw import GLContext
  File "/project/6029978/stone/work/mujoco-2.2.0/python/tmp/env/lib/python3.9/site-packages/mujoco/glfw/__init__.py", line 17, in <module>
    import glfw
  File "/project/6029978/stone/work/mujoco-2.2.0/python/tmp/env/lib/python3.9/site-packages/glfw/__init__.py", line 43, in <module>
    raise ImportError("Failed to load GLFW3 shared library.")
ImportError: Failed to load GLFW3 shared library.
```

I have tried to `module load glfw/3.3.2` and it didn't seem to make a difference, and I also tried to [build from source](https://www.glfw.org/docs/latest/compile.html) but have not been successful yet, but maybe this isn't the rabbit hole to be going down.

saran-t, 2022-07-16:
What's the GPU installed on your cluster? If it's Nvidia then you're better off with EGL: set the environment variable `MUJOCO_GL=egl` and see if that works.

Alternatively, if you don't need rendering (or just want to test that your base installation works) then set `MUJOCO_GL=off`.

Did you ever figure out what happened with GCC pretending to be Clang? I'm curious to find out the root cause there.

scottastone, 2022-07-16:
There is no GPU (as far as I know)- our model we are running is going to be A3C, so we're just running CPU-only nodes. And we're fine to go renderless.

Just to throw another wrinkle in - the entire reason I am trying to do this is because we are actually using `dm_control` for our environment - which at the top of our script we put in a `from dm_control import mujoco`

Now, when I run the above, it gives me this fun error, suggesting that I cannot set `MUJCOO_GL` to `off`.

```
>>> from dm_control import mujoco
Traceback (most recent call last):
  File "/project/6029978/stone/work/env/lib/python3.9/site-packages/dm_control/_render/__init__.py", line 60, in <module>
    import_func = _ALL_RENDERERS[BACKEND]
KeyError: 'off'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/project/6029978/stone/work/env/lib/python3.9/site-packages/dm_control/mujoco/__init__.py", line 18, in <module>
    from dm_control.mujoco.engine import action_spec
  File "/project/6029978/stone/work/env/lib/python3.9/site-packages/dm_control/mujoco/engine.py", line 41, in <module>
    from dm_control import _render
  File "/project/6029978/stone/work/env/lib/python3.9/site-packages/dm_control/_render/__init__.py", line 62, in <module>
    raise RuntimeError(
RuntimeError: Environment variable MUJOCO_GL must be one of odict_keys(['glfw', 'egl', 'osmesa']): got 'off'.
```

Also, there seems to be a difference between `mujoco` and the one wrapped by `dm_control` --- there is no class called `mujoco.Physics` in regular mujoco, but there is in the `dm_control` version.

However, if I just do a simple `import mujoco` it no longer complains. Sigh.

saran-t, 2022-07-16:
This is a `dm_control` bug, I'll fix that for our next release. Sorry about that.

For the time being, if you're able to install OSMesa, the easiest way to proceed is to install it and set `MUJOCO_GL=osmesa`.

scottastone, 2022-07-16:
Awesome, that seems to have fixed it. I am now able to load and run our code no problems (sans rendering, but that's ok).

I'm wondering if I should leave this issue open though, because there is something that seems to trick `gcc` into thinking it's `clang`. A decent work around is to just use `clang`, but that might not work for everyone. I can do a bit of a deeper dive and see if I can figure out what's going on

saran-t, 2022-07-16:
With OSMesa you'd actually be able to render correctly as well, just very slowly (since everything is done on the CPU).

I've never actually seen GCC pretending to be Clang before, so it might be something specific to your cluster. Perhaps it's worth asking your sysadmin?

scottastone, 2022-07-16:
I will ask the sysadmin about it, but I am also getting it on a local Ubuntu machine when I try to compile the bindings, so there may be an issue in the CMake files

saran-t, 2022-07-16:
You can see our GCC 11 build logs on GH Actions [here](https://github.com/deepmind/mujoco/runs/7371307356?check_suite_focus=true) for example.

If you're interested, you can fork this repo, and edit [build.yml](https://github.com/deepmind/mujoco/blob/main/.github/workflows/build.yml) to use the same GCC and Ubuntu versions  as (or as close as possible to) your machine to see if you can repro the issue that way. If you can, that'll give us a full log to look at.

saran-t, 2022-07-18:
Actually, the easiest thing to do would be to add `-v` to your `pip wheel` call. I really just want to see the first couple of lines from CMake (compiler identification lines).

scottastone, 2022-07-18:
I assume this is what you're looking for:

I'm running this on my local machine (Ubuntu 22.04) using Python 3.10.4
`gcc --version` output:
```
gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
(env) scott@frankenstein:~/code/sources/tmp/mujoco/python/dist$ pip wheel mujoco-2.2.0.tar.gz -v
Processing ./mujoco-2.2.0.tar.gz
  File was already downloaded /home/scott/code/sources/tmp/mujoco/python/dist/mujoco-2.2.0.tar.gz
  Running command python setup.py egg_info
  running egg_info
  creating /tmp/pip-pip-egg-info-22qpycqy/mujoco.egg-info
  writing /tmp/pip-pip-egg-info-22qpycqy/mujoco.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-pip-egg-info-22qpycqy/mujoco.egg-info/dependency_links.txt
  writing requirements to /tmp/pip-pip-egg-info-22qpycqy/mujoco.egg-info/requires.txt
  writing top-level names to /tmp/pip-pip-egg-info-22qpycqy/mujoco.egg-info/top_level.txt
  writing manifest file '/tmp/pip-pip-egg-info-22qpycqy/mujoco.egg-info/SOURCES.txt'
  reading manifest file '/tmp/pip-pip-egg-info-22qpycqy/mujoco.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  adding license file 'LICENSE'
  writing manifest file '/tmp/pip-pip-egg-info-22qpycqy/mujoco.egg-info/SOURCES.txt'
  Preparing metadata (setup.py) ... done
Collecting absl-py
  Using cached absl_py-1.1.0-py3-none-any.whl (123 kB)
Collecting glfw
  Using cached glfw-2.5.3-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38-none-manylinux2014_x86_64.whl (206 kB)
Collecting numpy
  Using cached numpy-1.23.1-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (17.0 MB)
Collecting pyopengl
  Using cached PyOpenGL-3.1.6-py3-none-any.whl (2.4 MB)
Saved ./absl_py-1.1.0-py3-none-any.whl
Saved ./glfw-2.5.3-py2.py27.py3.py30.py31.py32.py33.py34.py35.py36.py37.py38-none-manylinux2014_x86_64.whl
Saved ./numpy-1.23.1-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl
Saved ./PyOpenGL-3.1.6-py3-none-any.whl
Building wheels for collected packages: mujoco
  Running command python setup.py bdist_wheel
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-cpython-310
  creating build/lib.linux-x86_64-cpython-310/mujoco
  copying mujoco/bindings_test.py -> build/lib.linux-x86_64-cpython-310/mujoco
  copying mujoco/rollout_test.py -> build/lib.linux-x86_64-cpython-310/mujoco
  copying mujoco/__init__.py -> build/lib.linux-x86_64-cpython-310/mujoco
  copying mujoco/rollout.py -> build/lib.linux-x86_64-cpython-310/mujoco
  copying mujoco/render_test.py -> build/lib.linux-x86_64-cpython-310/mujoco
  creating build/lib.linux-x86_64-cpython-310/mujoco/glfw
  copying mujoco/glfw/__init__.py -> build/lib.linux-x86_64-cpython-310/mujoco/glfw
  creating build/lib.linux-x86_64-cpython-310/mujoco/osmesa
  copying mujoco/osmesa/__init__.py -> build/lib.linux-x86_64-cpython-310/mujoco/osmesa
  creating build/lib.linux-x86_64-cpython-310/mujoco/egl
  copying mujoco/egl/__init__.py -> build/lib.linux-x86_64-cpython-310/mujoco/egl
  copying mujoco/egl/egl_ext.py -> build/lib.linux-x86_64-cpython-310/mujoco/egl
  running build_ext
  Configuring CMake with the following arguments:
      -DPython3_ROOT_DIR:PATH=/home/scott/code/sources/tmp/mujoco/python/tmp/env
      -DPython3_EXECUTABLE:STRING=/home/scott/code/sources/tmp/mujoco/python/tmp/env/bin/python
      -DCMAKE_MODULE_PATH:PATH=/tmp/pip-req-build-79j72h9v/cmake
      -DCMAKE_BUILD_TYPE:STRING=Release
      -DCMAKE_LIBRARY_OUTPUT_DIRECTORY:PATH=build/temp.linux-x86_64-cpython-310
      -DCMAKE_INTERPROCEDURAL_OPTIMIZATION=:BOOLON
      -DCMAKE_Fortran_COMPILER:STRING=
      -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON
[... 147 lines cut ...]
  --
  -- mujoco::FindOrFetch: Using FetchContent to retrieve `eigen` - Done
  -- mujoco::FindOrFetch: checking for targets in package `pybind11`
  -- mujoco::FindOrFetch: checking for targets in package `pybind11` - target `pybind11::pybind11_headers` not defined.
  -- mujoco::FindOrFetch: Using FetchContent to retrieve `pybind11`
  -- pybind11 v2.10.0 dev1
  -- Performing Test HAS_FLTO
  -- Performing Test HAS_FLTO - Success
  -- mujoco::FindOrFetch: Using FetchContent to retrieve `pybind11` - Done
  -- Performing Test CAN_BUILD_AVX
  -- Performing Test CAN_BUILD_AVX - Success
  -- Configuring done
  -- Generating done
  -- Build files have been written to: /tmp/pip-req-build-79j72h9v/build/temp.linux-x86_64-cpython-310
  Building all extensions with CMake
  /usr/bin/cmake -S/tmp/pip-req-build-79j72h9v/mujoco -B/tmp/pip-req-build-79j72h9v/build/temp.linux-x86_64-cpython-310 --check-build-system CMakeFiles/Makefile.cmake 0
  /usr/bin/cmake -E cmake_progress_start /tmp/pip-req-build-79j72h9v/build/temp.linux-x86_64-cpython-310/CMakeFiles /tmp/pip-req-build-79j72h9v/build/temp.linux-x86_64-cpython-310//CMakeFiles/progress.marks
  /usr/bin/gmake  -f CMakeFiles/Makefile2 all
  gmake[1]: Entering directory '/tmp/pip-req-build-79j72h9v/build/temp.linux-x86_64-cpython-310'
  /usr/bin/gmake  -f _deps/abseil-cpp-build/absl/base/CMakeFiles/absl_log_severity.dir/build.make _deps/abseil-cpp-build/absl/base/CMakeFiles/absl_log_severity.dir/depend
  /usr/bin/gmake  -f _deps/abseil-cpp-build/absl/base/CMakeFiles/absl_spinlock_wait.dir/build.make _deps/abseil-cpp-build/absl/base/CMakeFiles/absl_spinlock_wait.dir/depend
[... cut ...] /abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/build.make _deps/abseil-cpp-build/absl/time/CMakeFiles/absl_time_zone.dir/depend
  /usr/bin/gmake  -f _deps/abseil-cpp-build/absl/numeric/CMakeFiles/absl_int128.dir/build.make _deps/abseil-cpp-build/absl/numeric/CMakeFiles/absl_int128.dir/depend
  /usr/bin/gmake  -f _deps/abseil-cpp-build/absl/profiling/CMakeFiles/absl_exponential_biased.dir/build.make _deps/abseil-cpp-build/absl/profiling/CMakeFiles/absl_exponential_biased.dir/depend
```

saran-t, 2022-08-01:
I'm going to close this issue since you've found a working solution and we've since added multiple GCC builds to our CI setup and they all seem fine.

closed 2022-08-01 by saran-t