google-deepmind/mujoco #382: CMake configuration breaks if building with MUJOCO_BUILD_TESTS=OFF

wpumacay, 2022-07-14:
Hi there.

I'm running into this issue when running the `CMake` configuration step, while using `-DMUJOCO_BUILD_TESTS=OFF`. It seems that tests are still requested for a python util via `MUJOCO_TEST_PYTHON_UTIL`, which enables testing, but it doesn't include `googletest`, therefore throwing an error during configuration, as shown in the screenshot below:

```bash
# CMake configuration with tests disabled
cmake -B build -DMUJOCO_BUILD_TESTS=OFF
```

[image]

Notice that [here][0] `GoogleTest` is being included, but it's not included [here][1] for that python util (yet, `gtest_add_test` is being called without having `GoogleTest` included)

A fix would be to check for `MUJOCO_BUILD_TESTS` also on the following line on the root [`CMakeLists.txt`][2]:

[image]

- **OS:** `Arch Linux`
- **MuJoCo version:** `2.2.0`

[0]: <https://github.com/deepmind/mujoco/blob/af12322e9d6fe1b5a7b59fe6eb4dc3f790f35580/test/CMakeLists.txt#L18> (tests-cmakefile)
[1]: <https://github.com/deepmind/mujoco/blob/af12322e9d6fe1b5a7b59fe6eb4dc3f790f35580/python/mujoco/util/CMakeLists.txt#L61> (tests-python-util-cmakefile)
[2]: <https://github.com/deepmind/mujoco/blob/af12322e9d6fe1b5a7b59fe6eb4dc3f790f35580/CMakeLists.txt#L164> (root-cmakefile)

saran-t, 2022-07-15:
I'd prefer the following

```
if(BUILD_TESTING AND MUJOCO_BUILD_TESTS)
  enable_testing()
  add_subdirectory(test)
  if(MUJOCO_TEST_PYTHON_UTIL)
    add_subdirectory(python/mujoco/util)
  endif()
endif()
```

Can you see if that works for you? If it does, feel free to send us a PR, I'll get it merged quickly. Thanks!

wpumacay, 2022-07-16:
Hi @saran-t sorry about the delay. I've just created a small PR for this fix :+1: . Everything works as expected :). Originally I also intended to exclude examples, and it also works:

```bash
cmake -B build -DMUJOCO_BUILD_TESTS=OFF -DMUJOCO_BUILD_EXAMPLES=OFF
```
:+1:

PR #385 (2022-07-16): Fixes #382 (compile with unittests disabled)
Small change to the `CMake` rules to fix #382
commit 7ee01ddc (2022-07-16): Fixes #382 (compile with unittests disabled)
closed 2022-07-16 by copybara-service[bot]