google-deepmind/mujoco #386: Compile error from source of "safeAddToBufferSize"

ElonKou, 2022-07-17:
**Compile error from source in function "safeAddToBufferSize"**

Hi,

I'm a engineer and I'm trying to build MuJoCo from source, and compile failed.

OS: ubuntu 20.04
compile : gcc 9.4.0
mujoco version : 220

After clone the repo, I run the command  "cmake .." in my build dir, and install some dependences by "sudo apt-get install libxinerama-dev libxcursor-dev libxi-dev", the "make" command the output error information as "compile error information" says.

"cmake .." output the information are list in "cmake information"

<details>
  <summary>compile error information </summary>

```bash
[  1%] Building C object _deps/ccd-build/src/CMakeFiles/ccd.dir/ccd.c.o
[  1%] Building C object _deps/ccd-build/src/CMakeFiles/ccd.dir/mpr.c.o
[  1%] Building C object _deps/ccd-build/src/CMakeFiles/ccd.dir/polytope.c.o
[  1%] Building C object _deps/ccd-build/src/CMakeFiles/ccd.dir/support.c.o
[  1%] Building C object _deps/ccd-build/src/CMakeFiles/ccd.dir/vec3.c.o
[  3%] Linking C static library ../../../lib/libccd.a
[  3%] Built target ccd
[  4%] Building CXX object CMakeFiles/lodepng.dir/_deps/lodepng-src/lodepng.cpp.o
[  4%] Linking CXX static library lib/liblodepng.a
[  4%] Built target lodepng
[  6%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/global_r.c.o
[  6%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/stat_r.c.o
[  6%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/geom2_r.c.o
[  6%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/poly2_r.c.o
[  6%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/merge_r.c.o
[  8%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/libqhull_r.c.o
[  8%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/geom_r.c.o
[  8%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/poly_r.c.o
[  8%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/qset_r.c.o
[  8%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/mem_r.c.o
[  9%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/random_r.c.o
[  9%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/usermem_r.c.o
[  9%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/userprintf_r.c.o
[  9%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/io_r.c.o
[  9%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/user_r.c.o
[ 11%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/accessors_r.c.o
[ 11%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/rboxlib_r.c.o
[ 11%] Building C object _deps/qhull-build/CMakeFiles/qhullstatic_r.dir/src/libqhull_r/userprintf_rbox_r.c.o
[ 11%] Linking C static library ../../lib/libqhullstatic_r.a
[ 11%] Built target qhullstatic_r
[ 11%] Building CXX object _deps/tinyxml2-build/CMakeFiles/tinyxml2.dir/tinyxml2.cpp.o
[ 12%] Linking CXX static library ../../lib/libtinyxml2.a
[ 12%] Built target tinyxml2
[ 12%] Building CXX object _deps/tinyobjloader-build/CMakeFiles/tinyobjloader.dir/tiny_obj_loader.cc.o
[ 12%] Linking CXX static library ../../lib/libtinyobjloader.a
[ 12%] Built target tinyobjloader
[ 12%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_callback.c.o
[ 12%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_collision_box.c.o
[ 12%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_collision_convex.c.o
[ 14%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_collision_driver.c.o
[ 14%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_collision_primitive.c.o
[ 14%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_core_constraint.c.o
[ 14%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_core_smooth.c.o
[ 14%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_derivative.c.o
[ 14%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_file.c.o
[ 16%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_forward.c.o
[ 16%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_inverse.c.o
[ 16%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_io.c.o
/home/elonkou/ELONKOU/mujoco/src/engine/engine_io.c: In function ‘safeAddToBufferSize’:
/home/elonkou/ELONKOU/mujoco/src/engine/engine_io.c:361:21: error: missing binary operator before token "("
  361 |     && __has_builtin(__builtin_add_overflow) && __has_builtin(__builtin_mul_overflow)
      |                     ^
make[2]: *** [CMakeFiles/mujoco.dir/build.make:206: CMakeFiles/mujoco.dir/src/engine/engine_io.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1269: CMakeFiles/mujoco.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

```
</details>

<details>
  <summary>cmake information </summary>

```bash
CHECK_STARTmujoco::FindOrFetch: checking for targets in package `qhull`
CHECK_FAILtarget `qhull` not defined.
CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve `qhull`
-- 
-- ========== qhull Build Information ==========
-- Build Version:                             8.1-alpha1
-- Install Prefix (CMAKE_INSTALL_PREFIX):     /usr/local
-- Binary Directory (BIN_INSTALL_DIR):        bin
-- Library Directory (LIB_INSTALL_DIR):       lib
-- Include Directory (INCLUDE_INSTALL_DIR):   include
-- Documentation Directory (DOC_INSTALL_DIR): share/doc/qhull
-- Man Pages Directory (MAN_INSTALL_DIR):     share/man/man1
-- Build Type (CMAKE_BUILD_TYPE):             Release
-- Build static libraries:                    ON
-- Build shared library:                      OFF
-- Use shared library for linking apps:       OFF
-- To override these options, add -D{OPTION_NAME}=... to the cmake command
--   Build the debug targets                  -DCMAKE_BUILD_TYPE=Debug
-- 
-- To build and install qhull, enter "make" and "make install"
-- To smoketest qhull, enter "ctest"
-- 
CHECK_PASSDone
CHECK_STARTmujoco::FindOrFetch: checking for targets in package `tinyxml2`
CHECK_FAILtarget `tinyxml2` not defined.
CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve `tinyxml2`
CHECK_PASSDone
CHECK_STARTmujoco::FindOrFetch: checking for targets in package `tinyobjloader`
CHECK_FAILtarget `tinyobjloader` not defined.
CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve `tinyobjloader`
CHECK_PASSDone
CHECK_STARTmujoco::FindOrFetch: checking for targets in package `ccd`
CHECK_FAILtarget `ccd` not defined.
CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve `ccd`
CMake Warning (dev) at build/_deps/ccd-src/CMakeLists.txt:9 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
    PROJECT_VERSION_MAJOR
    PROJECT_VERSION_MINOR
    PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

CHECK_PASSDone
CHECK_STARTmujoco::FindOrFetch: checking for targets in package `absl`
CHECK_FAILtarget `absl::core_headers` not defined.
CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve `abseil-cpp`
CHECK_PASSDone
CHECK_STARTmujoco::FindOrFetch: checking for targets in package `GTest`
CHECK_FAILtarget `gtest` not defined.
CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve `googletest`
CHECK_PASSDone
CHECK_STARTmujoco::FindOrFetch: checking for targets in package `benchmark`
CHECK_FAILtarget `benchmark::benchmark` not defined.
CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve `benchmark`
-- Failed to find LLVM FileCheck
-- git version: v1.6.1 normalized to 1.6.1
-- Version: 1.6.1
-- Performing Test HAVE_STD_REGEX -- success
-- Performing Test HAVE_GNU_POSIX_REGEX -- failed to compile
-- Performing Test HAVE_POSIX_REGEX -- success
-- Performing Test HAVE_STEADY_CLOCK -- success
CHECK_PASSDone
CHECK_STARTmujoco::FindOrFetch: checking for targets in package `mujoco`
CHECK_PASSfound
CHECK_STARTmujoco::FindOrFetch: checking for targets in package `glfw`
CHECK_FAILtarget `glfw` not defined.
CHECK_STARTmujoco::FindOrFetch: Using FetchContent to retrieve `glfw`
-- Using X11 for window creation
CHECK_PASSDone
-- Configuring done
-- Generating done
-- Build files have been written to: /home/elonkou/ELONKOU/mujoco/build

```

</details>

could you please help me to solve this question, thanks very much.

nimrod-gileadi, 2022-07-17:
Thanks. Looks like __has_builtin was only added to GCC in version 10 ([changelog](https://gcc.gnu.org/gcc-10/changes.html)), and our continuous integration is using GCC 11.

I'll fix this soon, but for now, is it feasible for you to upgrade to a more recent version of GCC?

ElonKou, 2022-07-17:
> Thanks. Looks like __has_builtin was only added to GCC in version 10 ([changelog](https://gcc.gnu.org/gcc-10/changes.html)), and our continuous integration is using GCC 11.
> 
> I'll fix this soon, but for now, is it feasible for you to upgrade to a more recent version of GCC?

@nimrod-gileadi Thanks a lot for your help, I solved the problem by upgrading to gcc-11. :)

closed 2022-07-17 by copybara-service[bot]