google-deepmind/mujoco #393: How to modelling the round hole precisely

flilqhu, 2022-07-20:
Hi,

I'm a student and I'm trying to use MuJoCo for peg-insertion task.

I'm looking for some help with round hole modelling.

In the documentation I see that
"
In order to model a non-convex object other than a height field, the user must decompose it into a union of convex geoms (which can be primitive shapes or meshes) and attach them to the same body. Tools such as the [HACD](https://github.com/kmammou/v-hacd) library can be used outside MuJoCo to automate this process.
"
However I found that the round holes can not be constructed precisely using VHACD.
Could you tell me the solution to get some accurate round peg in hole environments.
And how to  modelling precisely when I need to consider the collision during the precise operation.

Thank you.

yuvaltassa, 2022-07-20:
Also, take a look at the collision geometry of the [cup model](https://github.com/deepmind/mujoco/blob/main/model/mug/mug.xml)

closed 2022-07-20 by yuvaltassa

walt666, 2024-09-23:
> Tuning V-HACD parameters is indeed not straightforward. I found [this video](https://www.youtube.com/watch?v=Sa1NE-St8iA) from one of the developers helpful.
Hi，I have the same problem. But the video link won't open, I can't see the video.Could you send me the new link again, thank you!

kevinzakka, 2024-09-23:
@walt666 Looks like the video has been made private. I would recommend you try out [CoACD](https://github.com/SarahWeiii/CoACD).

yuvaltassa, 2024-09-23:
Or use `<replicate>` https://www.youtube.com/watch?v=5k0_wsIRAFc

walt666, 2024-09-24:
> @walt666 Looks like the video has been made private. I would recommend you try out [CoACD](https://github.com/SarahWeiii/CoACD).
Thank you very much！It's very useful to me. 😄