google-deepmind/mujoco #394: Example using Python bindings to visualize Mujoco scene

studywolf, 2022-07-20:
Hi,

I'm a researcher and I'm trying to use MuJoCo for training a neuromorphic controller.

I'm looking for some help with visualizing a simulation using the new Mujoco Python bindings. 

I previously used `mujoco-py` to do the interfacing with Mujoco, but I'm interested to update to the latest Mujoco version. There is the render_test.py example in the repo, but that doesn't actually show the full use case of visualizing a scene. I've scoured the documentation but not found a full description there either. I don't have a lot of familiarity with GLFW, so I'm hoping that someone can just whip up a quick python script that shows visualization of one of the default models. I'll keep working on it, but this would really help me out! (and I think be a useful example to have explicitly in the documentation)
Thanks!

yuvaltassa, 2022-07-20:
Take a look at our colabs (linked in the repo README). They use `dm_control` for rendering from Python.

We also intend to add some example Python code with no dependency on `dm_control`, but they would be a subset of want you can find in `dm_control`.

studywolf, 2022-07-20:
Ah, okeydoke. Is there a time line on that? My goal was to remove dependencies! :D Looking through the Colab and then the `dm_control` documentation it seems like the rendering is buried pretty deeply in there, and not something I'll be able to pull out quickly.

yuvaltassa, 2022-07-20:
Actually we already have a colab that just needs cleaning up and maybe some of the rendering stuff pulled into library functions. I'll try and see if we we can either finish that soon or if not maybe we can find a quick way to share it with you.
I hope to update here in the next few days.

rohanpsingh, 2022-07-22:
Hi @studywolf perhaps [this renderer I developed](https://github.com/rohanpsingh/mujoco-python-viewer) could be of use to you :)

studywolf, 2022-07-22:
Ah, great, thanks! This is much better than building out my own while waiting for the updates!

studywolf, 2022-09-15:
@yuvaltassa any updates?

yuvaltassa, 2022-09-20:
Hi @studywolf,

Thanks for the reminder. Yes, you can see a draft proposal of a new `Renderer` class in the LQR colab notebook we [just released](https://mujoco.readthedocs.io/en/latest/changelog.html). @rohanpsingh, FYI.

I hope to finish converting the old `dm_control`-based [tutorial notebook](https://github.com/deepmind/dm_control/blob/main/dm_control/mujoco/tutorial.ipynb) to native bindings soon as well.

If you wish to contribute, you might consider submitting this yourself :slightly_smiling_face:

closed 2022-09-20 by yuvaltassa

rohanpsingh, 2022-09-20:
@yuvaltassa 
I think link to the colab is broken: https://github.com/deepmind/mujoco/compare/main...rohanpsingh:mujoco:patch-2