google-deepmind/mujoco #395: Mujoco - Unity Plugin Crash - 2.2.1 on 2020.3

michelhabib, 2022-07-22:
I downloaded the latest 2.2.1 version of Mujoco, both windows binaries and source code.
I am running Windows 10, Unity 2020.3.36f1.
Create a vanilla 3D project from unity hub
I imported the package from source code (using package.json) as suggested. I copied the DLL in place.
I imported arm26.xml in unity (after checking it out in simulate program)
Saved the scene, hit play, crash!

[image]

[Editor_crash.log](https://github.com/deepmind/mujoco/files/9163694/Editor_crash.log)
?

michelhabib, 2022-07-22:
If it's worth noting, I installed 2021.3.6f1, same issue. 
I made sure there are no floating mujoco dlls of earlier releases, deleted them all.
I double checked with 2.2.0 - same issue, restarted my pc and check again, still the same.
I know I tried an earlier plugin version successfully many months ago (5+), but I don't recall which one.

jyt94, 2022-08-01:
I have met the same issue, also reverting to 2.2.0 solved the problem.

CheAbdullah, 2022-08-31:
Unity -- Project. Folder Package (where you put the mujoco.dll in this folder). Select this mujoco.dll. And see the Unity Inspector. Click Editor (ticked). Click Apply. 
Works with (https://github.com/deepmind/mujoco/releases) mujoco 2.20. 
Make sure you copy the mujoco.dll from the unzipped folder (mujoco-2.2.0-windows-x86_64\bin) to your User folder (C:\Users\YourName\MuJoCo\bin)

kevinzakka, 2023-04-28:
Closing this for now, please re-open if the issue persists.

closed 2023-04-28 by kevinzakka