google-deepmind/mujoco #398: Using mujoco rendering on CircleCI

vmoens, 2022-07-25:
We're trying to use mujoco (specifically dm_control)-based environments in [our tests](https://github.com/facebookresearch/rl/blob/main/test/test_env.py).

We'd like to use the egl backend, as osmesa is slow (it often stays idle for a very long time) and glfw is not available in headless environments.

We have installed all the [required dependencies](https://github.com/facebookresearch/rl/blob/8b1d2040af6e009cf3033fff94240355bddbb9e6/.circleci/unittest/linux/scripts/setup_env.sh#L59-L63) as pointed [here](https://github.com/deepmind/dm_control/issues/136) but the following error still occurs:
```
Collecting dm_control
  Downloading dm_control-1.0.4.tar.gz (38.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 38.2/38.2 MB 67.6 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'error'

Pip subprocess error:
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [25 lines of output]
      Traceback (most recent call last):
        File "/root/project/env/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/root/project/env/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/root/project/env/lib/python3.8/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 130, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-012xq985/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 177, in get_requires_for_build_wheel
          return self._get_build_requires(
        File "/tmp/pip-build-env-012xq985/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 159, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-012xq985/overlay/lib/python3.8/site-packages/setuptools/build_meta.py", line 174, in run_setup
          exec(code, locals())
        File "<string>", line 26, in <module>
        File "/tmp/pip-build-env-012xq985/overlay/lib/python3.8/site-packages/mujoco/__init__.py", line 52, in <module>
          from mujoco.egl import GLContext
        File "/tmp/pip-build-env-012xq985/overlay/lib/python3.8/site-packages/mujoco/egl/__init__.py", line 31, in <module>
          from mujoco.egl import egl_ext as EGL
        File "/tmp/pip-build-env-012xq985/overlay/lib/python3.8/site-packages/mujoco/egl/egl_ext.py", line 27, in <module>
          from OpenGL import EGL
        File "/tmp/pip-build-env-012xq985/overlay/lib/python3.8/site-packages/OpenGL/EGL/__init__.py", line 2, in <module>
          from OpenGL.raw.EGL._types import *
        File "/tmp/pip-build-env-012xq985/overlay/lib/python3.8/site-packages/OpenGL/raw/EGL/_types.py", line 87, in <module>
          raw_eglQueryString = _p.PLATFORM.EGL.eglQueryString
      AttributeError: 'NoneType' object has no attribute 'eglQueryString'
```

The strangest thing is that ssh-ing on the machine and executing the same code resolves the issue.

Any help would be appreciated.

saran-t, 2022-08-01:
> The strangest thing is that ssh-ing on the machine and executing the same code resolves the issue.

This implies that your machine has everything needed installed already. I don't know what to suggest since I don't know CircleCI, but if I'm investigating it I'd start by dumping all environment variables from your SSH session and on the CI workflow and see if there's any difference there.

vmoens, 2022-08-06:
Thanks for the tip! We solved the issue by not relying on docker to run the tests (which was messing up the env variables).

See [this](https://github.com/facebookresearch/rl/pull/339) for more context.

closed 2022-08-06 by vmoens