google-deepmind/mujoco #399: Some details and questions about the force-aspect sensors

yinzikang, 2022-07-26:
Hi, I am doing some contact tasks with sensors in Mujoco, and I made some summaries below according to #140, and #228.
Could you please point out whether I am right？ Any comment is welcomed.

1. `touch` sensor measures the normal force between two contacting independent geoms, while the `force` sensor and the `torque` sensor work between a child and a parent body. So it's unable to measure the force and torque between two independent geoms directly so far. (Please tell me if there are some methods I ignore)

2. To measure contact force and torque between two independent bodies, such as a `gripper` and a `bottle`, another body(`dummy body`) should be added to the `gripper` body, instead of just another geom. 
The `dummy body` must have a `geom`, a `site` for the sensor, a small but positive `inertial`, and no `joint`, like below.
```
<body name="gripper" pos="0 0 0.8">
    <inertial pos="0 0 0" mass="1" diaginertia="0.1 0.1 0.1" />
    <geom type="box" pos="0 0 0" size="0.1 0.1 0.1" rgba="1 0 0 1"/>
    <body name="dummy body" pos="0 0 -0.1">
        <inertial pos="0 0 0" mass="0.0001" diaginertia="0 0 0" />
        <geom type="box" pos="0 0 0" size="0.1 0.1 0.1" rgba="0 1 0 1"/>
        <site name="sensor" pos="0 0 0" rgba="1 0 0 0" size="0.01 0.01 0.01" type="box"/>
    </body>
</body>
```

3. in the 2 above, it is the `dummy body` which contacts the `bottle` instead of the `gripper`.The sensor actually still measures the force and torque between `gripper` and `dummy body`, which is very close to the force and torque between `dummy body` and `bottle` due to the small mass. The 'geom' that directly belongs to the gripper should stay away from the bottle, otherwise, the sensor data would be wrong.

BTW, I still got some problems unsolved.
1. does the position and the scale of the `dummy body`'s geom, site matter?
2. I still cannot get the correct force information about models with the `equality` attribute in #229. I‘d be obliged if there are some simple examples or tests.

Best Regards

yuvaltassa, 2022-08-20:
Sorry for the delay.

1. This is correct.
2. This is correct.
3. This is correct.

Regarding your unresolved questions:

1. No.
2. Yes. There are tests [here](https://github.com/deepmind/mujoco/blob/main/test/engine/engine_core_smooth_test.cc#L57) which reference test models [here](https://github.com/deepmind/mujoco/tree/main/test/engine/testdata/core_smooth/rne_post). Please inspect these and let us know if you have any further questions.

closed 2022-08-20 by yuvaltassa

hai-h-nguyen, 2023-06-03:
I want to clarify because it seems the position of the geometry inside the dummy body matters a lot in my experiment. Therefore, the position of the dummy body also matters. I have a peg and  hole system w/ the code below:

`            <body name="object" pos="0 0 0.16">
                <geom name="peg_body" type="box" size="0.02 0.02 0.05" group="1" material="plate_mat" solref=".004 1" density="1000" />
                <body name="tip" pos="0 0 0.03">
                    <site name="ft_sensor" pos="0 0 -0.0" rgba="1 0 0 1" size="0.02 0.02 0.02" type="box"/>
                    <site name="end_peg" pos="0 0 0.02" rgba="1 0 0 0" size="0.005"/>
                    <inertial pos="0 0 0" mass="1e-6" diaginertia="0 0 0" />
                    <geom pos="0 0 0.0" group="1" size="0.02 0.02 0.02" type="box" />
                </body>
            </body>
`

And I test three cases here with different positions of the sensor site (red) and the geometry (grey) inside the dummy body.

[image]
[image]
[image]

In my experiments, the sensor site's position does not matter. However, the geometry must cover the end of the tip so that the F/T sensor signals are there when there are contacts between the tip and the peg (and the table). Changing the site position does not affect anything.