google-deepmind/mujoco #400: [ Help wanted ] Eye-in-hand Camera rendering issue

jaku-jaku, 2022-07-26:
Hi,

I'm a (student) and I'm trying to use MuJoCo for research developments.

I'm looking for some help with eye-in-hand stereo camera vision modelling.

Issue: The camera seems fail to render the occluded objects close to the camera, and I am wondering if this is a bug or due to my configuration issue.

====

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
                        <!-- [Optional] ZED Camera on forearm link, TODO: uncertain for inertial params -->
                        <body name="wam/sensor/zed" pos="-0.18 -0.15 0" quat="0 0.707107 0 0.707107">
                            <inertial               pos="0 0 0.015" mass="0.124"/>
                            <geom                   class="wam/viz" mesh="zed"/>
                            <geom                   class="wam/col" mesh="zed_collision_mesh"/>
                            <camera name="camera/zed/R" pos="-0.06 0 0.042"  quat="0 0 -0.707107 0.707107" fovy="60"/>
                            <camera name="camera/zed/L" pos=" 0.06 0 0.042"  quat="0 0 -0.707107 0.707107" fovy="60"/>
                        </body>
                        
```
</details>

Here is a screenshot / video, illustrating my question:
[image]

Many Thanks!
Jack

Balint-H, 2022-07-27:
Hello,

This looks like its the clipping plane in action, you could alter the values of znear in the XML perhaps?
[See the znear property here](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-map)

jaku-jaku, 2022-07-27:
Dear Balint,

Thank you so much, yeah, it was something related to znear configuration.

Sincerely,
Jack

closed 2022-07-27 by jaku-jaku