google-deepmind/mujoco #402: I'm trying to inlcude mujoco into my own CMakeLists.txt for my project and I'm getting lots of `undefined reference` errors

MotorCityCobra, 2022-07-28:
I built and installed MuJoCo from source on my Linux machine. I am able to include mujoco in my C++ and it looks like not every part of mujoco I try to use throws an error but many do.  

Here is my CMakeLists.txt  
```

cmake_minimum_required(VERSION 3.5.1)
project(mujoco_gym)

set(CMAKE_CXX_STANDARD 17)
# It prevents the decay to C++98 when the compiler does not support C++14
set(CMAKE_CXX_STANDARD_REQUIRED ON)
# It disables the use of compiler-specific extensions
# e.g. -std=c++14 rather than -std=gnu++14
set(CMAKE_CXX_EXTENSIONS OFF)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall")

set(Torch_DIR libtorch/share/cmake/Torch)
find_package(Torch PATHS ${Torch_DIR} NO_DEFAULT REQUIRED)
if (Torch_FOUND)
    message(STATUS "Torch library found!")
    message(STATUS "    include path: ${TORCH_INCLUDE_DIRS}" \n)
else ()
    message(FATAL_ERROR "Could not locate Torch" \n)
endif()

include_directories(mujoco/build/_deps/glfw-src/include)
include_directories(mujoco/build/lib)
include_directories(mujoco/build)
include_directories(mujoco/sample)
include_directories(mujoco/include)

find_package(mujoco REQUIRED)
link_libraries(mujoco::mujoco)

find_package(GSL REQUIRED)
link_libraries(GSL::gsl)

file(GLOB SOURCE_FILES small_simulator.cpp)

#CMAKE_PROJECT_NAME is from 'project(mujoco_gym)' on the second line of this script
add_executable(${CMAKE_PROJECT_NAME} ${SOURCE_FILES})

target_link_libraries (
        ${CMAKE_PROJECT_NAME}
        ${TORCH_LIBRARIES}
        ${MUJOCO_LIBRARIES}
        ${GSL_LIBRARIES}
)
```

`$cmake ..` completes without errors but `cmake --build .` causes these errors  
```
$ cmake --build . 
Consolidate compiler generated dependencies of target mujoco_gym
[ 50%] Linking CXX executable mujoco_gym
/usr/bin/ld: CMakeFiles/mujoco_gym.dir/small_simulator.cpp.o: in function `mouse_button(GLFWwindow*, int, int, int)':
small_simulator.cpp:(.text+0x87): undefined reference to `glfwGetMouseButton'
/usr/bin/ld: small_simulator.cpp:(.text+0xa4): undefined reference to `glfwGetMouseButton'
/usr/bin/ld: small_simulator.cpp:(.text+0xc1): undefined reference to `glfwGetMouseButton'
/usr/bin/ld: small_simulator.cpp:(.text+0xea): undefined reference to `glfwGetCursorPos'
/usr/bin/ld: CMakeFiles/mujoco_gym.dir/small_simulator.cpp.o: in function `mouse_move(GLFWwindow*, double, double)':
small_simulator.cpp:(.text+0x1a1): undefined reference to `glfwGetWindowSize'
/usr/bin/ld: small_simulator.cpp:(.text+0x1b2): undefined reference to `glfwGetKey'
/usr/bin/ld: small_simulator.cpp:(.text+0x1c8): undefined reference to `glfwGetKey'
/usr/bin/ld: CMakeFiles/mujoco_gym.dir/small_simulator.cpp.o: in function `main':
small_simulator.cpp:(.text+0x79c): undefined reference to `glfwInit'
/usr/bin/ld: small_simulator.cpp:(.text+0x7d8): undefined reference to `glfwCreateWindow'
/usr/bin/ld: small_simulator.cpp:(.text+0x7ee): undefined reference to `glfwMakeContextCurrent'
/usr/bin/ld: small_simulator.cpp:(.text+0x7f8): undefined reference to `glfwSwapInterval'
/usr/bin/ld: small_simulator.cpp:(.text+0x889): undefined reference to `glfwSetKeyCallback'
/usr/bin/ld: small_simulator.cpp:(.text+0x8a2): undefined reference to `glfwSetCursorPosCallback'
/usr/bin/ld: small_simulator.cpp:(.text+0x8bb): undefined reference to `glfwSetMouseButtonCallback'
/usr/bin/ld: small_simulator.cpp:(.text+0x8d4): undefined reference to `glfwSetScrollCallback'
/usr/bin/ld: small_simulator.cpp:(.text+0xa44): undefined reference to `glfwGetFramebufferSize'
/usr/bin/ld: small_simulator.cpp:(.text+0xabc): undefined reference to `glfwSwapBuffers'
/usr/bin/ld: small_simulator.cpp:(.text+0xac1): undefined reference to `glfwPollEvents'
/usr/bin/ld: small_simulator.cpp:(.text+0x104b): undefined reference to `glfwWindowShouldClose'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/mujoco_gym.dir/build.make:106: mujoco_gym] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/mujoco_gym.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
$ 
```

My C++ file is one from the sample directory in the repo. It's large and could be any sample .cc file in the repo so I am not including.  
I know MuJoCo installed on my machine in various paths under /usr/local and I have included every path I can think of from the mujoco repo I have downloaded so it's not clear to me why I can't include the library and build with cmake.

saran-t, 2022-07-28:
CMake doesn't add dependencies transitively, so you need to `add_dependencies` on GLFW as well as `mujoco::mujoco`.

In other news, I think there's something a bit broken with our CMake setup since I'm setting missing MuJoCo symbols. Still investigating.

MotorCityCobra, 2022-07-28:
This works for you?  
Is there a problem with my CMakeLists.txt? I thought I already have what you recommend. 
For GLFW I did `include_directories(mujoco/build/_deps/glfw-src/include/GLFW)`. I don't know how to add it with `add dependencies`.  

I've tried a few variations and I get the same error as in the original post.

saran-t, 2022-07-28:
`array_safety.h` and `uitools.h` aren't currently officially part of MuJoCo itself, rather they're shipped as part of our sample applications. This may change in the near future, but in the meantime if your codebase needs these two headers you should copy and incorporate them into your own project.

MotorCityCobra, 2022-07-29:
> only `target_dependencies` should suffice. And you still need to add a dependency on GLFW as I mentioned above.

I assume you mean `add_dependencies` for `glfw`. I've been trying all day to get MuJoCo to use `glfw`. Like I said, I'm new to cmake. Since I couldn't get `add_dependencies` to work I cloned the glfw repo, built and installed it. I tried to build the `simulate.cc` file in the sample folder and I got the same errors as above.  

 This is how I brought the glfw library after I built it from the glfw github repo  
```

set(glfw3_DIR /glfw/build)
find_package(glfw3 PATHS ${glfw3_DIR} NO_DEFAULT REQUIRED)
if (glfw3_FOUND)
    message(STATUS "glfw3 library found!")
    message(STATUS "    include path: ${glfw3_INCLUDE_DIRS}" \n)
else ()
    message(FATAL_ERROR "Could not locate glfw3" \n)
endif()

find_package(GSL REQUIRED)
link_libraries(GSL::gsl)

file(GLOB SOURCE_FILES mujoco_gym.cpp)

#CMAKE_PROJECT_NAME is from 'project(mujoco_gym)' on the second line of this script
add_executable(${CMAKE_PROJECT_NAME} ${SOURCE_FILES})

target_link_libraries (
        ${CMAKE_PROJECT_NAME}
        ${TORCH_LIBRARIES}
        ${GSL_LIBRARIES}
        ${mujoco_LIBRARIES}
        ${glfw3_LIBRARIES}
)
```

saran-t, 2022-08-01:
When you place `uitools.h` and `uitools.c` into your own project, you either need to add it in the sources list of your binary/library target, or put them into a dedicated library and `target_link_libraries` to that.

guofeng509, 2024-06-14:
where can I find the file of  uitools.h and uitools.c

yuvaltassa, 2024-06-16:
These makefiles are old and unsupported. You can fix them and send a PR or we can just delete them. @saran-t has been wanting to delete them for a while...

guofeng509, 2024-06-23:
ok, Thank you very much

commit 1601ab60 (2024-07-01): Remove Makefiles. Fixes #402

These are no longer needed, as the open-source build uses CMake.

PiperOrigin-RevId: 648340374
Change-Id: Ibc704ca3aaab1c48d245e24b071bed6cd846c110