google-deepmind/mujoco #404: mj_copyData not available in the Python bindings

rpapallas, 2022-07-28:
It seems that the function `mujoco.mj_copyData` is not implemented for Python. Python gives the following error:

```
AttributeError: module 'mujoco' has no attribute 'mj_copyData'
```

Although I can see some evidence that this should have been available here https://github.com/deepmind/mujoco/blob/d3a86bb748b272951185a24570f49ceccf07f4e7/introspect/functions.py#L584 it seems that is not.

saran-t, 2022-07-28:
Please use the usual Pythonic to copy these objects: https://github.com/deepmind/mujoco/blob/d3a86bb748b272951185a24570f49ceccf07f4e7/python/mujoco/bindings_test.py#L401

This is sort of documented [here](https://github.com/deepmind/mujoco/blob/main/python/mujoco/functions.cc#L125) but I now notice that we haven't mentioned it in our documentation.

rpapallas, 2022-07-28:
Thank you @saran-t, I have just actually just stumbled upon this and managed to make a copy of data using copy.deepcopy.

closed 2022-07-28 by rpapallas