google-deepmind/mujoco #406: Getting a C1853 error in uitools.c: Problem with preprocessing

st168992, 2022-07-29:
Hi,

first of all my expertise with mujoco and programming in general is pretty low.
I'm currently in the process of trying to re-compile the code on visual studio 2017.

Now after fixing every error message that popped up, there's still a single C1853 error in uitools.c left.

For some reason pch.h doesn't work for this specific file.

What I've tried already with no success:

- changing uitools.c settings from _use precompiled header_ to _not use precompiled header_ (got 133 errors as result (LNK2005 & LNK2001)

- if I'm deleting the line with #include <pch.h> I'm getting a C1010 error (tells me that I might forgot to add #include <pch.h> at the top)

Right now, I don't know how to progress from here on.
Help is deeply appreciated.

additional info:

using
- Windows 10 Enterprise
- mujoco 2.2.1
- Visual Studio 2017

line of code in question in bold:
uitools.c:

**#include <pch.h>**
#include "uitools.h"
#include <stdio.h>
#include <string.h>

saran-t, 2022-08-01:
What's the resolution for this in the end?

st168992, 2022-08-05:
Had no solution lol. Figured out that I only need basic.cc for now as my main purpose is to visualize my build right now.

Should I reopen the issue for future users who might have a solution to this?

st168992, 2022-08-19:
> What's the resolution for this in the end?

st168992, 2022-08-19:
> What's the resolution for this in the end?

Maybe important extra info: I had a bunch of "4996: 'strncat'(also 'sscanf' and 'fopen') may be unsafe"-errors before and chose to ignore them by disabling deprecation.

Also the glfw-release I'm working with is the latest version: 3.3.8

Added to the project, for additional info, are all cc-, and c-files, as well as all header-files.

thanks in advance!

nimrod-gileadi, 2024-01-02:
Apologies if this is still an issue for you. Closing since there hasn't been any activity on this issue in a while, and there have been a lot of changes to MuJoCo since then.

closed 2024-01-02 by nimrod-gileadi