google-deepmind/mujoco #407: Unity MJ Scene Import - geom mass and density values are lost

michelhabib, 2022-07-29:
using 2.2.0 and Unity 2020.3
I built a single muscle mujoco model to use with ML Agents. However, I noticed that the model actuation behaves differently from the simulator, which bugged me for a while. It took a bit of tracing to figure out the problem, but I will state the issue first.

I started by importing the muscle model in Unity using "Import Mujoco Scene". [singlemuscle.txt](https://github.com/deepmind/mujoco/files/9222544/singlemuscle.txt)
I set the actuator to 0.445, the muscle pulls the arm all the way up in unity, but pulls it only moderately in the simulator.
sensordata in simulator shows a tendon length of 0.21, while in unity, the actuator debug value shows 0.07

I started by comparing the original xml to the xml saved by the unity script, I suspected the muscle parameters turned into general, but I was wrong.
I kept trying to get the 2 files close to each other (Original and Script generated), until I figured it has to do with the mass/density values, these values get completely lost in translation (pun intended)
The script generated xml ignores any mass/density values

The problem happens here:
```
public unsafe GameObject ImportFile(string filePath) {
    // If MuJoCo can't parse the mjcfString, we abort the entire process.
    // MjEngineTool.LoadModelFromString throws an exception when MuJoCo fails to parse the provided
    // mjcfString.
    var name = Path.GetFileNameWithoutExtension(filePath) + $"{UnityEngine.Random.Range(0,999)}";
    string newPath = Path.Combine(Application.temporaryCachePath, $"{name}.xml");
    _mjModel = MjEngineTool.LoadModelFromFile(filePath);
    MjEngineTool.SaveModelToFile(newPath, _mjModel);
    Debug.Log($"Imported MJCF loaded, saved to {newPath}");
    string mjcfString = File.ReadAllText(newPath);
    GameObject root = null;
    try {
      root = ImportString(mjcfString, name, filePath);
    } finally {
      MujocoLib.mj_deleteModel(_mjModel);
    }
    return root;
  }
```

Briefly, the original XML is loaded into mjModel, the mjModel is saved to new XML, the new XML is loaded into Unity.
The problem happens somewhere in the middle, probably when the mjModel is saved to XML?
If I knew how to read mass/density values from mjModel, probably I could trace this more, but I think that's the best I can do at the moment.

Actually, I changed the code above to ImportString directly from the original XML and all worked fine for me, so it's an acceptable workaround, knowing that I am skipping the failsafe XML Validation, but I know we should take advantage of chances in life sometimes.

[image]

quagla, 2022-08-02:
Hi Michel, I have tested your model with the XML load/save and if I load it, save it, and load it again, then all internal MuJoCo data structure are exactly identical, so I am inclined to think that the issue is not coming from saving the XML, but rather from the Unity plugin. Can you reproduce the bug in any way if you load-save-load in MuJoCo rather than Unity? Thanks.

michelhabib, 2022-08-02:
yes, it is indeed plugin related. I am comparing to the simulator, as my reference of confidence.
The code I am referring to, above, is an extract from the unity plugin code that I am suspecting.
this line:
MjEngineTool.SaveModelToFile(newPath, _mjModel);
saves the model back to xml, without the mass/density attributes.
If you'd like, I can show you how the xml looks like after that code saves it.
As I mentioned, skipping the xml load/save (for the purpose of format verification) fixes it, which confirms that the load model/save model step above is most probably the source of the problem.

quagla, 2022-08-02:
Yes please I would like to look at the XML. AFAIK, the plugin is calling the same load/save XML functions that are used in MuJoCo, so I'm surprised that the issue shows up only in the plugin.

michelhabib, 2022-08-02:
had to rename to text to be able to upload to github.
First one is the original xml file containing 2 nodes with mass=0
Second one is the file saved in temp folder by the plugin, mass value disappeared.
Same behavior happens for density as well, but I didn't include density in this test file.

I am using the "Import Mujoco Scene" from the Assets Menu, Unity 2020.3, Plugin 2.2.0 (2.2.1 currently crashes, so I am not usin g it and reported it separately)

I am surprised as well, however the scenario is straight forward. I reviewed the XML reference to make sure I am using the correct syntax for assigning mass/density in Mujoco.
Is it possible that the MjGeom Scripts are not picking up the values correctly, and they may be involved in the process of loading/saving?

[singlemuscle-original.txt](https://github.com/deepmind/mujoco/files/9242122/singlemuscle-original.txt)
[singlemuscle-savedbyplugin.txt](https://github.com/deepmind/mujoco/files/9242123/singlemuscle-savedbyplugin.txt)

quagla, 2022-08-02:
Do you see the mass disappearing also in the inspector? For me it's there.
[image]

michelhabib, 2022-08-02:
this screenshot is showing the xml file - which I assume is the file I uploaded.
Did you try loading the xml using MJ Import Scene and look at the geom nodes with zero mass?

quagla, 2022-08-02:
I get a crash with both 2.2.1 and 2.2.0 with the latest Unity..

michelhabib, 2022-08-02:
2.2.1 unity plugin seems corrupt, I already reported that in a separate issue, and this issue here is reported for 2.2.0. 
2.2.0 plugin works fine, unless you accidentally let it pick up the 2,2.1 dll

quagla, 2022-08-05:
I tried with Unity 2022.1 and whatever caused the plugin to crash is now fixed. I can import your files so I will take a look.

quagla, 2022-08-05:
OK that's interesting because the XML in my temp folder does have mass="0 (see attachment) and I see mass 0 also in the properties of the geom in Unity (see screenshot). What is your version of the unity plugin? The zero mass issue was solved on May 25th: https://github.com/deepmind/mujoco/commit/223874c4cab35cd88332947e0d9b13d5ec3822e4

[image]

[singlemuscle_temp.txt](https://github.com/deepmind/mujoco/files/9266990/singlemuscle_temp.txt)

michelhabib, 2022-08-05:
You are actually showing the problem i reported. mass is zero but density is 1000. density shouldn't be 1000 when mass is zero.
If you try the steps I indicated in the main issue, the results are different between unity and simulator. did you try them?

The commit you are referring to is after 2.2.0 - so I cannot confirm it, until 2.2.1 crash is resolved - you mentioned you are using 2.2.0 as well. I'd be happy to test it if fixed.

michelhabib, 2022-08-05:
it's interesting that the mass is zero in your temp folder, that's different from what I have, which unity plugin version are you using?

kevinzakka, 2024-08-30:
Closing because stale.

closed 2024-08-30 by kevinzakka