google-deepmind/mujoco #408: I have a newb question about sizes of arrays and data types

MotorCityCobra, 2022-07-31:
My ultimate goal is to flatten out and concatenate a few data observations, but I will be verbose.

I am fairly new to C++ and I would like to know what these are exactly and how to get all the numbers quickly...  
`d->qpos, d->qvel, d->cinert, d->cvel, d->qfrc_actuator, d->cfrc_ext`  
When I tired to print the `typeid(q->pos).name()` it just printed db.

I know what data they are showing but I don't know how best to access it. Take `q->pos` when running the Humanoid model for example. Do I need to look at the XML for the amount of numbers in this `q->pos` object? When I `std::cout` (print) `q->pos` I get a memory address.  
I can index the object `q->pos` to get one of the positions for one of the joints. Would it be best to run a loop over the object?

kevinzakka, 2022-07-31:
Hi @MotorCityCobra, the [API reference](https://mujoco.readthedocs.io/en/latest/APIreference.html) and specifically [MjModel](https://mujoco.readthedocs.io/en/latest/APIreference.html#mjmodel) should be helpful in understanding the sizes of the different structs and arrays.

MotorCityCobra, 2022-07-31:
> So yes, you can use a for loop

Okay, but just because I don't understand I'll ask and maybe I'll learn something about C++...  
Why can't I grab the whole array at once and plug it into Libtorch? Why do I have to loop through for each one individually?  
I'm coming from Python where it would already be in a format / data structure that I could access all at once. Or, that's what Python makes it seem. Maybe it's all loops underneath Python?

kevinzakka, 2022-07-31:
You're probably used to using numpy in python, in which case yes, all vectorization is under the hood calling C code with for loops. Actually probably useful to read the numpy docs on this, see [here](https://numpy.org/doc/stable/user/whatisnumpy.html).

closed 2022-08-01 by MotorCityCobra