google-deepmind/mujoco #414: How to get the 3D position of rope end?

wangjunyi9999, 2022-08-02:
Hi, I am new in Mujoco. I'm looking for some help with acquiring the rope end 3D coordinates position.

- FIrstly, as far as I refer the tutorial, it seems that we can only know the start coordinate of rope rather than the end one.  Also, only two methods can provide "pos" information: directly define the "pos" coordinate or using "joint" attribute "ref". If I add a joint at the end of rope, they cannot be connected. Therefore, could you provide me a tip how to get the rope end coordinate? 

- Secondly, I tried my best to find other solution to solve the problem, for example, I defined a new small "capsule" type body, and connect it with the rope end, so I can receive the almost same coordinate of rope end. But I also find the capsule would float, then the rope end would separate from it. I think using "anchor" attribute can solve it but actually not. I am not sure if I should twick the param or other params, such as "solimp", the "pos" of two bodies or else.

Here is a model which explains my question:

[reacher_update.txt](https://github.com/deepmind/mujoco/files/9245471/reacher_update.txt)

Here is a screenshot / video, illustrating my question:
[image]

Thanks!

saran-t, 2022-08-02:
I'm not really sure what you actually did in your second point, but if all you want is the position of the tip of the rope, you should add a `<site>` in the innermost body, make sure that it's positioned correctly within that body so that it's at the tip of your last capsule, then read what you want from `d->site_xpos`.

wangjunyi9999, 2022-08-03:
@saran-t  Thanks for your answer! Actually, the sencond point is very similar with the "site" method, so just leave it now. 

By using the function of site, I could know the last capsule position. But when I rotate the arms, the end of rope would be detached with the "site".  What should I do to fix them together, so whatever the arm rotates, the "site" would reprsent the rope end position?

wangjunyi9999, 2022-08-03:
I tried <equality> but <site> is not a body, so I cannot connect them. Also,  if I using <tendon>,  it would not match my physical model because I just need a rope and the position of the rope end.

Balint-H, 2022-08-03:
Maybe you could still use equality constraints between the arm and the rope's body, and use the "relpose" attribute in the xml to set it according to where the site / rope end is.

wangjunyi9999, 2022-08-03:
@Balint-H  Hi, I follow your advice but receive an error:
XML Error: Schema violation: unrecognized attribute: 'relpose'
this is the code,  it seems not support "relpose" attribute
`<equality>
        <connect body1="body1" body2="CB9" anchor="0 0 0" relpose="0 0 0.38 0 0 0 0" />
    </equality>`

Balint-H, 2022-08-03:
Hello, `relpose` is only available in the `weld` type equality constraints. With it you can define any spatial relationship, even if its not satisfied at the start (at which point it will quickly snap to satisfy it). `weld` also constrains orientation to match as well, which might be fine for your case or not. If not, you should stick to connect, however then the constraint is assumed to be satisfied at the start. You should move the target to the end of the rope in the MJCF, and define the anchor location from body1's frame of reference to be the tip of the rope. In this case you can remove the fingertip object and connect the target to CB9 directly.

wangjunyi9999, 2022-08-03:
Thank you! I find another instead method, which is defining a very small freejoint box at the end of rope. Thus I would track the rope end, but I am not sure if it would affect my simulation result. And I would also try your suggestion later.

wangjunyi9999, 2022-09-01:
I would close the issue because I find a way to solve the question. We can set two sites between the second joint end and tip, then using tendon to connect them.

closed 2022-09-01 by wangjunyi9999

baixianger, 2023-04-23:
> Thank you! I find another instead method, which is defining a very small freejoint box at the end of rope. Thus I would track the rope end, but I am not sure if it would affect my simulation result. And I would also try your suggestion later.

would you like to show this method to us? that would be a major help to me. Thank you in advance!