google-deepmind/mujoco #420: Cylinder/Box collision issue.

benureau, 2022-08-04:
Hi,

I'm a researcher and I'm trying to use MuJoCo for robotics simulations.

I'm looking for some help with simple object collisions. I have a box (static) and a cylinder (freejoint). I wish to drop the cylinder on the table and get a somewhat believable interaction. 

[image]

Here's the model:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <body name="table">
        <geom type="box" size="0.3 0.6 0.3" pos="0 0 0" rgba="1 1 1 1"/>
    </body>

    <body name="cylinder">
        <freejoint/>
        <geom type="cylinder" pos="0 0.2 0.6" size=".06 .06" rgba="1 0 0 1"/>
    </body>
  </worldbody>
</mujoco>

```

</details>

The result, with MuJoCo 2.2.1:

https://user-images.githubusercontent.com/2088743/182828701-9c304124-3966-4260-8718-9279c0420b85.mov

Depending on the high/position, the cylinder will slide in many different directions, sometimes gaining great velocity, etc. 

If I drop it in the exact center of the table, it (correctly) lands upright:

[image]

I have fiddled with many different scenarios/parameters of the simulation (timestep/integrator/friction/mass/condim/margin, etc.), and this behavior stays brittle to changes in cylinder size/drop heigh/drop position. I feel I am missing something obvious (should I specify the inertia or is it not already done automatically? Is there a way to have minimal penetration at collision time, unlike the behavior here?). Or is it an inherent limitation of MuJoCo?

yuvaltassa, 2022-08-04:
This is likely due to box-cylinder being handled by the convex collider, though there might be other issues. Please load this model in `simulate` where you can visualise contacts and contact forces and also slow down the simulation ('C', 'F' and '-' keys respectively)

Try enabling multiccd using 
```xml
<option>
  <flag multiccd="enable"/>
</option>
```

This option should increase the number of contacts between the box and the cylinder and certainly alleviate the issue. It could also be the case that the default contact is too soft (MuJoCo's defaults are rather soft), in which case you should probably try decreasing `solref[0]` for all your geoms (probably in a `default` clause would be easiest).

I'll close this as I suspect these suggestions will solve your issue but please write here if it doesn't.

Very generally using `simulate` as a visual debugger is very highly recommended.

closed 2022-08-04 by yuvaltassa

benureau, 2022-08-04:
@yuvaltassa: Thank you! Indeed, enabling multiccd and decreasing solref to 0.001 did the trick, as far as I can observe.