google-deepmind/mujoco #423: Skin Breaks Away from Bodies It Is Bound To

gvittrup, 2022-08-04:
Hi,

I'm a student and software developer looking to add MuJoCo to a realtime motion capture software. I am using 2.2.1 and switching between MacOS and Windows!

I have very slowly developed an understanding of creating skin assets, and thanks to patient assistance, I've gotten to the point where I can implement any 3D model I have created thus far as a skin in the engine; at least before running a simulation that is. Once I run a simulation, I notice that my skin will break away from the MuJoCo bodies that I bound the bones to in the skin. For this request, I am working with a cylinder with 21 segments and 176 vertices. I wrote the skin asset within the composite rope model, and find the skin will stick with the rope until it collides with the capsule beneath it. 

One interesting thing I noticed about the skin is that it is successfully bound to the rope body 'B10' (which I colored in white), though nothing about my approach changes throughout creating the bones.  

Here is my declaration of the bones in the skin:

```XML
<bone body="B0" bindpos="0 0 0" bindquat="1 0 0 0" vertid="0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15" vertweight="1 1 1 1 1 1 1 1 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 "/>
<bone body="B1" bindpos="0 0 0" bindquat="1 0 0 0" vertid="8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B2" bindpos="0 0 0" bindquat="1 0 0 0" vertid="16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B3" bindpos="0 0 0" bindquat="1 0 0 0" vertid="24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B4" bindpos="0 0 0" bindquat="1 0 0 0" vertid="32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B5" bindpos="0 0 0" bindquat="1 0 0 0" vertid="40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B6" bindpos="0 0 0" bindquat="1 0 0 0" vertid="48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B7" bindpos="0 0 0" bindquat="1 0 0 0" vertid="56 57 58 59 60 61 62 63 64 65 66 67 68 69 70 71" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B8" bindpos="0 0 0" bindquat="1 0 0 0" vertid="64 65 66 67 68 69 70 71 72 73 74 75 76 77 78 79" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B9" bindpos="0 0 0" bindquat="1 0 0 0" vertid="72 73 74 75 76 77 78 79 80 81 82 83 84 85 86 87" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B10" bindpos="0 0 0" bindquat="1 0 0 0" vertid="80 81 82 83 84 85 86 87 88 89 90 91 92 93 94 95" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B11" bindpos="0 0 0" bindquat="1 0 0 0" vertid="88 89 90 91 92 93 94 95 96 97 98 99 100 101 102 103" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B12" bindpos="0 0 0" bindquat="1 0 0 0" vertid="96 97 98 99 100 101 102 103 104 105 106 107 108 109 110 111" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B13" bindpos="0 0 0" bindquat="1 0 0 0" vertid="104 105 106 107 108 109 110 111 112 113 114 115 116 117 118 119" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B14" bindpos="0 0 0" bindquat="1 0 0 0" vertid="112 113 114 115 116 117 118 119 120 121 122 123 124 125 126 127" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B15" bindpos="0 0 0" bindquat="1 0 0 0" vertid="120 121 122 123 124 125 126 127 128 129 130 131 132 133 134 135" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B16" bindpos="0 0 0" bindquat="1 0 0 0" vertid="128 129 130 131 132 133 134 135 136 137 138 139 140 141 142 143" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B17" bindpos="0 0 0" bindquat="1 0 0 0" vertid="136 137 138 139 140 141 142 143 144 145 146 147 148 149 150 151" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B18" bindpos="0 0 0" bindquat="1 0 0 0" vertid="144 145 146 147 148 149 150 151 152 153 154 155 156 157 158 159" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B19" bindpos="0 0 0" bindquat="1 0 0 0" vertid="152 153 154 155 156 157 158 159 160 161 162 163 164 165 166 167" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5"/>
<bone body="B20" bindpos="0 0 0" bindquat="1 0 0 0" vertid="160 161 162 163 164 165 166 167 168 169 170 171 172 173 174 175" vertweight="0.5 0.5 0.5 0.5 0.5 0.5 0.5 0.5 1 1 1 1 1 1 1 1"/>
```

Here are two screenshots, before and after colliding with the capsule:

[image]
[image]

I did not change anything from the provided composite rope model. I have tried several different approaches to fixing this, but after a point I believe I was shooting in the dark once I did not see any expected or noticeable outcomes. I am not sure if I provided all necessary information to adequately solve this problem, so please let me know what else I can provide to assist me with this.

quagla, 2022-08-05:
Hi, could you please provide an XML file of the example that fails so I can have a look? Thanks!

gvittrup, 2022-08-05:
Here you go, I couldn't upload it directly on Git so hopefully this works!
[https://drive.google.com/file/d/1cd20Uz-b3IVZ3IBXuLzb1djSqlBFjzk9/view?usp=sharing](url)

quagla, 2022-08-06:
Thank you, I will have a look on Monday morning.

quagla, 2022-08-08:
Hi, just to make sure I understand: Did you do the skin by hand or do you have code that generates it? I've done a quick test with a skin code that we haven't released yet (adapted from the cloth code, so the skin is a flat yellow ribbon contained in the geometry as you can see in the attached picture) and I did't experienced issues in this example. So my best guess is that there is some bug in your code that produces the list of vertid. If you want you can share your skin code as well, although I'm not sure the bug will be easy to spot.

[image]

gvittrup, 2022-08-08:
I hand wrote the skin asset, which is wholly included in the XML file I provided, so you should be able to review it there. I extrapolated the data directly from an OBJ file exported from Blender, and inferred the vertex indices from the order of the coordinates. I tan a test case on the same composite rope with 3 links instead of the included 21, and this seemed to work just fine!

quagla, 2022-08-09:
I would recommend against inferring vertex numbers by hand which is very error prone. Making your custom skin code would be simpler. It would mean making a new skin function for a rope like the one already available for cloth
https://github.com/deepmind/mujoco/blob/main/src/user/user_composite.cc#L1046 and then it would work for any rope, not just the 21 links you sent.

closed 2022-08-20 by yuvaltassa