google-deepmind/mujoco #426: cannot see name of a joint

ShahhhVihaan, 2022-08-05:
Hi,
I'm a student trying to build a simulation of a quadruped leg. It's a simple model with three parts of the leg attached using two revolute joints.

I'm having some trouble simulating the urdf file that I have exported from solidworks.

I exported a urdf file from solidworks and the I converted it into a mujoco xml file using compile.
**The model renders perfectly okay but I cannot see the name of the second joint that is "knee_joint" in the window. Thus I cannot see the qpos. Is there a problem with my xml file? Can someone please help me fix this?**

<details>
<summary>XML</summary>
```
<mujoco model="2dof_leg_module_v3">
    <compiler angle="radian" meshdir="../meshes/" />
    <size njmax="500" nconmax="100" />
    <asset>
        <mesh name="hip_module" file="hip_module.STL" />
        <mesh name="upper_leg" file="upper_leg.STL" />
        <mesh name="lower_leg" file="lower_leg.STL" />
    </asset>
    <option gravity = "0 0 0"/>
    <worldbody>
        <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.752941 0.752941 0.752941 1" mesh="hip_module" />
        <geom type="mesh" rgba="0.752941 0.752941 0.752941 1" mesh="hip_module" />
        <body name="upper_leg" pos="0.066577 -0.011968 -0.0185">
            <inertial pos="-0.0578208 -0.0212739 -0.0555578" quat="0.640784 0.254162 0.299142 0.659781" mass="0.0942771" diaginertia="0.000160483 0.000154497 1.3996e-05" />
            <joint name="hip_joint" pos="0 0 0" axis="0 -1 0" />
            <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.752941 0.752941 0.752941 1" mesh="upper_leg" />
            <geom type="mesh" rgba="0.752941 0.752941 0.752941 1" mesh="upper_leg" />
            <body name="lower_leg" pos="-0.11534 -0.0324 -0.11089">
                <inertial pos="0.0478755 -0.0138956 -0.0466193" quat="0.281232 0.654101 0.648775 0.26861" mass="0.037401" diaginertia="0.000102588 0.000101714 1.74264e-06" />
                <joint name="knee_joint" pos="0 0 0" axis="0 1 0" />
                <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.752941 0.752941 0.752941 1" mesh="lower_leg" />
                <geom type="mesh" rgba="0.752941 0.752941 0.752941 1" mesh="lower_leg" />
            </body>
        </body>
    </worldbody>
</mujoco>
```
</details>

yuvaltassa, 2022-08-05:
Sorry, I don't understand 

> cannot see the name of the second joint that is "knee_joint" in the window

What window? Perhaps attach a screenshot?

ShahhhVihaan, 2022-08-05:
[image]
Here's the image.

yuvaltassa, 2022-08-06:
Please attach a loadable model and I will take a look. (I need the meshes. You can attach everything as a zip)

yuvaltassa, 2022-08-09:
I see the knee joint?
[image]

ShahhhVihaan, 2022-08-09:
May I ask what version of Mujoco you are using? I am using an 210.

For someone reason I cannot see it.

yuvaltassa, 2022-08-10:
I loaded this in MuJoCo 2.2.1 on an M1 Mac.

Can anyone else here replicate the OP's problem?

ShahhhVihaan, 2022-08-10:
I am having this problem in MuJoCo 2.2.1 and 210 in Ubuntu.

yuvaltassa, 2022-08-10:
Okay, I will try to reproduce on Ubuntu with 2.2.1

To clarify, you mean the prebuit binary, you are not building from source. Correct?

ShahhhVihaan, 2022-08-10:
I first tried the binary and I also built it from the source.

yuvaltassa, 2022-08-11:
Hi!

So the good news is we can repro this. It's a very strange issue. Note that this is "solved" by opening the control tab, in which case the slider is fine but the "Clear all" button loses the text. This is reversible. Close the control tab and the joint slider goes blank. So the symptom is, **the last GUI element on the RHS doesn’t get its text rendered, but only sometimes and only on Linux**. We will keep looking into this. Thanks for your reproducible bug report. Leaving the issue open.

[image]
[image]

nimrod-gileadi, 2023-04-20:
Seems to be fixed. Probably by some GLFW update.