google-deepmind/mujoco #430: how to get actual speed of a site without delay in mujoco

yinzikang, 2022-08-08:
Hi, I am trying to transfer from mujoco-py to mujoco.
I found that many valuable of `PyMjData` in mujoco-py were deleted, such as `site_xvelp`, `site_xvelr`.
So I tried to get the speed of a `site` with the function `mj_objectVelocity` as well as the sensor `velocimeter` only to got an unexpected result.

Here is a small test where a box falls freely with v0 = 0 at the height of 5m.
After running `sim.step()` in mujoco-py (or `mujoco.mj_step(model, data)` in mujoco) for the first time, `data.time` becomes 0.002, and the actual speed of the box should have been more than 0. 
However, except the function `sim.data.get_body_xvelp('box')`, every other way in mujoco-py and mujoco return a zero, such as `mujoco.mj_objectVelocity(model, data, mujoco.mjtObj.mjOBJ_BODY, id, vel2, True)` in mujoco and `sensordata` in both `mujoco-py` and `mujoco`.
The speed I get seems to be one step later than what it actually is.

So my question is: how to get actual speed of a site without delay in mujoco?
Any suggestions is welcomed, thanks so much.

The following is the output from mujoco-py and mujoco and corresponding files.
mujoco-py 2.1.2.14:
```
time: 0.002  sensor vel: [0. 0. 0.]  get_body_xvelp: [ 0.    0.   -0.02] pos: [0. 0. 5.]
time: 0.004  sensor vel: [ 0.    0.   -0.02]  get_body_xvelp: [ 0.    0.   -0.04] pos: [0.      0.      4.99996]
time: 0.006  sensor vel: [ 0.    0.   -0.04]  get_body_xvelp: [ 0.    0.   -0.06] pos: [0.      0.      4.99988]
time: 0.008  sensor vel: [ 0.    0.   -0.06]  get_body_xvelp: [ 0.    0.   -0.08] pos: [0.      0.      4.99976]
```

mujoco 2.2.1:
```
time: 0.002  sensor vel: [0. 0. 0.]  mj_objectVelocity: [0. 0. 0. 0. 0. 0.] pos: [0. 0. 5.]
time: 0.004  sensor vel: [ 0.    0.   -0.02]  mj_objectVelocity: [ 0.    0.    0.    0.    0.   -0.02] pos: [0.      0.      4.99996]
time: 0.006  sensor vel: [ 0.    0.   -0.04]  mj_objectVelocity: [ 0.    0.    0.    0.    0.   -0.04] pos: [0.      0.      4.99988]
time: 0.008  sensor vel: [ 0.    0.   -0.06]  mj_objectVelocity: [ 0.    0.    0.    0.    0.   -0.06] pos: [0.      0.      4.99976]
```

<details>
  <summary>vel.xml</summary>

```XML
<mujoco>
  <option gravity="0 0 -10"/>
  <worldbody>
    <light name="light" mode="targetbody" target="box" directional="true" pos="1 1 3" />
    <geom name='floor' pos='0 0 0' size='2.5 2.5 0.1'  type='plane'/>
    <body name="box" pos="0 0 5">
        <inertial pos="0 0 0" mass="1" diaginertia="0.1 0.1 0.1" />
        <geom type="box" size="0.1 0.1 0.1"  rgba="0 0 1 1"/>
        <joint type="free"/>
        <site name="speed_sensor" pos="0 0 0" rgba="1 0 0 0" size="0.01 0.01 0.01" type="sphere"/>
    </body>
  </worldbody>
  <sensor>
      <velocimeter name="xvelp" site="speed_sensor"/>
      <gyro name="xvelr" site="speed_sensor"/>
  </sensor>
</mujoco>
```

</details>

<details>
  <summary>mujoco</summary>

```
import mujoco
import numpy as np

model = mujoco.MjModel.from_xml_path(filename='vel.xml')
data = mujoco.MjData(model)
for i in range(5):
    mujoco.mj_step(model, data)

    vel1 = np.array(data.sensor('xvelp').data)
    id = mujoco.mj_name2id(model, mujoco.mjtObj.mjOBJ_BODY, 'box')
    vel2 = np.zeros(6)
    mujoco.mj_objectVelocity(model, data, mujoco.mjtObj.mjOBJ_BODY, id, vel2, True)
    pos = np.array(data.body('box').xpos)

    print('time:', data.time, ' sensor vel:', vel1,' mj_objectVelocity:', vel2, 'pos:', pos)

```

</details>

<details>
  <summary>mujoco-py</summary>

```
import mujoco_py as mp
import numpy as np

mjc_model = mp.load_model_from_path('vel.xml')
sim = mp.MjSim(model=mjc_model)
for i in range(5):
    sim.step()

    vel1 = np.array(sim.data.sensordata[:3])
    vel2 = np.array(sim.data.get_body_xvelp('box'))
    pos = np.array(sim.data.get_body_xpos('box'))

    print('time:', sim.data.time, ' sensor vel:', vel1,' get_body_xvelp:', vel2, 'pos:', pos)

```

</details>

yinzikang, 2022-08-08:
Furthermore, when I start debug and run `mujoco.mj_step(model, data)` for the first time.
the `qpos`, 'qvel' has changed, but the `sensordata`, `site_xpos` remains all zeros.
[image]

Is it a designed mechanism that I ignored or a bug?

saran-t, 2022-08-08:
You can check out [this section](https://mujoco.readthedocs.io/en/latest/computation.html#forward-dynamics) in our documentation for the description of what happens when you call `mj_step`. In particular, note that the forward kinematics calculation is _not_ repeated after time integration.

The most simple solution to update the entirety of `mjData` to reflect the state after time integration is via the `mj_forward` function. However, note that if you always call `mj_forward` immediately after each `mj_step`, your simulation loop will slow down quite considerably, since you would be repeating most of the computational stages.

A solution that `dm_control` uses is to replace the call to `mj_step` with `mj_step2` followed by `mj_step1` (see [here](https://github.com/deepmind/dm_control/blob/main/dm_control/mujoco/engine.py#L162)), which is equivalent to running the pipeline stages 18-24 followed by 1-17 (see implementation [here](https://github.com/deepmind/mujoco/blob/main/src/engine/engine_forward.c#L801)). If you want to do this, make sure that you call `mj_forward` once before you do `mj_step2` for the first time.

yinzikang, 2022-08-09:
Thank you so much for your detailed reply.
It solved my problem perfectly.
Have a nice day!

closed 2022-08-09 by yinzikang